BiancoRoyal/node-red-contrib-modbus #67: QUESTION: Modbus TX Enable Shield

candronikos, 2017-11-02:
Hi Klaus , really appreciate your assistance so far :)

For our application it's a requirement for the TX Enable toggled correctly. Is it possible to set a GPIO pin to use with Raspberry PI RS485 shields that use a GPIO pin?

biancode, 2017-11-02:
keywords could be ttyma0 or pi uart

candronikos, 2017-11-02:
Not sure what you mean here. Should I add them to the title?

biancode, 2017-11-02:
keywords could be ttyama0 or pi uart to search for

biancode, 2017-11-02:
You want to connect serial RS485 via GPIO on RPi, right?

candronikos, 2017-11-02:
If we need to bit bang on GPIO pins to make this work then I'm ok with it. Whatever works best.

We were using a UART to rs485 shield but found that the TX enable is tied to the TX pin. My colleague wants a proper shield to use and also drivers that will support it.

biancode, 2017-11-02:
OK, but I have never worked with the GPIO for serial or TX. I use [USB sticks](http://usb-rs485.eu/) with RS485 or RS232.

biancode, 2017-11-02:
Please, reopen if there is more or let me know about your experiences if it works well for you ;)

closed 2017-11-02 by biancode

candronikos, 2017-11-03:
Right, yes I was using USB converters to loop back the serial for testing. Unfortunately, it's a requirement that we do not use these in the field. I will definitely keep an eye on this project :)