BiancoRoyal/node-red-contrib-modbus #69: Problem when timout for more than 10 minutes

icarogregorio, 2017-11-06:
We have a continuous scan inject node (5 seconds) on the input of the modbus-flex-getter node and when the connection is broken, we got a status message "timeout". But when this connection is broken by more than 10 minutes the status message are changed for "not ready to read". But when we plug the connection again the node does not connect to the device again and freezes with this message: "not ready to read" and we can't connect to the device again without doing a deploy on the node-red. When we do a deploy after this event, the connection is established normally. 

PS: In this case, we don't need a queue.

Thanks

biancode, 2017-11-06:
a log file would help in which is the state of the client and which ERRNO it has

biancode, 2017-11-06:
without the queue, it should be in state ACTIVATED if the connection to the device is running

icarogregorio, 2017-11-07:
[image]

icarogregorio, 2017-11-07:
In the print of the log sent above the network connection of device was re-established after 17:16:14 after showing the error "FSM Reset on state activated". But the Flex Getter does not go back to work.

icarogregorio, 2017-11-07:
[Log - Failed.log](https://github.com/biancode/node-red-contrib-modbus/files/1450957/Log.-.Failed.log)

biancode, 2017-11-07:
the log file is not valid

evandrodp, 2017-11-07:
Hi @biancode, I am working with @icarogregorio to solve this problem. What is exactly a valid log file? This problem don't give us more information, just this sequence:
1 - Received message form previous node (loop while offline)
2 - FSM broken
3 - Try Reconnect
4 - Not connect anymore.

What we can get exactly for you?

icarogregorio, 2017-11-07:
Sorry... I don't understood what do you need ? This log file above is corrupted ?

biancode, 2017-11-07:
@evandrodp try to download your linked file ;)

biancode, 2017-11-07:
As I can see here in your Debug view, the device did not reconnect. It gets never the state CONNECTED after INIT and that means the hardware or Modbus library does not work correctly. see [FSM](https://github.com/biancode/node-red-contrib-modbus/wiki/Modbus-FSM)

evandrodp, 2017-11-07:
@biancode the file is ok for me. The same file with other extension:
[Log - Failed.txt](https://github.com/biancode/node-red-contrib-modbus/files/1451008/Log.-.Failed.txt)

icarogregorio, 2017-11-07:
`7 Nov 14:16:16 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071376005}
7 Nov 14:16:16 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071376064}
7 Nov 14:16:16 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071376127}
7 Nov 14:16:16 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071376182}
7 Nov 14:16:16 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071376245}
7 Nov 14:16:16 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071376334}
7 Nov 14:16:16 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071376398}
7 Nov 14:16:16 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071376479}
7 Nov 14:16:16 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071376545}
7 Nov 14:16:16 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071376595}
7 Nov 14:16:16 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071376655}
7 Nov 14:16:16 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071376722}
7 Nov 14:16:16 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071376783}
7 Nov 14:16:16 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071376939}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071377001}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071377061}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071377122}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071377177}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071377237}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071377299}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071377359}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071377414}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071377474}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071377533}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071377593}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071377684}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071377745}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071377831}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071377892}
7 Nov 14:16:17 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071377953}
7 Nov 14:16:18 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071378013}
7 Nov 14:16:18 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071378114}
7 Nov 14:16:18 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"a04e1110.8f884","timestamp":1510071378177}
7 Nov 14:16:18 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.send","msgid":"a04e1110.8f884","timestamp":1510071378279}
[... cut ...]
7 Nov 14:38:54 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"c2bfe688.9461e8","timestamp":1510072734931}
7 Nov 14:39:04 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"8a872e3.59485d","timestamp":1510072744940}
7 Nov 14:39:14 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"f717b55a.a0daf8","timestamp":1510072754950}
7 Nov 14:39:24 - [metric] {"level":99,"nodeid":"7de56d77.42fad4","event":"node.modbus-flex-getter.receive","msgid":"792a7886.cb17e8","timestamp":1510072764961}
`

evandrodp, 2017-11-07:
@biancode we have an SCAN -> [Instructions] -> modbus-flex-getter. Every 5 seconds our scan activate the modbus, but after more than 10 minutes, the node give an FSM Not Ready to Read error, when we put the equipment online again, modbus-flex-getter don't reconnect.

biancode, 2017-11-07:
@evandrodp I understand your problem, but without log information, I can not help you. There could be a modbusErrorHandling in the log file, but I can not download your attached file  - please see [DEBUG](https://github.com/biancode/node-red-contrib-modbus/wiki/DEBUG)

biancode, 2017-11-07:
You can send your log file via E-Mail, too.

biancode, 2017-11-07:
FSM Not Ready to Read means you try to read without a working connection

evandrodp, 2017-11-07:
@biancode I followed the DEBUG wiki, but now reading all source codes, i saw:
`let internalDebugLog = require('debug')('contribModbus:flex:connector')`
`let internalDebugLog = require('debug')('contribModbus:flex:getter')`
`let internalDebugLog = require('debug')('contribModbus:flex:server')`
`let internalDebugLog = require('debug')('contribModbus:flex:write')`
`let internalDebugLog = require('debug')('contribModbus:getter')`
`let internalDebugLog = require('debug')('contribModbus:queue')`
`let internalDebugLog = require('debug')('contribModbus:read')`
`let internalDebugLog = require('debug')('contribModbus:response')`
`let internalDebugLog = require('debug')('contribModbus:server')`
`let internalDebugLog = require('debug')('contribModbus:write')`
`de.biancoroyal.modbus.core.client.internalDebug = de.biancoroyal.modbus.core.client.internalDebug || require('debug')('contribModbus:core:client') // `eslint-disable-line no-use-before-define
`de.biancoroyal.modbus.core.client.modbusSerialDebug = de.biancoroyal.modbus.core.client.modbusSerialDebug || require('debug')('modbus-serial') // eslint-disable-line `no-use-before-define`

All reffers to "contribModbus", the correct term for DEBUG is: `DEBUG=contribModbus` ?

evandrodp, 2017-11-07:
@biancode I getted debug contribModbus:* without modbus-serial.
[modbus-debug.txt](https://github.com/biancode/node-red-contrib-modbus/files/1451348/modbus-debug.txt)

biancode, 2017-11-07:
Please, try to set reconnect delay to 2 sec. not 500 msec. if you reconnect to fast, then your device will not answer sometimes.

biancode, 2017-11-07:
please, could you do that again with modbus-serial in the log file?

evandrodp, 2017-11-07:
@biancode But this changes only increase the time of the problem. With 500ms the problem occurs about 10~20 min. If i increase to 2s maybe it occurs about 40 ~60min, but not solve the problem.

When machine get State = BROKEN, it stop all retries, on modbus-client:
`    node.on('readModbus', function (msg, cb, cberr) {
      let state = node.statlyMachine.getMachineState()

      if (node.messageAllowedStates.indexOf(state) === -1) {
        cberr(new Error('FSM Not Ready To Read'), msg)
        return
      }
`

biancode, 2017-11-07:
not because of the code - it is because of the TCP device does nothing and this is not OK - you're to fixed to think you have found an error in code maybe - you should try to see the whole picture, please.

evandrodp, 2017-11-07:
@biancode We using 3G connection in a industry, this difficult ours tests. Tomorrow I will generate new log with serial and I will upload soon as possible.

biancode, 2017-11-07:
otherwise, it needs a timeout on connecting TCP process

biancode, 2017-11-07:
the ERRNO would help if there is one on 3G connection try

biancode, 2017-11-08:
... and if you press Deploy, then it works again?

evandrodp, 2017-11-08:
@biancode, yes, if deploy again, the connection restart.

biancode, 2017-11-08:
the ERRNO is EHOSTUNREACH on which we have to react on TCP

biancode, 2017-11-08:
try v2.2.2 and log again, please

biancode, 2017-11-09:
feel free to [become a happy backer](https://bianco-royal.cloud/supporter/)

closed 2017-11-09 by biancode

biancode, 2017-11-09:
minimum give us a star to that repository 
[image]