BiancoRoyal/node-red-contrib-modbus #73: Relationship between "node-red-node-serialport" and "node-red-contrib-modbus"

kangskii, 2017-11-29:
Hello,
I found on https://www.npmjs.com/package/node-red-contrib-modbus that node-red-node-serialport is one dependency library of node-red-contrib-modbus.

Q1: Does it mean node-red-contrib-modbus internally invoke the two nodes： serial in and serial out in node-red-node-serialport ?

To my understanding, if the node-red-contrib-modbus is used, we don't need to drag the node-red-node-serialport nodes into the modbus flow.

Q2: Actually, I now had some issues in using serial out node in node-red-node-serialport. If I now use some nodes in node-red-contrib-modbus, will I have the SAME problem ?

Thanks.

biancode, 2017-11-29:
Hi,

I'm sorry, node-red-node-serialport is not a dependency it is a separate package from another developer. My package uses modbus-serial too if you mean that in your Q1. Do you mean that library? 

If your question is, how you could use serial from my package, then the answer is: Just select Serial or Serial Expert in the config node (connection client). Default selection is TCP.

[image]

[image]

[image]

Does it help?

Happy wiring!
Klaus

kangskii, 2017-11-30:
Hi， 
Thanks for your reply. 
There are 8 dependencies for your package: bson, debug, modbus-serial, node-modbus, serialport, stampit, stately.js, vm2. To make your package run on Node-RED, do I have to install them additionally ?

biancode, 2017-11-30:
Hi,

The dependencies are managed by npm install. There is nothing to install additionally, but read the Github Wiki and NPM information about npm install.

To get more about the nodes and what they can do for your flow, please hover over in the list or drag and drop them to the flow window and watch the info tab at the right. 

Watch my YouTube videos and you will learn how it could work for you. 😉

Happy learning!
Klaus _

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

biancode, 2017-11-30:
Search YouTube for my package name to find...

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2017-11-30 by biancode

kangskii, 2017-11-30:
Hi,
Thanks for you quick response.
I had limited access to youtube.

In the node-red-contrib-modbus library, there is no modbus client. And I find it in the Manage Palette, see attached figure, but the Modbus client is a dashed box.

Could you please give some suggestions if I want to develop a modbus client in node-red-contrib-modbus ?

[image]

biancode, 2017-11-30:
Hi Eric,

Please, use the Node-RED documentation on their website first and learn how to work with that tool. There are many courses and tutorials about Node-RED. ;)

The modbus-client node is hidden by Node-RED, because it is a configuration node. As I said, please teach yourself about that basics of Node-RED. You could also contact the community via Slack in the Node-RED channel.

Happy learning!
Klaus

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

kangskii, 2017-11-30:
Hi Klaus,

Well received. Thank you so much.

With regards,

Eric 

> 在 2017年11月30日，19:34，Klaus Landsdorf <notifications@github.com> 写道：
> 
> Klaus