BiancoRoyal/node-red-contrib-modbus #74: Possible memory leak in Flex Getter

MonkeyDo, 2017-12-05:
Hello!

I am making heavy use of the nodes you created, and have found the improvements over time to be very valuable. Thanks a lot for you work on this library and node-modbus.

I think I may have identified a memory leak. You can find a reduced test case [as a gist here](https://gist.github.com/MonkeyDo/0b2a5eb06e8408aeaffb2b18e595fc14).

My set-up:
I am running node-red in a docker container. There are 6 modbus-flex-getter nodes, each polling from a Wago PLC on a local network. The modbus client config nodes are set to queue commands.
Each flex-getter nodes receives multiple input messages every second with read instructions with various fc codes and quantities.
I am on a testing setup: 5 of the PLCs are simulated on another computer in the network, and one is a real PLC.
I am using node-red-contrib-modbus@2.3.2.

The issue:
Node-red occasionally crashes after running for a certain arbitrary amount of time. Using node's inspect protocol to debug what is happening, I have noticed a dramatic and continuous memory increase. The memory footprint of my node-red application more than quadrupled in size from its base ~300Mb footprint.
Comparing memory heap snapshots with Chrome devtools, I noticed a huge amount of arrays originating from ModbusFlexGetter nodes, all the same allocation size. Here is a screenshot as an example:

[image]

In this particular session, with the application running for half a day, I have about 50 000 such arrays. That does not seem right.
Can you help me shed some light on what is happening?
I am happy to provide any information you might need.

MonkeyDo, 2017-12-05:
Here is a screenshot of memory usage using the same flow running with node-red-contrib-modbus@2.0.1:

[image]

MonkeyDo, 2017-12-05:
After some more digging, I suspect the command queues of my getter nodes are increasing linearly in size.
I am surprised they are allowed to grow indefinitely, and saw that the Modbus Queue Info node can be used to purge the queue. Unfortunately that is not satisfactory for my use case.

I still don't understand why the commands are piling up. Each getter node gets between 3 and 6 commands each second; that should be plenty of time to process those commands.
did you encounter that problem?

biancode, 2017-12-05:
Hello,

i the most cases it grows up by writing or reading/polling to fast to your Modbus device.
You will see that in a growing queue info.

The messages could grow if there are to many errors in reads and writes. That is a point, which I could improve in that case. 

Who does the queue grow? 

cya
Klaus

biancode, 2017-12-05:
Are there Modbus errors?

biancode, 2017-12-05:
[image]

biancode, 2017-12-05:
gc working well for me - how does it work in your case?

[image]

MonkeyDo, 2017-12-06:
Thanks for your help.
The problem is not with node-red-contrib-modbus but indeed with my queue management.

What I really need is a signal when the queue is at 0 to trigger a new round of polls, as in my use case I can't rely on adjusting the poll interval.
I am going to fork the repo and try adding to the Queue Info node a signal on low / low low level the same way you have on high / high high level. I will let you know if this is successful, in case you want to merge it.

biancode, 2017-12-06:
Did you watch the [YouTube video](https://www.youtube.com/watch?v=ND9ZYkMOwZs&list=PLdcYFhtEB7okPHU1-NUYTeKFNbJOW_-gH&index=11) about that? You could use that functionality.

MonkeyDo, 2017-12-06:
I did, and know about the functionality.
Unfortunately, for my application it is critical that I do not lose write messages that might be in the queue.
Better to trigger a new round of reads once the queue is empty.

MonkeyDo, 2017-12-06:
I was able to set up my read pool without modifying the library.
It took me a little bit to figure out how to set the Queue Info node to get a signal when the queue is empty. I had to set low low level to -1 and low level to 1. That way I get a signal when I want.

I also wanted to throttle that loop so that instead of starting a new round of reading as soon the previous one was finished, it limits it to once a second.

Here is the code I used to achieved my rate-limited, queue-conscious read loop. It is set up with a fake modbus server with a long response delay to simulate slow network:

`[{"id":"94be6227.1ae7c","type":"modbus-flex-getter","z":"b6aeedf6.a4a9b","name":"modbus flex getter","showStatusActivities":true,"showErrors":true,"server":"f2a898c7.def828","x":450,"y":120,"wires":[["4a123cc2.da9c64"],[]]},{"id":"cdb083b0.414d8","type":"function","z":"b6aeedf6.a4a9b","name":"multiple read instructions","func":"var msgOne = {\n    topic:\"polling_topic_one\",\n    payload:{\"fc\":3, \"address\": 12, \"quantity\":100}\n};\nvar msgTwo = {\n    topic:\"polling_topic_two\",\n    payload:{\"fc\":3, \"address\": 0, \"quantity\":67}\n};\nnode.send([[msgOne,msgTwo]]);","outputs":"1","noerr":0,"x":170,"y":120,"wires":[["94be6227.1ae7c"]]},{"id":"3b6c63fd.77b8cc","type":"switch","z":"b6aeedf6.a4a9b","name":"read results splitter","property":"input.topic","propertyType":"msg","rules":[{"t":"eq","v":"polling_topic_one","vt":"str"},{"t":"eq","v":"polling_topic_two","vt":"str"}],"checkall":"false","outputs":2,"x":890,"y":120,"wires":[["9b9d7964.856bb8"],["b820d37a.4dbff"]]},{"id":"c8cf78f7.a0ddd8","type":"delay","z":"b6aeedf6.a4a9b","name":"max 1/sec","pauseType":"rate","timeout":"1","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":true,"x":310,"y":180,"wires":[["cdb083b0.414d8"]]},{"id":"999c9028.50f7a","type":"switch","z":"b6aeedf6.a4a9b","name":"if queue empty","property":"items","propertyType":"msg","rules":[{"t":"eq","v":"0","vt":"num"}],"checkall":"true","outputs":1,"x":140,"y":180,"wires":[["c8cf78f7.a0ddd8"]]},{"id":"deae15ed.9ab778","type":"modbus-queue-info","z":"b6aeedf6.a4a9b","name":"","unitid":"","lowLowLevel":"-1","lowLevel":"1","highLevel":150,"highHighLevel":300,"server":"f2a898c7.def828","errorOnHighLevel":false,"x":219.5,"y":234,"wires":[["999c9028.50f7a"]]},{"id":"4a123cc2.da9c64","type":"change","z":"b6aeedf6.a4a9b","name":"extract data","rules":[{"t":"set","p":"payload","pt":"msg","to":"payload.data","tot":"msg"},{"t":"delete","p":"values","pt":"msg"}],"action":"","property":"","from":"","to":"","reg":false,"x":670,"y":120,"wires":[["3b6c63fd.77b8cc"]]},{"id":"8f0dd8b6.bfce78","type":"modbus-server","z":"b6aeedf6.a4a9b","name":"","logEnabled":false,"serverPort":10502,"responseDelay":"1200","delayUnit":"ms","coilsBufferSize":10000,"holdingBufferSize":10000,"inputBufferSize":10000,"x":720,"y":340,"wires":[[],[],[]]},{"id":"9b9d7964.856bb8","type":"debug","z":"b6aeedf6.a4a9b","name":"","active":true,"console":"false","complete":"false","x":1150,"y":102,"wires":[]},{"id":"b820d37a.4dbff","type":"debug","z":"b6aeedf6.a4a9b","name":"","active":true,"console":"false","complete":"false","x":1150,"y":140,"wires":[]},{"id":"f2a898c7.def828","type":"modbus-client","z":"","name":"Modbus server","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":true,"tcpHost":"127.0.0.1","tcpPort":"10502","tcpType":"DEFAULT","serialPort":"/dev/ttyS0","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"","unit_id":"","commandDelay":"","clientTimeout":"5000","reconnectTimeout":"5000"}]`

MonkeyDo, 2017-12-14:
I spoke too soon. I do see a linear increase in memory. The ModbusFlexGetter bufferMessageList keeps growing and growing, and the entries are not being removed even though the queue works as expected.

Look at the number of entries in the bufferMessageList !
[image]

biancode, 2017-12-15:
Please, send me a flow of what are you doing and with that problem!

biancode, 2017-12-19:
Please, test with v2.5.0

closed 2017-12-19 by biancode

MonkeyDo, 2017-12-20:
Yes, looks like that was it !
Thanks to @milosjovac and yourself !