BiancoRoyal/node-red-contrib-modbus #75: High memory usage in modbus-getter

milosjovac, 2017-12-19:
Hi Klaus,

I have a problem with memory consumption of the modbus-getter on the raspberry pi 3. Memory continuously piles up till ENOMEM error starts to popup. It behaves the same on 2 environments:

Linux 4.9.59-v7+ arm LE
Node-RED version: v0.17.5
Node.js  version: v8.9.3
node-redpcontrib-modbus version: v2.4.1

Linux 4.9.59-v7+ arm LE
Node-RED version: v0.17.4
Node.js  version: v4.8.2
node-redpcontrib-modbus version: v2.3.2

[image]

Real flow logic is a bit more complex but leaking problem is reproducible with a [flow from the image.](https://gist.github.com/milosjovac/ba02adf79c851d7fd9c661531f4778ff)

milosjovac, 2017-12-19:
Ok, I have a new hint about the problem

[image]

milosjovac, 2017-12-19:
Ok, I think I found the problem. [In this line](https://github.com/biancode/node-red-contrib-modbus/blob/fa51fa895855d2a634beca791c63e40050ad49fd/src/modbus-getter.js#L77) You set new message with messageId, but this is is always different (incremental), and I do not see elsewhere in code that you call node.bufferMessageList.delete() or bufferMessageList.clear()

[image]

milosjovac, 2017-12-19:
Or [you do](https://github.com/biancode/node-red-contrib-modbus/blob/fa51fa895855d2a634beca791c63e40050ad49fd/src/core/modbus-core.js#L22). Hmm..

biancode, 2017-12-19:
it asnyc in 
`    function buildMessage (values, response, msg) {
      let origMsg = mbCore.getOriginalMessage(node.bufferMessageList, msg) || msg`

biancode, 2017-12-19:
maybe you have a problem with Map.delete in your node version?

biancode, 2017-12-19:
messages are deleted on success and error

milosjovac, 2017-12-19:
Yes, I see. I will have to check. It behaves the same on 2 node versions btw. Will let you know about the results.

milosjovac, 2017-12-19:
[origMsg](https://github.com/biancode/node-red-contrib-modbus/blob/fa51fa895855d2a634beca791c63e40050ad49fd/src/core/modbus-core.js#L22) has no property in payload called messageId

So map is always deleted with undefined.

You should just omit .paylod. in between and it should work. 

[image]

biancode, 2017-12-19:
Yes, I see.

biancode, 2017-12-19:
Please, test with v2.5.0

closed 2017-12-19 by biancode

milosjovac, 2017-12-19:
Tested. Memory footprint is now OK 👍