BiancoRoyal/node-red-contrib-modbus #76: Modbus non respond RTU-BUFFERED

ciaowecia81, 2018-01-03:
Hello,
I use the Modbus read module in RTU-BUFFERED mode.
My ethernet to serial gateway is a Moxa Nport
If I try to connect to the modbus device directly from a tool like Putty or something similar, it responds correctly.
If I use the RTU-BUFFERED mode , it doesn't respond.
Is possible to view the command request send to the TCP client ?

I try to create a local TCP connection to read the command sent to the client, and I see that it use a FC \x06 for Holding Register read.
are there an option to decide to use the FC \x03 ?
Thanks

biancode, 2018-01-03:
Hi Marco,

please, try:
`DEBUG=modbus-serial* node-red -v`

and use DEFAULT TCP

[more](https://github.com/biancode/node-red-contrib-modbus/wiki/DEBUG)

ciaowecia81, 2018-01-08:
Hello,

I will try it.
For now I connect my device by a modbus RTU to TCP converter and it does work.
I'll try with the debug and keep you updated
Many thanks

closed 2018-01-17 by biancode