BiancoRoyal/node-red-contrib-modbus #78: TypeError: msg.payload.value.indexOf is not a function

tomtom30, 2018-01-19:
Hi,

I'am using your nice modbus contrib but i have a little issue when I use boolean with modbus-flex-write like this :

`[{"id":"b7568f9b.53757","type":"ui_switch","z":"5cd16ef.103a39","name":"","label":"Lampe Automate","group":"228ed415.3f3ba4","order":1,"width":0,"height":0,"passthru":false,"decouple":"false","topic":"Lumiere Bureau","style":"","onvalue":"true","onvalueType":"bool","onicon":"","oncolor":"","offvalue":"false","offvalueType":"bool","officon":"","offcolor":"","x":390,"y":200,"wires":[["b820c6ad.d288d8"]]},{"id":"b820c6ad.d288d8","type":"function","z":"5cd16ef.103a39","name":"Write 25281 by FC5","func":"msg.payload = { 'value': msg.payload, 'fc': 5, 'unitid': 1, 'address': 25281 , 'quantity': 1 };\nreturn msg;","outputs":1,"noerr":0,"x":692.375,"y":225,"wires":[[]]},{"id":"228ed415.3f3ba4","type":"ui_group","z":"","name":"Bureau","tab":"6891b93c.bea8a","order":2,"disp":true,"width":"6"},{"id":"6891b93c.bea8a","type":"ui_tab","z":"","name":"Lumieres","icon":"dashboard","order":6}]`

I have this error in console log :

`20 Jan 00:18:51 - [error] [modbus-flex-write:8733a326.9ee8f] TypeError: msg.payload.value.indexOf is not a function
    at ModbusFlexWrite.<anonymous> (/usr/src/node-red/node_modules/node-red-contrib-modbus/modbus/modbus-flex-write.js:104:54)
    at emitOne (events.js:96:13)
    at ModbusFlexWrite.emit (events.js:188:7)
    at ModbusFlexWrite.Node.receive (/usr/src/node-red/node_modules/node-red/red/runtime/nodes/Node.js:215:14)
    at FunctionNode.Node.send (/usr/src/node-red/node_modules/node-red/red/runtime/nodes/Node.js:202:14)
    at sendResults (/usr/src/node-red/node_modules/node-red/nodes/core/core/80-function.js:52:18)
    at FunctionNode.<anonymous> (/usr/src/node-red/node_modules/node-red/nodes/core/core/80-function.js:198:21)
    at emitOne (events.js:96:13)
    at FunctionNode.emit (events.js:188:7)
    at FunctionNode.Node.receive (/usr/src/node-red/node_modules/node-red/red/runtime/nodes/Node.js:215:14)
    at FunctionNode.Node.send (/usr/src/node-red/node_modules/node-red/red/runtime/nodes/Node.js:202:14)
    at sendResults (/usr/src/node-red/node_modules/node-red/nodes/core/core/80-function.js:52:18)
    at FunctionNode.<anonymous> (/usr/src/node-red/node_modules/node-red/nodes/core/core/80-function.js:198:21)
    at emitOne (events.js:96:13)
    at FunctionNode.emit (events.js:188:7)
    at FunctionNode.Node.receive (/usr/src/node-red/node_modules/node-red/red/runtime/nodes/Node.js:215:14)
    at JoinNode.Node.send (/usr/src/node-red/node_modules/node-red/red/runtime/nodes/Node.js:137:22)
    at completeSend (/usr/src/node-red/node_modules/node-red/nodes/core/logic/17-split.js:311:18)
    at JoinNode.<anonymous> (/usr/src/node-red/node_modules/node-red/nodes/core/logic/17-split.js:461:21)
    at emitOne (events.js:96:13)
    at JoinNode.emit (events.js:188:7)
    at JoinNode.Node.receive (/usr/src/node-red/node_modules/node-red/red/runtime/nodes/Node.js:215:14`

I have this error when msg.payload comes to 'true'

Can you help me?
Thanks

biancode, 2018-01-21:
Hi,

please, check your versions of node.js and the package and let me know them.
It is not to reproduce on my side.

tomtom30, 2018-01-21:
Hi,
I am on docker with :
npm version 3.10.10
node.js version 6.11.3
node-red-contrib-modbus version 3.0.3

thanks for your help

biancode, 2018-01-21:
Hi,
did you check the modbus package version via "Manage palette"?

[image]

If yes, then it is all OK.

The error means, that there is a string input which is not "true" or "false"

[image]

Is it an import or built with the actual running Node-RED?

biancode, 2018-01-21:
`[{"id":"7cfb14b1.cc376c","type":"tab","label":"Flow 1"},{"id":"47a9bdc8.4b3e0c","type":"ui_switch","z":"7cfb14b1.cc376c","name":"","label":"switch","group":"9a62a347.8b4ce8","order":0,"width":0,"height":0,"passthru":true,"decouple":"false","topic":"","style":"","onvalue":"true","onvalueType":"bool","onicon":"","oncolor":"","offvalue":"false","offvalueType":"bool","officon":"","offcolor":"","x":290,"y":200,"wires":[["8acfdfd3.e7fcb"]]},{"id":"8acfdfd3.e7fcb","type":"function","z":"7cfb14b1.cc376c","name":"","func":"if(msg.payload) {\n    msg.payload = { 'value': true, 'fc': 5, 'unitid': 1, 'address': 0 , 'quantity': 1 };\n}\nelse {\n    msg.payload = { 'value': false, 'fc': 5, 'unitid': 1, 'address': 0 , 'quantity': 1 };\n}\n\nreturn msg;","outputs":1,"noerr":0,"x":420,"y":200,"wires":[["cd827494.abf5d"]]},{"id":"3423ad19.17cb3a","type":"inject","z":"7cfb14b1.cc376c","name":"","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"x":130,"y":180,"wires":[["47a9bdc8.4b3e0c"]]},{"id":"7c6b2d0a.0978dc","type":"inject","z":"7cfb14b1.cc376c","name":"","topic":"","payload":"false","payloadType":"bool","repeat":"","crontab":"","once":false,"x":130,"y":220,"wires":[["47a9bdc8.4b3e0c"]]},{"id":"49853982.adb438","type":"modbus-server","z":"7cfb14b1.cc376c","name":"","logEnabled":false,"serverPort":"18502","responseDelay":100,"delayUnit":"ms","coilsBufferSize":10000,"holdingBufferSize":10000,"inputBufferSize":10000,"x":420,"y":380,"wires":[[],[],[]]},{"id":"cd827494.abf5d","type":"modbus-flex-write","z":"7cfb14b1.cc376c","name":"","showStatusActivities":false,"showErrors":false,"server":"36ec4f72.39c82","x":620,"y":200,"wires":[["2d4aebd8.5300dc"],["fa94cd5b.9a4b8"]]},{"id":"fa94cd5b.9a4b8","type":"modbus-response","z":"7cfb14b1.cc376c","name":"","registerShowMax":20,"x":870,"y":220,"wires":[]},{"id":"2d4aebd8.5300dc","type":"debug","z":"7cfb14b1.cc376c","name":"","active":true,"console":"false","complete":"false","x":850,"y":180,"wires":[]},{"id":"61f378f5.cfa9b","type":"debug","z":"7cfb14b1.cc376c","name":"","active":true,"console":"false","complete":"false","x":610,"y":260,"wires":[]},{"id":"6091f019.8ceaa8","type":"modbus-getter","z":"7cfb14b1.cc376c","name":"","showStatusActivities":false,"showErrors":false,"unitid":"","dataType":"Coil","adr":"0","quantity":"1","server":"36ec4f72.39c82","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":420,"y":280,"wires":[["61f378f5.cfa9b"],["83967d25.56651"]]},{"id":"83967d25.56651","type":"modbus-response","z":"7cfb14b1.cc376c","name":"","registerShowMax":20,"x":630,"y":300,"wires":[]},{"id":"df054f5f.8d13c8","type":"inject","z":"7cfb14b1.cc376c","name":"","topic":"","payload":"","payloadType":"date","repeat":"10","crontab":"","once":false,"x":230,"y":280,"wires":[["6091f019.8ceaa8"]]},{"id":"9a62a347.8b4ce8","type":"ui_group","z":"","name":"Bureau","tab":"4960ce05.62e618","order":2,"disp":true,"width":"6"},{"id":"36ec4f72.39c82","type":"modbus-client","z":"7cfb14b1.cc376c","name":"","clienttype":"tcp","bufferCommands":true,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"18502","tcpType":"DEFAULT","serialPort":"/dev/ttyUSB","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"},{"id":"4960ce05.62e618","type":"ui_tab","z":"","name":"Lumieres","icon":"dashboard","order":6}]`

tomtom30, 2018-01-21:
Ok i have found an issue.
I have 2.2.3 version with manage palette but 3.0.3 with npm in command shell.
I think it is my problem.
I try to resolve it and come back to you

biancode, 2018-01-21:
OK and yes 2.2.3 has that problem.

tomtom30, 2018-01-21:
Ok really thanks, it was this problem, difference between npm command version and palette version.
I think it was a wrong path installation.
No problem with 3.0.3 version.
Thanks 
thomas

closed 2018-01-21 by tomtom30