BiancoRoyal/node-red-contrib-modbus #79: Modbus RTU read with no result

rjbaat, 2018-01-31:
Hi, 
I had node-red-contrib-modbus working on an embeddedpi on the ttyAMA0 port. It is still sending the reading correct, but not receiving the results anymore. Insted i get a timeout.

I have connected the wires to a usb to rs485 converter to a windows laptop running Simply Modbus slave software.

**What works:**
When i use 2x the rs485 converters.
1 rs485 usb converter connected to an rpi3 <--> 1 rs485 usb converter connected to a windows laptop. Then when i do a read FC3 unit 1 from 0-2 it does read the registers and get positive return. The slave software is showing the right conditions also.

**What doesn't work:**
When i use the ttyAMA0 port on the embeddedpi <--> rs485 usb converter on a windows laptop. 
The slave software is showing me the right read parameters that i send (FC3 unit 1 from 0-2). It is showing it is returning data aswell. But it looks like the data never arrives on the embeddedpi. 

Is there something i can do to make it work again? 

In the attachment i have added a screen of the slave software. 

[image]

biancode, 2018-01-31:
Hi,

Please, I need to know all versions:

* nodejs
* node-red
* node-red-contrib-modbus

biancode, 2018-01-31:
What was the version before as it worked?

rjbaat, 2018-02-02:
I have tested the box with mbpoll aswell. But it has the same problem. Then i tested with a plain serial connection from box to box and box to windows. The result are:

box -> windows (message received = OK)
box <- windows (message NOT received)
box -> box (message received = OK)
box <- box (message received = OKE)

It seems that its a problem with the rs485 to usb converters. Since the RX lights on the box doesn't light up on the RX end. Only on the TX. When i get new converters i will try again.

closed 2018-02-02 by rjbaat