BiancoRoyal/node-red-contrib-modbus #80: ModeBus with serial interface

shahbour, 2018-02-05:
Hello

This is not a buy but a question seeking help as I could not find them in documents.

I am trying to use this module to connect to a battery that supports Modbus and read data. Using the software that came with battery is working so my connections and battery are working, but I am not able to configure the different parameters.

I am always getting timeout when trying simple getter as below 

```
[{"id":"88deba32.459178","type":"tab","label":"Flow 1"},{"id":"7443f003.00d78","type":"modbus-getter","z":"88deba32.459178","name":"","showStatusActivities":true,"showErrors":true,"unitid":"38","dataType":"Coil","adr":"4096","quantity":"1","server":"e71eb878.ef5e08","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":340,"y":80,"wires":[["ae00dbc8.95ef18"],[]]},{"id":"73612a54.d8e954","type":"inject","z":"88deba32.459178","name":"","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"x":140,"y":40,"wires":[["7443f003.00d78"]]},{"id":"ae00dbc8.95ef18","type":"debug","z":"88deba32.459178","name":"","active":true,"console":"false","complete":"true","x":530,"y":40,"wires":[]},{"id":"e71eb878.ef5e08","type":"modbus-client","z":"88deba32.459178","name":"battery1","clienttype":"serial","bufferCommands":true,"stateLogEnabled":true,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"COM3","serialType":"RTU","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":38,"commandDelay":10,"clientTimeout":1000,"reconnectTimeout":2000}]
```

From the document of the battery, I concluded the following 

_BMS code switch value range 0～15，The BMS communication base address 38，BMS communication address is sum of code switch value and communication base address._
**UnitId = 38**

RegId for voltage is 0x1000 so I used Address **4096**

and I used quantity 1.

Is my conclusion correct about the different settings, is their a definition of what is UnitId, Address, RegId

Thanks

```
  contribModbus:core:client Client -> {"info":"queue read msg","message":{"value":1517841108577,"unitid":"38","fc":1,"address":"4096","quantity":"1","messageId":"5a786ad41c82fb1d1c27e4f0"},"state":"EMPTY","queueLength":1} Serial@COM3:9600bit/s default Unit-Id: 38 +1ms
  contribModbus:getter {"topic":"7443f003.00d78","payload":{"value":1517841108577,"unitid":"38","fc":1,"address":"4096","quantity":"1","messageId":"5a786ad41c82fb1d1c27e4f0"},"_msgid":"d053c53f.5f1068","queueNumber":0,"queueUnit":38} +19ms
  contribModbus:core:client Client -> {"state":"QUEUEING","message":"dequeue command serial","delay":10} Serial@COM3:9600bit/s default Unit-Id: 38 +15ms
  contribModbus:core:client Client -> {"type":"queue check","unitid":38,"sendAllowed":true,"queueLength":1} Serial@COM3:9600bit/s default Unit-Id: 38 +0ms
  contribModbus:core:client Client -> {"type":"serial sending and wait","unitid":38,"queueLength":0,"sendAllowedForNext":false,"delay":10} Serial@COM3:9600bit/s default Unit-Id: 38 +3ms
  contribModbus:core:client Client -> {"info":"read msg","message":{"value":1517841108577,"unitid":"38","fc":1,"address":"4096","quantity":"1","messageId":"5a786ad41c82fb1d1c27e4f0"},"unitid":38,"timeout":1000,"state":"QUEUEING"} Serial@COM3:9600bit/s default Unit-Id: 38 +3ms
  contribModbus:core:client Client -> {"info":"queue response activate sending","message":{"value":1517841108577,"unitid":"38","fc":1,"address":"4096","quantity":"1","messageId":"5a786ad41c82fb1d1c27e4f0"}} Serial@COM3:9600bit/s default Unit-Id: 38 +1s
  contribModbus:core:client Client -> event: activate old: QUEUEING new: ACTIVATED Serial@COM3:9600bit/s default Unit-Id: 38 +3ms
  contribModbus:core:client Client -> event: queue old: ACTIVATED new: QUEUEING Serial@COM3:9600bit/s default Unit-Id: 38 +4ms
  contribModbus:core Remove Message In:<5a786ad41c82fb1d1c27e4f0> Out:<5a786ad41c82fb1d1c27e4f0> +2m
  contribModbus:getter Timed out +1s
  modbus-serial modbusErrorHandling:Timed out +15s
  contribModbus:core:client Client -> {"state":"QUEUEING","message":"dequeue command serial","delay":10} Serial@COM3:9600bit/s default Unit-Id: 38 +12ms
  contribModbus:core:client Client -> {"type":"queue check is not a unit","unitid":null} Serial@COM3:9600bit/s default Unit-Id: 38 +3ms
```

biancode, 2018-02-05:
Hi, do you have a specification for your BMS?

[like that](http://www.batteryspace.com/prod-specs/BMS%20MODBUS%20Specification.pdf)

biancode, 2018-02-05:
... maybe it is FC4 not FC1?
... if you fill the Unit-ID in connector it don't has to be filled out in the getter ;)
... normally it is RTU buffered the most
... please send me your BMS manual for Modbus. :)

`[{"id":"38c9a6ac.2e70aa","type":"tab","label":"Flow 1"},{"id":"5223d303.ad307c","type":"modbus-getter","z":"38c9a6ac.2e70aa","name":"","showStatusActivities":true,"showErrors":true,"unitid":"","dataType":"InputRegister","adr":"4096","quantity":"1","server":"477fa506.f0a844","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":340,"y":80,"wires":[["b84dd349.d1a148"],[]]},{"id":"8dad8c63.686a6","type":"inject","z":"38c9a6ac.2e70aa","name":"","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"x":140,"y":40,"wires":[["5223d303.ad307c"]]},{"id":"b84dd349.d1a148","type":"debug","z":"38c9a6ac.2e70aa","name":"","active":true,"console":"false","complete":"true","x":530,"y":40,"wires":[]},{"id":"477fa506.f0a844","type":"modbus-client","z":"38c9a6ac.2e70aa","name":"battery1 buffered","clienttype":"simpleser","bufferCommands":true,"stateLogEnabled":true,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"COM3","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"38","commandDelay":"10","clientTimeout":"1000","reconnectTimeout":"2000"}]`

shahbour, 2018-02-05:
Hello 

Thanks for your help, I will try your suggestion above.

 please find the document for the battery Modbus https://drive.google.com/open?id=13SSdmbsvifwuzHyZLjf8FbgQ4IDQtLl2  

Thanks

biancode, 2018-02-05:
address = dec (start from zero) ...

0xFFFF = 65535 (65534) is max
0x1000 = 4096 (4095) with factor 0.01

with UnitID from 38 to 43 ( 38 + code switch value [0~15] ) - What is the code switch value here?
please test FC4 from address 4095 with quantity 9 **or** test FC4 from address 4096 also with quantity 9

all Flags (16Bit UInt) registers are to split into separate Bits

happy reading ...

shahbour, 2018-02-06:
Hello

Thanks for your help, I tried different combination but none did work so I installed Serial Port Monitor to compare what I am sending using node-red and the factory program below are the results (note that the program does retrieve a lot of info voltage, temperature, alarms...

Using node-red
```
[06/02/2018 12:09:06] - Open port COM3 (C:\Program Files\nodejs\node.exe) 
 
[06/02/2018 12:09:15] Written data (COM3) 
    26 04 10 00 00 09 32 1b                           &.....2.        
```

Using the factory program (It poll all the data every 5 seconds)
```
[06/02/2018 11:51:31] Written data (COM3)	
    7e 32 35 30 31 34 36 43 31 30 30 30 30 46 44 39   ~250146C10000FD9	
    41 0d                                             A.              	
[06/02/2018 11:51:31] Read data (COM3)	
    33 33 31 33 36 32 44 33 31 32 45 33 31 33 30 32   331362D312E31302	
    30 32 30 32 30 46 35 38 34 0d                     02020F584.      	
[06/02/2018 11:51:36] Written data (COM3)	
    7e 32 35 30 31 34 36 43 32 30 30 30 30 46 44 39   ~250146C20000FD9	
    39 0d                                             9.              	
[06/02/2018 11:51:36] Read data (COM3)	
    7e 32 35 30 30 34 36 30 30 36 30 32 38 33 34 33   ~250046006028343	
    38 33 31 33 30 33 30 32 44 33 31 33 36 33 30 33   83130302D3136303	
    32 32 44 33 32 33 30 33 31 33 37 33 30 33 37 33   22D3230313730373	
    30 33 30 33 31 46 35 39 35 0d                     03031F595.      	
[06/02/2018 11:51:36] Written data (COM3)	
    7e 32 35 30 31 34 36 34 32 45 30 30 32 30 31 46   ~25014642E00201F	
    44 33 30 0d                                       D30.            	
[06/02/2018 11:51:36] Read data (COM3)	
    7e 32 35 30 30 34 36 30 30 46 30 37 41 30 30 30   ~25004600F07A000	
    31 31 30 30 43 44 34 30 43 44 34 30 43 44 34 30   1100CD40CD40CD40	
    43 44 34 30 43 44 35 30 43 44 34 30 43 44 35 30   CD40CD50CD40CD50	
    43 44 34 30 43 44 34 30 43 44 34 30 43 44 34 30   CD40CD40CD40CD40	
    43 44 34 30 43 44 34 30 43 44 35 30 43 44 35 30   CD40CD40CD50CD50	
    43 44 34 30 36 30 42 38 32 30 42 38 36 30 42 38   CD4060B820B860B8	
    34 30 42 37 44 30 42 39 42 30 42 43 36 30 30 30   40B7D0B9B0BC6000	
    30 43 44 34 34 31 33 41 30 30 33 32 37 31 30 30   0CD4413A00327100	
    30 30 30 32 37 31 30 45 32 41 42 0d               0002710E2AB.    	
[06/02/2018 11:51:37] Written data (COM3)	
    7e 32 35 30 31 34 36 34 34 45 30 30 32 30 31 46   ~25014644E00201F	
    44 32 45 0d                                       D2E.            	
[06/02/2018 11:51:37] Read data (COM3)	
    7e 32 35 30 30 34 36 30 30 30 30 34 43 30 30 30   ~25004600004C000	
    31 31 30 30 30 30 30 30 30 30 30 30 30 30 30 30   1100000000000000	
    30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000	
    30 30 30 30 36 30 30 30 30 30 30 30 30 30 30 30   0000600000000000	
    30 30 30 30 30 30 30 30 30 30 30 30 36 30 30 30   0000000000006000	
    30 30 30 30 30 30 30 30 30 45 46 34 41 0d         000000000EF4A.  	
[06/02/2018 11:51:42] Written data (COM3)	
    7e 32 35 30 31 34 36 43 31 30 30 30 30 46 44 39   ~250146C10000FD9	
    41 0d                                             A.              	
[06/02/2018 11:51:42] Read data (COM3)	
    7e 32 35 30 30 34 36 30 30 36 30 32 38 35 30 33   ~250046006028503	
    31 33 36 35 33 33 35 33 30 34 31 32 44 33 37 33   136533530412D373	
    33 33 31 33 36 32 44 33 31 32 45 33 31 33 30 32   331362D312E31302	
    30 32 30 32 30 46 35 38 34 0d                     02020F584.      	
[06/02/2018 11:51:42] Written data (COM3)	
    7e 32 35 30 31 34 36 43 32 30 30 30 30 46 44 39   ~250146C20000FD9	
    39 0d                                             9.              	
[06/02/2018 11:51:42] Read data (COM3)	
    7e 32 35 30 30 34 36 30 30 36 30 32 38 33 34 33   ~250046006028343	
    38 33 31 33 30 33 30 32 44 33 31 33 36 33 30 33   83130302D3136303	
    32 32 44 33 32 33 30 33 31 33 37 33 30 33 37 33   22D3230313730373	
    30 33 30 33 31 46 35 39 35 0d                     03031F595.      	
[06/02/2018 11:51:42] Written data (COM3)	
    7e 32 35 30 31 34 36 34 32 45 30 30 32 30 31 46   ~25014642E00201F	
    44 33 30 0d                                       D30.            	
[... 39 lines cut ...]
[06/02/2018 11:51:48] Read data (COM3)	
    7e 32 35 30 30 34 36 30 30 46 30 37 41 30 30 30   ~25004600F07A000	
    31 31 30 30 43 44 34 30 43 44 34 30 43 44 34 30   1100CD40CD40CD40	
    43 44 34 30 43 44 35 30 43 44 34 30 43 44 35 30   CD40CD50CD40CD50	
    43 44 34 30 43 44 34 30 43 44 34 30 43 44 34 30   CD40CD40CD40CD40	
    43 44 34 30 43 44 34 30 43 44 35 30 43 44 35 30   CD40CD40CD50CD50	
    43 44 34 30 36 30 42 38 31 30 42 38 36 30 42 38   CD4060B810B860B8	
    34 30 42 37 44 30 42 39 42 30 42 43 36 30 30 30   40B7D0B9B0BC6000	
    30 43 44 34 34 31 33 41 30 30 33 32 37 31 30 30   0CD4413A00327100	
    30 30 30 32 37 31 30 45 32 41 43 0d               0002710E2AC.    	
[06/02/2018 11:51:48] Written data (COM3)	
    7e 32 35 30 31 34 36 34 34 45 30 30 32 30 31 46   ~25014644E00201F	
    44 32 45 0d                                       D2E.            	
[06/02/2018 11:51:48] Read data (COM3)	
    7e 32 35 30 30 34 36 30 30 30 30 34 43 30 30 30   ~25004600004C000	
    31 31 30 30 30 30 30 30 30 30 30 30 30 30 30 30   1100000000000000	
    30 30 30 30 30 30 30 30 30 30 30 30 30 30 30 30   0000000000000000	
    30 30 30 30 36 30 30 30 30 30 30 30 30 30 30 30   0000600000000000	
    30 30 30 30 30 30 30 30 30 30 30 30 36 30 30 30   0000000000006000	
    30 30 30 30 30 30 30 30 30 45 46 34 41 0d         000000000EF4A.  	
```

So based on the top I could not compare what they are sedning , any idea ?

Thanks

biancode, 2018-02-06:
That looks like ASCII Modbus ...

biancode, 2018-02-06:
26 04 10 00 00 09 32 1b  --->  UnitID 26 FC4 address 1000 quantity 9 ?

shahbour, 2018-02-07:
Yes basically this is what I am doing, but for unit id, I am using 38 that is converted to 0x26

biancode, 2018-02-07:
Yes sure - it is hex! Then it should work. You could test with [the necessary lib](https://www.npmjs.com/package/modbus-serial), how it works on direct access?

biancode, 2018-02-07:
You could test the ASCII mode with the node. Did you test before?

shahbour, 2018-02-07:
I will test that today when I reach office

yaacov, 2018-02-07:
Hi, all
The battary software message is not pure modbus syntax, but does look like ascii modbus:

~ | delimiter (in ascii modbus the delimiter is `:` )
--| ---
25 | UnitID
01 | Read Coil Status (FC=01)
46C2 | The Data Address of the first coil to read. 
0000 | The total number of coils requested. 
FD99 | should be LRC

My 2 cents are:
1. the software use some proprietary dialect that is not pure modbus.
2. try to use ascii  modbus.
3. try also UnitID 25, it seems to work for the dialect the battery software use :-)

biancode, 2018-02-07:
please reopen if there is a problem ...

closed 2018-02-07 by biancode