BiancoRoyal/node-red-contrib-modbus #81: Modbus core / serial times out after a few hours of running

3phreal, 2018-02-06:
Hi I've just started using Node-red because I'm trying to make an interface to my Ventilation system which is controlled by a CTS602 I do not know if this is specific to my device alone or is a common controller. Anyways I've manage to create access to it both from a Python app and node-red-contrib-modubs. And it runs find for some time (few hours in some cases a whole day in others) before it just keeps timing out. 

I've tried debugging the modbus interface using the log and i keep getting the following. 

```
v/ttyUSB0:19200bit/s default Unit-Id: 30 +2ms
  contribModbus:flex:getter {"topic":"Humidity","payload":{"value":"Humidity","unitid":30,"fc":4,"address":221,"quantity":1,"messageId":"5a797dad7b998c7c6fa30fd9"},"_msgid":"a2a8ad5a.27718","queueNumber":0,"queueUnit":30} +4ms
  contribModbus:core:client Client -> {"state":"QUEUEING","message":"dequeue command serial","delay":50} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +49ms
  contribModbus:core:client Client -> {"type":"queue check","unitid":30,"sendAllowed":true,"queueLength":1} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +1ms
  contribModbus:core:client Client -> {"type":"serial sending and wait","unitid":30,"queueLength":0,"sendAllowedForNext":false,"delay":50} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +0ms
  contribModbus:core:client Client -> {"info":"read msg","message":{"value":"Humidity","unitid":30,"fc":4,"address":221,"quantity":1,"messageId":"5a797dad7b998c7c6fa30fd9"},"unitid":30,"timeout":1000,"state":"QUEUEING"} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +1ms
  modbus-serial { action: 'send serial rtu buffered',
  modbus-serial   data: <Buffer 1e 04 00 dd 00 01 a3 9f>,
  modbus-serial   unitid: 30,
  modbus-serial   functionCode: 4 } +1s
  contribModbus:core:client Client -> {"info":"queue response activate sending","message":{"value":"Humidity","unitid":30,"fc":4,"address":221,"quantity":1,"messageId":"5a797dad7b998c7c6fa30fd9"}} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +1s
  contribModbus:core:client Client -> event: activate old: QUEUEING new: ACTIVATED Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +0ms
  contribModbus:core:client Client -> event: queue old: ACTIVATED new: QUEUEING Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +2ms
  contribModbus:flex:getter Timed out +1s
  modbus-serial modbusErrorHandling:Timed out +1s
  contribModbus:core:client Client -> {"state":"QUEUEING","message":"dequeue command serial","delay":50} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +53ms
  contribModbus:core:client Client -> {"type":"queue check is not a unit","unitid":null} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +1ms
  contribModbus:flex:getter Add Message 5a797db07b998c7c6fa30fda +1s
  contribModbus:core:client Client -> {"info":"push to Queue by Unit-Id","message":{"value":"Temperatur","unitid":30,"fc":4,"address":200,"quantity":16,"messageId":"5a797db07b998c7c6fa30fda"},"unit":30} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +1s
  contribModbus:core:client Client -> event: queue old: EMPTY new: QUEUEING Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +1ms
  contribModbus:core:client Client -> {"info":"queue read msg","message":{"value":"Temperatur","unitid":30,"fc":4,"address":200,"quantity":16,"messageId":"5a797db07b998c7c6fa30fda"},"state":"EMPTY","queueLength":1} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +1ms
  contribModbus:flex:getter {"topic":"Temp","payload":{"value":"Temperatur","unitid":30,"fc":4,"address":200,"quantity":16,"messageId":"5a797db07b998c7c6fa30fda"},"_msgid":"d47eb796.928308","queueNumber":0,"queueUnit":30} +4ms
  contribModbus:flex:getter Add Message 5a797db07b998c7c6fa30fdb +3ms
  contribModbus:core:client Client -> {"info":"push to Queue by Unit-Id","message":{"value":1517911472120,"unitid":30,"fc":4,"address":1100,"quantity":5,"messageId":"5a797db07b998c7c6fa30fdb"},"unit":30} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +4ms
  contribModbus:core:client Client -> {"info":"queue read msg","message":{"value":1517911472120,"unitid":30,"fc":4,"address":1100,"quantity":5,"messageId":"5a797db07b998c7c6fa30fdb"},"state":"QUEUEING","queueLength":2} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +1ms
  contribModbus:flex:getter {"topic":"Air-Flow-Vent-ctrl","payload":{"value":1517911472120,"unitid":30,"fc":4,"address":1100,"quantity":5,"messageId":"5a797db07b998c7c6fa30fdb"},"_msgid":"3b9d9344.3a865c","queueNumber":1,"queueUnit":30} +2ms
  contribModbus:core:client Client -> {"state":"QUEUEING","message":"dequeue command serial","delay":50} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +44ms
  contribModbus:core:client Client -> {"type":"queue check","unitid":30,"sendAllowed":true,"queueLength":2} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +0ms
  contribModbus:core:client Client -> {"type":"serial sending and wait","unitid":30,"queueLength":1,"sendAllowedForNext":false,"delay":50} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +0ms
  contribModbus:core:client Client -> {"info":"read msg","message":{"value":"Temperatur","unitid":30,"fc":4,"address":200,"quantity":16,"messageId":"5a797db07b998c7c6fa30fda"},"unitid":30,"timeout":1000,"state":"QUEUEING"} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +1ms
  modbus-serial { action: 'send serial rtu buffered',
  modbus-serial   data: <Buffer 1e 04 00 c8 00 10 72 57>,
  modbus-serial   unitid: 30,
  modbus-serial   functionCode: 4 } +3s
  contribModbus:core:client Client -> {"info":"queue response activate sending","message":{"value":"Temperatur","unitid":30,"fc":4,"address":200,"quantity":16,"messageId":"5a797db07b998c7c6fa30fda"}} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +1s
  contribModbus:core:client Client -> event: activate old: QUEUEING new: ACTIVATED Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +0ms
  contribModbus:core:client Client -> event: queue old: ACTIVATED new: QUEUEING Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +2ms
  contribModbus:flex:getter Timed out +1s
  modbus-serial modbusErrorHandling:Timed out +3s
  contribModbus:core:client Client -> {"state":"QUEUEING","message":"dequeue command serial","delay":50} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +52ms
  contribModbus:core:client Client -> {"type":"queue check","unitid":30,"sendAllowed":true,"queueLength":1} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +1ms
  contribModbus:core:client Client -> {"type":"serial sending and wait","unitid":30,"queueLength":0,"sendAllowedForNext":false,"delay":50} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +0ms
  contribModbus:core:client Client -> {"info":"read msg","message":{"value":1517911472120,"unitid":30,"fc":4,"address":1100,"quantity":5,"messageId":"5a797db07b998c7c6fa30fdb"},"unitid":30,"timeout":1000,"state":"QUEUEING"} Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30 +0ms
  modbus-serial { action: 'send serial rtu buffered',
  modbus-serial   data: <Buffer 1e 04 04 4c 00 05 f2 81>,
  modbus-serial   unitid: 30,
  modbus-serial   functionCode: 4 } +1s
```
It starts working again if i restart node-red or redeploy my code which is shown below. please not its still a work in progress. The main idea was to use a signal Flex Getter to handle all read actions. 

`[{"id":"a23ad08.b70e23","type":"tab","label":"Modbus","disabled":false,"info":""},{"id":"fd0230b9.28ce2","type":"tab","label":"Dashboard","disabled":false,"info":""},{"id":"7d0fe24a.6ec4ec","type":"tab","label":"Statistics","disabled":false,"info":""},{"id":"cc1b3ba4.956328","type":"tab","label":"Control","disabled":false,"info":""},{"id":"c8ebdfe6.aaeae","type":"ui_base","theme":{"name":"theme-dark","lightTheme":{"default":"#0094CE","baseColor":"#0094CE","baseFont":"-apple-system,BlinkMacSystemFont,Segoe UI,Roboto,Oxygen-Sans,Ubuntu,Cantarell,Helvetica Neue,sans-serif","edited":true,"reset":false},"darkTheme":{"default":"#097479","baseColor":"#097479","baseFont":"-apple-system,BlinkMacSystemFont,Segoe UI,Roboto,Oxygen-Sans,Ubuntu,Cantarell,Helvetica Neue,sans-serif","edited":true,"reset":false},"customTheme":{"name":"Untitled Theme 1","default":"#4B7930","baseColor":"#4B7930","baseFont":"-apple-system,BlinkMacSystemFont,Segoe UI,Roboto,Oxygen-Sans,Ubuntu,Cantarell,Helvetica Neue,sans-serif","reset":false},"themeState":{"base-color":{"default":"#097479","value":"#097479","edited":false},"page-titlebar-backgroundColor":{"value":"#097479","edited":false},"page-backgroundColor":{"value": [... cut ...] nd:\"#C0DEED\"};\n        break;\n    case 3:\n        b3 = {color:\"#000000\",background:\"#C0DEED\"};\n        break;\n    case 4:\n        b4 = {color:\"#000000\",background:\"#C0DEED\"};\n        break;\n}\nreturn [b1, b2, b3, b4, intake, outlet];","outputs":"6","noerr":0,"x":398,"y":225,"wires":[["a89772b3.0e156"],["d3212280.7be72"],["56ab50ae.e8e57"],["b76e0276.33f98"],["7ac27291.cd6dcc"],["4517cb87.e9b034"]],"outputLabels":["Button_1","Button_2","Button_3","Button_4","Intake","Outlet"]}]`

biancode, 2018-02-06:
Hi, 

a) some Modbus devices need to ask in a minimum time cycle. Otherwise, some reads have timeouts.

b) other devices sometimes have timeouts, but it is no problem because next read works well again 

c) you are too high with the baud rate or to fast in reading cycle

What tells the Manual of the device about timeouts and polling?

cya
Klaus

3phreal, 2018-02-06:
Hi klaus

My only source is this document
https://reader.livedition.dk/nilan/272/html5/

Regards
Anders

biancode, 2018-02-06:
OK, as I see it is all correct. Does the communication stop on timeout? What if you set the timeout to 5000 (5sec)?

3phreal, 2018-02-06:
I haven't tried a larger timeout because.
1. I have kinda made a stress test activating all my calls multiple times and it does not make it fail. 
2. Why can't it recover, if all I need is to redeploy or restart node-red. Then it looks like it's stuck in an unrecovable state. 

When I use my above system, does the modbus core wait for response before sending the next command.? 

Regards
Anders 

Regards

3phreal, 2018-02-07:
Hi 

I have now set the time out to 5000ms, additionally I've set the Queue delay to 500 ms. Still only runs for a few hours before locking up. 

Regards

biancode, 2018-02-07:
Set the queue  back to 1ms

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

3phreal, 2018-02-07:
Hi 

I've set the Queue delay time to 1 ms

I have the log from when it goes awray. though i cannot see whats wrong. its at line 9246. with a delay time of 633ms from there it stops working.

[modbus_screen_dump.log](https://github.com/biancode/node-red-contrib-modbus/files/1702987/modbus_screen_dump.log)

regards

biancode, 2018-02-07:
... as I see it stopped to send the response to requests but did not close the connection, in that case, it should accept new requests without any reconnect otherwise, it has to disconnect by Modbus error

3phreal, 2018-02-07:
Hi 

I've noticed something odd. When I start node-red either after reset or redeploy. I get two error messeges from the Modbus-core. 

`Client -> try to reconnect by init in 2000 ms Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30`
and
`Client -> reconnect in 2000 ms Serial@/dev/ttyUSB0:19200bit/s default Unit-Id: 30`

and after that it works. 

However when the timeout occurs and it locks up. I never get any errors or debug printouts. it seems odd to me. 

I can see from the consol log that it keeps getting timeout error
`modbus-serial modbusErrorHandling:Timed out +3s`

regards

biancode, 2018-02-07:
with the redeploy it close the serial and reconnect - a chance is to connect directly via [modbus-serial](https://www.npmjs.com/package/modbus-serial) to see if there some errors in communication, which are not handled or testing with other Modbus client ...

3phreal, 2018-02-07:
How are you handling Modbus exceptions ?

biancode, 2018-02-07:
You will see the exception if you get one - did you see some?

biancode, 2018-02-07:
[image]

biancode, 2018-02-07:
not "working" means it has to reconnect

3phreal, 2018-02-07:
I do not get any errors from the modbus after I have restarted.

3phreal, 2018-02-07:
Except the 2 at the beginning

3phreal, 2018-02-07:
The slave can send modbus exceptions as response . Such wait, not ready yet ect. Do you handle thoes or are they user specific. Some of the exception responses requires you to stop sending data.

3phreal, 2018-02-07:
But it is still active.

biancode, 2018-02-07:
Yes, an exception is handled, but there is no exception in your case ...

biancode, 2018-02-07:
you could try the new v3.2.2 too

3phreal, 2018-02-07:
I get the same problem with that version. How do I debug the serial-modbus code?

biancode, 2018-02-07:
debug is with `modbus-serial*` tag - but did you test just with the [modbus-serial](https://www.npmjs.com/package/modbus-serial) lib?

3phreal, 2018-02-07:
Well I'm new to node-red and i do not have that much knowledge about JS and such. and it dosn't looks like i can load the modbus-serial inside node-red ?

biancode, 2018-02-07:
Ah, sorry  no I mean please test it on command line and use the[ examples of the modbus-serial lib](https://github.com/yaacov/node-modbus-serial/tree/master/examples)

biancode, 2018-02-07:
`node logger.js`   and you have toi set the parameter in logger.js to reach your device

biancode, 2018-02-07:
please reopen if you find the problem ...

closed 2018-02-07 by biancode