BiancoRoyal/node-red-contrib-modbus #82: Issue on build

rjbaat, 2018-02-19:
Hi,
I have an issue when building the node for Resin.io on a rpi.

This is the error that occurred:

`[34m[Build]   [39m make: Entering directory '/usr/local/lib/node_modules/node-red-contrib-modbus/node_modules/node-modbus/node_modules/serialport/build'
[34m[Build]   [39m   CXX(target) Release/obj.target/serialport/src/serialport.o
[34m[Build]   [39m   CXX(target) Release/obj.target/serialport/src/serialport_unix.o
[34m[Build]   [39m   CXX(target) Release/obj.target/serialport/src/poller.o
[34m[Build]   [39m   SOLINK_MODULE(target) Release/obj.target/serialport.node
[34m[Build]   [39m   COPY Release/serialport.node
[34m[Build]   [39m make: Leaving directory '/usr/local/lib/node_modules/node-red-contrib-modbus/node_modules/node-modbus/node_modules/serialport/build'
[34m[Build]   [39m [91mprebuild-install[0m[91m [0m[91minfo[0m[91m [0m[91mbegin Prebuild-install version 2.5.1
[34m[Build]   [39m [0m[91mprebuild-install [0m[91minfo [0m[91mlooking for local prebuild @[0m[91m prebuilds/serialport-v6.1.0-node-v59-linuxmusl-arm.tar.gz
[34m[Build]   [39m [0m[91mprebuild-install [0m[91minfo[0m[91m looking for cached prebuild @[0m[91m /root/.npm/_prebuilds/https-github.com-node-serialport-node-serialport-releases-download-v6.1.0-serialport-v6.1.0-node-v59-linuxmusl-arm.tar.gz
[34m[Build]   [39m [0m[91mprebuild-install [0m[91mhttp request[0m[91m GET https://github.com/node-serialport/node-serialport/releases/download/v6.1.0/serialport-v6.1.0-node-v59-linuxmusl-arm.tar.gz
[34m[Build]   [39m [0m[91mprebuild-install [0m[91mhttp[0m[91m 404[0m[91m https://github.com/node-serialport/node-serialport/releases/download/v6.1.0/serialport-v6.1.0-node-v59-linuxmusl-arm.tar.gz
[34m[Build]   [39m [0m[91mprebuild-install WARN[0m[91m install[0m[91m No prebuilt binaries found (target=9.4.0 runtime=node arch=arm platform=linux)
[34m[Build]   [39m [0mmake: Entering directory '/usr/local/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build'
[34m[Build]   [39m   CXX(target) Release/obj.target/serialport/src/serialport.o
[34m[Build]   [39m   CXX(target) Release/obj.target/serialport/src/serialport_unix.o
[34m[Build]   [39m   CXX(target) Release/obj.target/serialport/src/poller.o
[34m[Build]   [39m   CXX(target) Release/obj.target/serialport/src/serialport_linux.o
[34m[Build]   [39m [91m../src/serialport_linux.cpp: In function 'int linuxSetCustomBaudRate(int, unsigned int)':
[34m[Build]   [39m ../src/serialport_linux.cpp:10:18: error: 'TCGETS2' was not declared in this scope
[34m[Build]   [39m      if(ioctl(fd, TCGETS2, &t)) {
[34m[Build]   [39m                   ^~~~~~~
[34m[Build]   [39m [0m[91m../src/serialport_linux.cpp:18:18: error: 'TCSETS2' was not declared in this scope
[34m[Build]   [39m      if(ioctl(fd, TCSETS2, &t)) {
[34m[Build]   [39m                   ^~~~~~~
[34m[Build]   [39m [0m[91m../src/serialport_linux.cpp: In function 'int linuxGetSystemBaudRate(int, int*)':
[34m[Build]   [39m ../src/serialport_linux.cpp:29:16: error: 'TCGETS2' was not declared in this scope
[34m[Build]   [39m    if(ioctl(fd, TCGETS2, &t)) {
[34m[Build]   [39m                 ^~~~~~~
[34m[Build]   [39m [0m[91mmake: *** [serialport.target.mk:96: Release/obj.target/serialport/src/serialport_linux.o] Error 1
[34m[Build]   [39m [0m[91mmake: *** Waiting for unfinished jobs....
[34m[Build]   [39m [0mmake: Leaving directory '/usr/local/lib/node_modules/node-red-contrib-modbus/node_modules/serialport/build'`

Is there something i need to adjust?
It has worked perfectly before.

rjbaat, 2018-02-19:
The previous version of serialport: 6.0.5 will install correct with the command: npm install serialport@6.0.5.
But then it will try to install the latest version as well when installing: npm install node-red-contrib-modbus.

biancode, 2018-02-20:
I am sorry, but it looks like the error comes from v6.1.0 ... please, report that issue to the serialport team!

rjbaat, 2018-02-21:
I have done this. And it looks like 6.1.0 has an issue indeed, what would be fixed in a new version. But 6.1.0 is the prerelease version. Isnt it better to include the stable version as dependency insted? Because this way the last release is always depending on the pre release. This is not really reliable when using it in production. And since docker is building it based on a dockerfile its a bit difficult to get a custom dependency up. Resin builds the img from a docker template so it will allways build a prerelease build insted of a stable build when you just include node red modbus.

biancode, 2018-02-21:
Yes, you are right.

rjbaat, 2018-02-22:
I tried to install it again. But it seams to require 6.1.0 again insted of 6.0.5. Is there a way to force to use the 6.0.5 version?
Also if i pull the node on OSX where it is never installed or cached it will get the 6.1.0 version.

biancode, 2018-02-22:
Yes, the [modbus-serial ](https://github.com/yaacov/node-modbus-serial/blob/master/package.json) did not set a fix library here. @yaacov maybe we should just provide stable versions of serialport?

rjbaat, 2018-02-22:
Well the node-serial has a fixed version of: 6.0.4.
https://github.com/yaacov/node-modbus-serial/blob/a9118c8859ff3b29d9d0bb0bbd83744a325594e0/package.json#L41

But the node-modus has version 6.1.0
https://github.com/biancode/node-modbus/blob/c293d8c42982a74343c859198146de4f06aa7f09/package.json#L21

biancode, 2018-02-22:
Yes, you are right - I was in the wrong window ...

rjbaat, 2018-02-22:
I see there is been an update, thanks for that. But there is one little thing to make it all work again. 
The problem i run into now is that there is a dependency to node-modbus of the old version 4.2.1 insted of 4.3.0

https://github.com/biancode/node-red-contrib-modbus/blob/56f2e576281bd30ed01a9ca5a40605335423fa0f/package.json#L11

This means when i install 3.3.2 of node-red-contrib-modbus the dependency of 6.1.0 of serialport still exist :(. 
```

node-red-contrib-modbus@3.3.2 
  ├── bson@1.0.4 
  ├─┬ debug@3.1.0 
  │ └── ms@2.0.0 
  ├── line-by-line@0.1.5 
  ├── modbus-serial@7.1.5 
  ├─┬ node-modbus@4.2.1 
  │ ├── bluebird@3.5.1 
  │ ├── crc@3.5.0 
  │ ├── net@1.0.2 
  │ ├── serialport@6.1.0 
  │ ├── stampit-event-bus@0.1.1 
  │ ├── stampit-log@0.3.0 
  │ └── stampit-state-machine@0.2.1 
  ├─┬ serialport@6.0.5 
  │ ├── bindings@1.3.0 
  │ ├── commander@2.14.1 
  │ ├── nan@2.8.0 
  │ ├─┬ prebuild-install@2.5.1 
  │ │ ├── detect-libc@1.0.3 
  │ │ ├── expand-template@1.1.0 
  │ │ ├── github-from-package@0.0.0 
  │ │ ├── minimist@1.2.0 
..

```

biancode, 2018-02-22:
Yes, sure! ;)
step by step ...

rjbaat, 2018-02-22:
Thnx! It works again :)

closed 2018-02-22 by biancode

rjbaat, 2018-08-03:
I think i run into the same problem again. I can't build it since node serialport is giving me build errors again. 

```
34m[main][39m  Executing busybox-1.28.4-r0.trigger
[34m[main][39m  OK: 205 MiB in 83 packages
[34m[main][39m  make: Entering directory '/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/build'
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport_unix.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/poller.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport_linux.o
[34m[main][39m  [91m../src/serialport_linux.cpp:4:24: fatal error: asm/ioctls.h: No such file or directory
[34m[main][39m   #include <asm/ioctls.h>
[34m[main][39m                          ^
[34m[main][39m  compilation terminated.
[34m[main][39m  [0m
[34m[main][39m  [91mmake: *** [serialport.target.mk:96: Release/obj.target/serialport/src/serialport_linux.o] Error 1
[34m[main][39m  make: *** Waiting for unfinished jobs....
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport_unix.cpp:24:26: fatal error: linux/serial.h: No such file or directory
[34m[main][39m   #include <linux/serial.h>
[34m[main][39m                            ^
[34m[main][39m  compilation terminated.
[34m[main][39m  [0m
[34m[main][39m  [91mmake: *** [serialport.target.mk:96: Release/obj.target/serialport/src/serialport_unix.o] Error 1
[34m[main][39m  [0m
[34m[main][39m  [91m../src/poller.cpp: In static member function 'static void Poller::onData(uv_poll_t*, int, int)':
[34m[main][39m  ../src/poller.cpp:69:29: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     obj->callback.Call(2, argv);
[34m[main][39m                               ^
[34m[main][39m  In file included from ../src/poller.cpp:1:0:
[34m[main][39m  ../../nan/nan.h:1618:3: note: declared here
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport.cpp: In function 'void EIO_AfterOpen(uv_work_t*)':
[34m[main][39m  ../src/serialport.cpp:95:30: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     data->callback.Call(2, argv);
[34m[main][39m                                ^
[34m[main][39m  In file included from ../src/./serialport.h:6:0,
[34m[main][39m                   from ../src/serialport.cpp:1:
[34m[main][39m  ../../nan/nan.h:1618:3: note: declared here
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport.cpp: In function 'void EIO_AfterUpdate(uv_work_t*)':
[34m[main][39m  ../src/serialport.cpp:150:30: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     data->callback.Call(1, argv);
[34m[main][39m                                ^
[34m[main][39m  In file included from ../src/./serialport.h:6:0,
[34m[main][39m                   from ../src/serialport.cpp:1:
[34m[main][39m  ../../nan/nan.h:1618:3: note: declared here
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport.cpp: In function 'void EIO_AfterClose(uv_work_t*)':
[34m[main][39m  ../src/serialport.cpp:188:30: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     data->callback.Call(1, argv);
[34m[main][39m                                ^
[34m[main][39m  In file included from ../src/./serialport.h:6:0,
[34m[main][39m                   from ../src/serialport.cpp:1:
[34m[main][39m  ../../nan/nan.h:1618:3: note: declared here
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport.cpp: In function 'void EIO_AfterFlush(uv_work_t*)':
[34m[main][39m  ../src/serialport.cpp:231:30: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     data->callback.Call(1, argv);
[34m[main][39m                                ^
[34m[main][39m  In file included from ../src/./serialport.h:6:0,
[34m[main][39m                   from ../src/serialport.cpp:1:
[... 24 lines cut ...]
[34m[main][39m  [91m../src/serialport.cpp: In function 'void EIO_AfterGetBaudRate(uv_work_t*)':
[34m[main][39m  ../src/serialport.cpp:383:30: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     data->callback.Call(2, argv);
[34m[main][39m                                ^
[34m[main][39m  In file included from ../src/./serialport.h:6:0,
[34m[main][39m                   from ../src/serialport.cpp:1:
[34m[main][39m  ../../nan/nan.h:1618:3: note: declared here
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[... cut ...]
[34m[main][39m  In file included from ../src/./serialport.h:6:0,
[34m[main][39m                   from ../src/serialport.cpp:1:
[34m[main][39m  ../../nan/nan.h:1618:3: note: declared here
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  make: Leaving directory '/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/build'
[34m[main][39m  Removing intermediate container 8c610b5127d4
```

Do you have an idea how to fix it?

rjbaat, 2018-08-03:
What i see is in the package.json that 6.1.1 (pre-release version) is used. Might that be the problem?
[https://github.com/biancode/node-red-contrib-modbus/blob/ac65a11b15a50728b90474bc48726a036008e716/package.json#L12](url)

biancode, 2018-08-03:
You have to use the correct nodejs version as expected from serialport. Is it the right one? I’m in Wacken time W:O:A ;)

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

rjbaat, 2018-08-03:
Well i just tried to build it with this latest Alpine docker base img:

https://github.com/resin-io-library/base-images/blob/5159250f897aa4e88c701d4c27dac62645bafcaa/node/raspberry-pi/alpine/0.10/Dockerfile

It seams to have nodejs version: 0.10.48

But i get the same results. 

```
[34m[main][39m  Step 1/9 : FROM resin/raspberrypi3-alpine-node:0.10.48-slim
[34m[main][39m   ---> 46a9c2b36740
[34m[main][39m  Step 2/9 : RUN mkdir -p /usr/src/app/
[34m[main][39m   ---> Running in fb26500e3558
[34m[main][39m   ---> ceb6eceab279
[34m[main][39m  Removing intermediate container fb26500e3558
[34m[main][39m  Step 3/9 : WORKDIR /usr/src/app
[34m[main][39m   ---> 2a0412617ae2
[34m[main][39m  Removing intermediate container fe8976e19575
[34m[main][39m  Step 4/9 : COPY ./app ./
[34m[main][39m   ---> 3de0661066b8
[34m[main][39m  Removing intermediate container be9cde7470b2
[34m[main][39m  Step 5/9 : COPY entry.sh /usr/bin/entry.sh
[34m[main][39m   ---> 4e53fa4dbf33
[34m[main][39m  Removing intermediate container ae3d549d9826
[34m[main][39m  Step 6/9 : COPY udev-rules/ /etc/udev/rules.d/
[34m[main][39m   ---> 6f40895ec386
[34m[main][39m  Removing intermediate container ba14080f9d1a
[34m[main][39m  Step 7/9 : RUN apk add --no-cache make gcc g++ python &&   JOBS=MAX npm install -g node-red-node-serialport --production --silent &&   JOBS=MAX npm install -g node-red-contrib-modbus --production --silent &&   apk del make gcc g++ python &&   npm cache clean --force && rm -rf /tmp/*
[34m[main][39m   ---> Running in 47854f1ebb21
[34m[main][39m  fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/main/armhf/APKINDEX.tar.gz
[34m[main][39m  fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/community/armhf/APKINDEX.tar.gz
[34m[main][39m  (1/13) Installing binutils (2.30-r5)
[34m[main][39m  (2/13) Installing isl (0.18-r0)
[34m[main][39m  (3/13) Installing libgomp (6.4.0-r8)
[34m[main][39m  (4/13) Installing libatomic (6.4.0-r8)
[34m[main][39m  (5/13) Installing mpfr3 (3.1.5-r1)
[34m[main][39m  (6/13) Installing mpc1 (1.0.3-r1)
[34m[main][39m  (7/13) Installing gcc (6.4.0-r8)
[34m[main][39m  (8/13) Installing musl-dev (1.1.19-r10)
[34m[main][39m  (9/13) Installing libc-dev (0.7.1-r0)
[34m[main][39m  (10/13) Installing g++ (6.4.0-r8)
[34m[main][39m  (11/13) Installing make (4.2.1-r2)
[34m[main][39m  (12/13) Installing gdbm (1.13-r1)
[34m[main][39m  (13/13) Installing python2 (2.7.15-r0)
[34m[main][39m  Executing busybox-1.28.4-r0.trigger
[34m[main][39m  OK: 205 MiB in 83 packages
[34m[main][39m  [91m
[34m[main][39m  [0m
[34m[main][39m  [91m/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/node_modules/prebuild-install/node_modules/which-pm-runs/index.js:11
[34m[main][39m  [0m
[34m[main][39m  [91m  const pmSpec = userAgent.split(' ')[0]
[34m[main][39m    ^^^^^
[34m[main][39m  [0m
[34m[main][39m  [91mSyntaxError: Use of const in strict mode.
[34m[main][39m      at Module._compile (module.js:439:25)
[34m[main][39m      at Object.Module._extensions..js (module.js:474:10)
[34m[main][39m      at Module.load (module.js:356:32)
[34m[main][39m      at Function.Module._load (module.js:312:12)
[34m[main][39m      at Module.require (module.js:364:17)
[34m[main][39m      at require (module.js:380:17)
[34m[main][39m      at Object.<anonymous> (/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/node_modules/prebuild-install/bin.js:5:19)
[34m[main][39m      at Module._compile (module.js:456:26)
[34m[main][39m      at Object.Module._extensions..js (module.js:474:10)
[34m[main][39m      at Module.load (module.js:356:32)
[34m[main][39m  [0m
[34m[main][39m  make: Entering directory '/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/build'
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport_unix.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/poller.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport_linux.o
[... 111 lines cut ...]
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport.cpp: In function 'void EIO_AfterDrain(uv_work_t*)':
[34m[main][39m  ../src/serialport.cpp:424:30: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     data->callback.Call(1, argv);
[34m[main][39m                                ^
[34m[main][39m  In file included from ../src/./serialport.h:6:0,
[34m[main][39m                   from ../src/serialport.cpp:1:
[34m[main][39m  ../node_modules/nan/nan.h:1618:3: note: declared here
[34m[main][39m     Call(int argc, v8::Local<v8::Value> argv[]) const {
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91mmake: *** [serialport.target.mk:91: Release/obj.target/serialport/src/poller.o] Error 1
[34m[main][39m  [0m
[34m[main][39m  make: Leaving directory '/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/build'
[34m[main][39m  Removing intermediate container 47854f1ebb21
[34m[main][39m  [31mThe command '/bin/sh -c apk add --no-cache make gcc g++ python &&   JOBS=MAX npm install -g node-red-node-serialport --production --silent &&   JOBS=MAX npm install -g node-red-contrib-modbus --production --si [... cut ...]
[================>                                  ] 33%[2K[G[32m[Success][39m  Successfully uploaded images
[31m[Error][39m  Some services failed to build:
[31m[Error][39m    Service: main
[31m[Error][39m      Error: The command '/bin/sh -c apk add --no-cache make gcc g++ python &&   JOBS=MAX npm install -g node-red-node-serialport --production --silent &&   JOBS=MAX npm install -g node-red-contrib-modbus --production --silent &&   apk del make gcc g++ python &&   npm cache clean --force && rm -rf /tmp/*' returned a non-zero code: 1
[31m[Error][39m  Not deploying release.

```

yaacov, 2018-08-03:
@rjbaat hi, adding links to node-serialport docs about alpine and rpi.

https://github.com/node-serialport/node-serialport/tree/master/packages/serialport#alpine-linux
https://github.com/node-serialport/node-serialport/tree/master/packages/serialport#raspberry-pi-linux

rjbaat, 2018-08-03:
hi @yaacov thnx! I have looked into it and checked if it would build with: 
--unsafe-perm --build-from-source but so far the same result :(.

Not sure what i can do to make it work again..

rjbaat, 2018-08-03:
@yaacov I added the extra libraries from the aline link you added. Now it looks like the fatal error is gone, but the build is still failing. 

```
[34m[main][39m  Step 7/9 : RUN apk add --no-cache make gcc g++ python linux-headers udev &&   JOBS=MAX npm install -g node-red-node-serialport --unsafe-perm --build-from-source --production --silent &&   JOBS=MAX npm install -g node-red-contrib-modbus --unsafe-perm --build-from-source --production --silent &&   apk del make gcc g++ python &&   npm cache clean --force && rm -rf /tmp/*
[34m[main][39m   ---> Running in f8029267be99
[34m[main][39m  fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/main/armhf/APKINDEX.tar.gz
[34m[main][39m  fetch http://dl-cdn.alpinelinux.org/alpine/v3.8/community/armhf/APKINDEX.tar.gz
[34m[main][39m  (1/14) Installing binutils (2.30-r5)
[34m[main][39m  (2/14) Installing isl (0.18-r0)
[34m[main][39m  (3/14) Installing libgomp (6.4.0-r8)
[34m[main][39m  (4/14) Installing libatomic (6.4.0-r8)
[34m[main][39m  (5/14) Installing mpfr3 (3.1.5-r1)
[34m[main][39m  (6/14) Installing mpc1 (1.0.3-r1)
[34m[main][39m  (7/14) Installing gcc (6.4.0-r8)
[34m[main][39m  (8/14) Installing musl-dev (1.1.19-r10)
[34m[main][39m  (9/14) Installing libc-dev (0.7.1-r0)
[34m[main][39m  (10/14) Installing g++ (6.4.0-r8)
[34m[main][39m  (11/14) Installing linux-headers (4.4.6-r2)
[34m[main][39m  (12/14) Installing make (4.2.1-r2)
[34m[main][39m  (13/14) Installing gdbm (1.13-r1)
[34m[main][39m  (14/14) Installing python2 (2.7.15-r0)
[34m[main][39m  Executing busybox-1.28.4-r0.trigger
[34m[main][39m  OK: 208 MiB in 84 packages
[34m[main][39m  [91m
[34m[main][39m  [0m
[34m[main][39m  [91m/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/node_modules/prebuild-install/node_modules/which-pm-runs/index.js:11
[34m[main][39m  [0m
[34m[main][39m  [91m  const pmSpec = userAgent.split(' ')[0]
[34m[main][39m    ^^^^^
[34m[main][39m  [0m
[34m[main][39m  [91mSyntaxError: Use of const in strict mode.
[34m[main][39m      at Module._compile (module.js:439:25)
[34m[main][39m      at Object.Module._extensions..js (module.js:474:10)
[34m[main][39m      at Module.load (module.js:356:32)
[34m[main][39m      at Function.Module._load (module.js:312:12)
[34m[main][39m      at Module.require (module.js:364:17)
[34m[main][39m      at require (module.js:380:17)
[34m[main][39m      at Object.<anonymous> (/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/node_modules/prebuild-install/bin.js:5:19)
[34m[main][39m      at Module._compile (module.js:456:26)
[34m[main][39m      at Object.Module._extensions..js (module.js:474:10)
[34m[main][39m      at Module.load (module.js:356:32)
[34m[main][39m  [0m
[34m[main][39m  make: Entering directory '/usr/local/lib/node_modules/node-red-node-serialport/node_modules/serialport/build'
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport_unix.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/poller.o
[34m[main][39m    CXX(target) Release/obj.target/serialport/src/serialport_linux.o
[34m[main][39m  [91m../src/poller.cpp: In constructor 'Poller::Poller(int)':
[34m[main][39m  ../src/poller.cpp:12:39: error: could not convert 'status' from 'int' to 'uv_err_t {aka uv_err_s}'
[34m[main][39m       Nan::ThrowError(uv_strerror(status));
[34m[main][39m                                         ^
[34m[main][39m  [0m
[34m[main][39m  [91m../src/poller.cpp: In member function 'void Poller::poll(int)':
[34m[main][39m  ../src/poller.cpp:39:43: error: could not convert 'status' from 'int' to 'uv_err_t {aka uv_err_s}'
[34m[main][39m       Nan::ThrowTypeError(uv_strerror(status));
[34m[main][39m                                             ^
[34m[main][39m  [0m
[34m[main][39m  [91m../src/poller.cpp: In member function 'void Poller::stop()':
[34m[main][39m  ../src/poller.cpp:47:43: error: could not convert 'status' from 'int' to 'uv_err_t {aka uv_err_s}'
[34m[main][39m       Nan::ThrowTypeError(uv_strerror(status));
[34m[main][39m                                             ^
[34m[main][39m  [0m
[34m[main][39m  [91m../src/poller.cpp: In static member function 'static void Poller::onData(uv_poll_t*, int, int)':
[34m[main][39m  ../src/poller.cpp:58:75: error: could not convert 'status' from 'int' to 'uv_err_t {aka uv_err_s}'
[34m[main][39m       argv[0] = v8::Exception::Error(Nan::New<v8::String>(uv_strerror(status)).ToLocalChecked());
[34m[main][39m                                                                             ^
[34m[main][39m  [0m
[34m[main][39m  [91m../src/poller.cpp:69:29: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     obj->callback.Call(2, argv);
[34m[main][39m                               ^
[... 73 lines cut ...]
[34m[main][39m     ^~~~
[34m[main][39m  [0m
[34m[main][39m  [91m../src/serialport.cpp: In function 'void EIO_AfterDrain(uv_work_t*)':
[34m[main][39m  ../src/serialport.cpp:424:30: warning: 'v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const' is deprecated [-Wdeprecated-declarations]
[34m[main][39m     data->callback.Call(1, argv);
[34m[main][39m                                ^
[... cut ...]
[31m[Error][39m    Service: main
[31m[Error][39m      Error: The command '/bin/sh -c apk add --no-cache make gcc g++ python linux-headers udev &&   JOBS=MAX npm install -g node-red-node-serialport --unsafe-perm --build-from-source --production --silent &&   JOBS=MAX npm install -g node-red-contrib-modbus --unsafe-perm --build-from-source --production --silent &&   apk del make gcc g++ python &&   npm cache clean --force && rm -rf /tmp/*' returned a non-zero code: 1
[31m[Error][39m  Not deploying release.
```

rjbaat, 2018-08-03:
As suggested by Resin engineer i used another Alpine base img: 8.11.3 what was a LTS version and now it does build successfully again! 👍