BiancoRoyal/node-red-contrib-modbus #84: modbus flex:connector RTU mode problem

behi88, 2018-02-24:
Hi
I used the recent node-red-contrib-modbus and when i add new l=flex connector and use server in RTU mode the node-red occupy my RAM and CPU .
my hardware is RPI3.
it looks like it is a none block loop.
any ideas?

behi88, 2018-02-24:
`[{"id":"f448f23f.2cf62","type":"tab","label":"Modbus RTU Master","disabled":false,"info":""},{"id":"5a58d4c5.b66f5c","type":"inject","z":"f448f23f.2cf62","name":"","topic":"Select * From Device ORDER BY unitId ASC;","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":380,"y":340,"wires":[["d67db9a3.3add78","8d8a986f.c64dd8"]]},{"id":"d67db9a3.3add78","type":"debug","z":"f448f23f.2cf62","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","x":870,"y":240,"wires":[]},{"id":"8d8a986f.c64dd8","type":"modbus-flex-connector","z":"f448f23f.2cf62","name":"","maxReconnectsPerMinute":4,"emptyQueue":false,"showStatusActivities":false,"showErrors":false,"server":"3b31095e.770306","x":950,"y":340,"wires":[]},{"id":"3b31095e.770306","type":"modbus-client","z":"","name":"Serial Port","clienttype":"serial","bufferCommands":false,"stateLogEnabled":false,"tcpHost":"127.0.0.1","tcpPort":"502","tcpType":"DEFAULT","serialPort":"/dev/ttyAMA0","serialType":"RTU-BUFFERD","serialBaudrate":"9600","serialDatabits":"8","serialStopbits":"1","serialParity":"none","serialConnectionDelay":"100","unit_id":"1","commandDelay":"1","clientTimeout":"1000","reconnectTimeout":"2000"}]`

biancode, 2018-02-24:
Hi, I need some logs with DEBUG=modbus-serial*,contribModbus:flex:connector*, please!

biancode, 2018-02-24:
the flex connector just send an event to the connector and thats it, there is no loop

behi88, 2018-02-24:
this is my $ htop report if it helps:
[image]

behi88, 2018-02-24:
there is no signaling for DEBUG.
as soon as I add flex connector to my project and assign a SERVER with RTU for connection type it occupy my RAM and CPU usage.
and only if I Remove the NODE and the SERVER together , my CPU load will came back to normal.

biancode, 2018-02-25:
Please, could you try to test just to connect via serialport lib and modbus-serial lib?
Do you have this problem with that libs too?

behi88, 2018-02-25:
let me check...

behi88, 2018-02-25:
You are right there is a problem with that lib too:
I checked modbus-serial lib and there is a similar problem
when I add the the node "modbus-serial Out" to my project and add a server and Deploy , every thing seams to be fine.
but when I send a dummy payload to the node, my RAM stays at 15% but my CPU go's up up 50%!
strange...

behi88, 2018-02-25:
and the node-red stop working with these errors:
26 Feb 01:35:37 - [info] [modbusSerialConfig:faacf2e6.503cc] Error: Error: Timed out
26 Feb 01:35:56 - [info] Stopping flows
26 Feb 01:35:57 - [red] Uncaught Exception:
26 Feb 01:35:57 - Error: Error: Bad file descriptor, calling write

biancode, 2018-02-25:
@yaacov could you help, please?

behi88, 2018-02-25:
Maybe the problem is  when there's no actual device connected to the port.
I could run some test on a few slave devices back in lab tomorrow .

yaacov, 2018-02-26:
> It seems the problem is with native RP3 serial port

raspberry pi serial tty is 3.3v, rs232 (usb->serial) is 12v, what do you use to connect the raspberry pi serial pins (tty ? ) to the device serial (rs232 ? ) line ?

behi88, 2018-02-26:
We use MAX3232 TTL to RS232 Level converter and ADAM4520 for RS232 to RS458 level converter

yaacov, 2018-02-26:
> We use MAX3232 TTL to RS232 Level converter and ADAM4520 for RS232 to RS458 level converter

AFAIK rs485 connection requires a direction pin that the serial library we use [1] does not support, so if you need rs485 you probably need a usb-to-rs485 device, or use electronics to create a direction signal from tx/rx ( like the usb module does ).

[1] https://github.com/node-serialport/node-serialport/issues/903

behi88, 2018-02-26:
ADAM4520 module Does that, it create a direction signal from tx/rx and also buffer the data.
in other word it do not need direction pin and work without one.

yaacov, 2018-02-26:
> ADAM4520 module Does that, it create a direction signal from tx/rx and also buffer the data.
in other word it do not need direction pin and work without one.

Nice :+1: 

My next guess will be baud-rate, start,stop,data bits and parity :-( but I can't think of anything other than that :-( 

p.s.
you can also check that the serial pins work by connecting to a device that talk tty and send some data, just to be sure the serial pins in the raspberry are not fried.

behi88, 2018-02-26:
I check the tx/rx pins and they are fine and also I checked the baud rate, stop bit and parity with LIBMODBUS c library seems to work fine.
as I send my 'htop' result afew post earlier the nod-red occupy mu CPU and RAM
the question is: is hardware failure can make node-red occupy the CPU and RAM?
personally I don't mind using USB to Serial converter
but my problem with USB is "Error FSM Not Ready to Read" when the USB is not connected and the node want to read from it. and it is not reset until the next deploy or node-red reset
do you now how could  reset the Error without reset the whole thing?

yaacov, 2018-02-26:
> the question is: is hardware failure can make node-red occupy the CPU and RAM?
> ...
> do you now how could reset the Error without reset the whole thing?

@biancode ?

biancode, 2018-02-27:
"Error FSM Not Ready to Read" - the serial port should try to reconnect after some time - could send me more log with DEBUG=contribModbus*,modbus-serial

biancode, 2018-02-27:
if the port is open, then it should read or write if the USB is connected without a deploy

biancode, 2018-02-27:
"Port Not Open" brings a reconnect while it is connected

behi88, 2018-02-27:
Ok this is what i do:

1-connected my ttyUSB0 port
2-send a dummy read to Modbus flex reader
3- because of no salve device I've got "timeout"
4- I disconnect the USB port
5- send a dummy  read to Modbus flex again
6-I've god "initialize'
7- send the dummy read again 
8- I've got "Error FSM Not Read to Read"
9- wait for 10 seconds and reconnect the USB
10 nothing happens 
11- wait for 2 min and send the dummy byte and still getting "Error FSM ..."
12- as I type this my node still in "Error..." and my USB is Still Connected

this is the above DEBUG report:

  contribModbus:core:client Client -> initialized +0ms
27 Feb 23:53:07 - [warn] [modbus-client:9f276dcd.05f55] Client -> reconnect in 2000 ms Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1
  contribModbus:core:client Client -> event: init old: NEW new: INIT Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +7ms
27 Feb 23:53:07 - [info] Started flows
  contribModbus:core:client Client -> RTU buffered port serial Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +2s
  contribModbus:core:client Client -> event: openserial old: INIT new: OPENED Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +26ms
  contribModbus:core:client Client -> time to open Unit 1 Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +101ms
  modbus-serial modbus connection opened +0ms
  contribModbus:core:client Client -> {"state":"QUEUEING","message":"dequeue command serial","delay":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +5ms
  contribModbus:core:client Client -> {"type":"queue check is not a unit","unitid":null} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +1ms
  contribModbus:flex:getter Add Message 5a95be55a872a80b87a8c4d2 +0ms
  contribModbus:core:client Client -> {"info":"push to Queue by Unit-Id","message":{"unitid":1,"fc":1,"address":0,"quantity":1,"messageId":"5a95be55a872a80b87a8c4d2"},"unit":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +40s
  contribModbus:core:client Client -> {"info":"queue read msg","message":{"unitid":1,"fc":1,"address":0,"quantity":1,"messageId":"5a95be55a872a80b87a8c4d2"},"state":"EMPTY","queueLength":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +2ms
  contribModbus:core:client Client -> {"state":"QUEUEING","message":"dequeue command serial","delay":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +12ms
  contribModbus:core:client Client -> {"type":"queue check","unitid":1,"sendAllowed":true,"queueLength":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +1ms
  contribModbus:core:client Client -> {"type":"serial sending and wait","unitid":1,"queueLength":0,"sendAllowedForNext":false,"delay":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +0ms
  contribModbus:core:client Client -> {"info":"read msg","message":{"unitid":1,"fc":1,"address":0,"quantity":1,"messageId":"5a95be55a872a80b87a8c4d2"},"unitid":1,"timeout":1000,"state":"QUEUEING"} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +3ms
  modbus-serial { action: 'send serial rtu buffered',
  modbus-serial   data: <Buffer 01 01 00 00 00 01 fd ca>,
  modbus-serial   unitid: 1,
  modbus-serial   functionCode: 1 } +0ms
  contribModbus:core:client Client -> {"info":"queue response activate sending","message":{"unitid":1,"fc":1,"address":0,"quantity":1,"messageId":"5a95be55a872a80b87a8c4d2"}} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +1s
  contribModbus:flex:getter Timed out +1s
  contribModbus:core Remove Message In:<5a95be55a872a80b87a8c4d2> Out:<5a95be55a872a80b87a8c4d2> +0ms
  modbus-serial modbusErrorHandling:Timed out +41s
  contribModbus:core:client Client -> {"state":"QUEUEING","message":"dequeue command serial","delay":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +6ms
  contribModbus:core:client Client -> {"type":"queue check is not a unit","unitid":null} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +1ms
  contribModbus:flex:getter Add Message 5a95be6ba872a80b87a8c4d3 +21s
  contribModbus:core:client Client -> {"info":"push to Queue by Unit-Id","message":{"unitid":1,"fc":1,"address":0,"quantity":1,"messageId":"5a95be6ba872a80b87a8c4d3"},"unit":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +21s
  contribModbus:core:client Client -> {"info":"queue read msg","message":{"unitid":1,"fc":1,"address":0,"quantity":1,"messageId":"5a95be6ba872a80b87a8c4d3"},"state":"EMPTY","queueLength":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +1ms
  contribModbus:core:client Client -> {"state":"QUEUEING","message":"dequeue command serial","delay":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +4ms
  contribModbus:core:client Client -> {"type":"queue check","unitid":1,"sendAllowed":true,"queueLength":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +0ms
  contribModbus:core:client Client -> {"type":"serial sending and wait","unitid":1,"queueLength":0,"sendAllowedForNext":false,"delay":1} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +1ms
  contribModbus:core:client Client -> {"info":"read msg","message":{"unitid":1,"fc":1,"address":0,"quantity":1,"messageId":"5a95be6ba872a80b87a8c4d3"},"unitid":1,"timeout":1000,"state":"QUEUEING"} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +1ms
  contribModbus:core:client Client -> {"info":"queue response activate sending","message":{"unitid":1,"fc":1,"address":0,"quantity":1,"messageId":"5a95be6ba872a80b87a8c4d3"}} Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 1 +3ms
  contribModbus:flex:getter Port Not Open +13ms
[... cut ...]
  contribModbus:core Remove Message In:<5a95be78a872a80b87a8c4d4> Out:<5a95be78a872a80b87a8c4d4> +13s
  contribModbus:flex:getter Add Message 5a95bea5a872a80b87a8c4d5 +46s
  contribModbus:flex:getter FSM Not Ready To Read +1ms
  contribModbus:core Remove Message In:<5a95bea5a872a80b87a8c4d5> Out:<5a95bea5a872a80b87a8c4d5> +46s
  contribModbus:flex:getter Add Message 5a95beb4a872a80b87a8c4d6 +14s
  contribModbus:flex:getter FSM Not Ready To Read +1ms
  contribModbus:core Remove Message In:<5a95beb4a872a80b87a8c4d6> Out:<5a95beb4a872a80b87a8c4d6> +14s

behi88, 2018-02-27:
I hope it helps. :)

biancode, 2018-02-28:
> modbus-serial error: Error: Error: No such file or directory, cannot open /dev/ttyUSB0 } +0ms
> modbus-serial modbusErrorHandling:Error: No such file or directory, cannot open /dev/ttyUSB0 +4s

modbusErrorHandling has to differentiate between Serial and TCP - I'll do that.

behi88, 2018-02-28:
Great! thanks
I know it is not appropriate to send suggestion in issue form but as i may...
I tested modbus tcp so many times, it is a very robust component
but it could really use dynamic server change in read and write.
let's say I have more the 10 Modbus TCP slave. I configure the servers at start up with Modbus flex connector I need to add one Modbus flex getter for each server to read their data.
sure I could just use 1 Flex connector and change the server every time I want to connect, but it takes time and do not work with big Modbus networks and also some slaves could not handle the TCP disconnect very well and stay connected for a long time and the second request from the same master become a new connection to them and eventually they overflow.
in Modbus RTU it is different: you have only 1 bus specification (baud rate , stop bit ,...) and the slaves distinguished by their Slave ID , so you don't need multi getter to connect to slaves
anyway it would be REALLY practical if you could add a server name (or IP) to flex getter and writer parameters
this way  we could have 1 getter for all servers.

Thanks again.

biancode, 2018-02-28:
@behi88 the last is a new issue, please! Delete it from here and add a new issue!

biancode, 2018-03-05:
please, test v3.4.0

behi88, 2018-03-06:
Hi'
Thank you biancode.
I tested it with any scenario I could think of and the problem 's gone and it works great.
but in in 3.4.0 in when you want to add a server in RTU mode there where a tools to search for active serial ports and it's stop working. 
you still could add the port by typing it's name. but the search tool was very handy.
maybe it is a browser issue. I could check that to be sure tonight.
I could open and new issue and close this one though if you said so.

anyway well done and great work. :)

biancode, 2018-03-06:
Please, try a fresh install of the package. On MACOS it works well with the lookup for ports.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

behi88, 2018-03-06:
no not working with fresh install either ...
I tried with different browser as well
after click on search button it deactivated and no serial port shown.
I could capture a video if it helps.

behi88, 2018-03-07:
sorry @biancode I do not think it is a big deal.
anyway I'm closing this issue for others reference.
;)

closed 2018-03-07 by behi88