BiancoRoyal/node-red-contrib-modbus #86: Error Data length when reading

shahbour, 2018-03-03:
Hello 

I am working on the battery Modbus we talked about in [issue](https://github.com/biancode/node-red-contrib-modbus/issues/80) and found my problem was I am using port RS232 instead of 485 (Battery has both ports).

So now I moved forward and I am getting a reply from the battery but it is not showing up in node-red 

I am getting `error Data length error, expected 9 got 1`, but when sniffy the serial port I see the below, 9 bytes with correct values have been returned.

Is it possible because I am using USB to RS485?

>       0x26 (&)
>       0x10
>       0x00
>       0x00
>       0x02
>       0x73 (s)
>       0xfc

>       0x26 (&)
>       0x04
>       0x04
>       0x14
>       0x8e
>       0x27 (')
>       0x10
>       0xd3
>       0x61 (a)

Thanks

biancode, 2018-03-03:
@yaacov is that maybe an issue for modbus-serial?

yaacov, 2018-03-03:
> @yaacov is that maybe an issue for modbus-serial?

are you using the buffered serial port ?

shahbour, 2018-03-03:
I kept the default , if you mean RTU no I  am using Modbus RTU not buffered . 

I tried the flow in raspberry it worked couple of time but most often I receive , expected 9 got 2 or 4 ...

Now I am ordering ttl to 485 to get rid of USB and try . 

Thanks

yaacov, 2018-03-03:
I thinks that the `RTUBufferedPort` should solve this, but AFAIK it is the one used by node-red, @biancode ?

biancode, 2018-03-03:
RTU Buffered is default and you need that mode to run if you don't have a Modbus gateway.

biancode, 2018-03-03:
The mode just RTU in serial means that there is a gateway, which has a queue and handles completion of the response. In that case we don't have to do the overhead to check every messages for complete state.

shahbour, 2018-03-03:
Ok I see , I will give it a try with RTUBuffered . Till Monday as battery is very heavy only at office

shahbour, 2018-03-05:
Worked perfectly,  now I am able to read the data. I noticed sometimes it is showing timeout but still i get the data in the debug.

Thanks for your help

closed 2018-03-05 by biancode