BiancoRoyal/node-red-contrib-modbus #89: Questions and possible enhancement in queues

bashlakov, 2018-03-29:
Hi!

I'm trying to use modbus-flex-getter to poll devices in a loop with a minimum delay. Currently, I connect it to a function that dynamically generates several requests at once and triggered by injecting a node with a certain repetition interval. In this setup, sometimes I get a slowly increasing queue length. The increasing cycle period seems to solve my problem, but I want to send requests as soon as possible. So, I have 2 feature requests:
1) Is there abillity to add queue length limit? In my case, it's okay to lose some requests, but speed does matters, so I can not completely turn off the queue or increase the cycle period. At the same time, the queue, which grows without any restrictions, can create problems with ending up of RAM, especially if there are many Modbus nodes and low RAM on server. Also I think that there may be a performance issue with node-red if the queue becomes too large. Therefore, I think that it makes sense to limit the queue at some adjustable level to prevent its uncontrolled expansion and still get good performance and other benefits that we get through the queue.
2) Is there ability to implement reply for queues status requests in Queue-Info nodes? I would like to export the queue metrics to an external monitoring system, but I do not see the possibility of getting this information from the Queue-Info now, I can only receive the message if the limit setted in this module is exceeded, which is not I actually expect. I think it will be great if this node could provide complete statistics of the queue by request.

What do you think, is this a good ideas?

biancode, 2018-04-03:
Hi, did you watch the [YouTube videos](https://www.youtube.com/watch?v=ND9ZYkMOwZs&t=0s&index=12&list=PLdcYFhtEB7okPHU1-NUYTeKFNbJOW_-gH)?

The queue handling works via catch node.

bashlakov, 2018-04-04:
Hi! 
Thanks for link. As I understood, I need to add Queue info module for every UnitID in this case, am I right? In my environment I have multiple devices behind single modbus-tcp converter and function block that read config and dynamically generates queries, so adding Queue info module for every UnitID is manual work that I wants to avoid.
Also, I would like to monitoring queue length to know about possible problems, in this example I only can set trappers for reset messages, that is not flexible solution I think.

biancode, 2018-04-09:
OK, please write down the requirements and describe precisely what you expect!

biancode, 2018-04-29:
reopen if with the requirements

closed 2018-04-29 by biancode