BiancoRoyal/node-red-contrib-modbus #90: ObjectID error

gemini86, 2018-04-02:
Hello,

I had been using the modbus read node successfully until an update a little while ago. What's happening now is that when I pass the output to a dashboard template node, I get "Error: invalid ObjectId, ObjectId.id must be either a string or a Buffer, but is [{"_placeholder":true,"num":1}]". I cannot seem to track this down to any other source of input into the template node, only seems to happen when the modbus read node is connected. Any help is much appreciated.

biancode, 2018-04-02:
Hello,

some logs or stacktrace to see what happens?

gemini86, 2018-04-03:
Thanks for your time!

`3 Apr 05:50:03 - [error] [ui_template:valve message] Error: invalid ObjectId, ObjectId.id must be either a string or a Buffer, but is [{"_placeholder":true,"num":1}]
    at ObjectID.toHexString (/home/[user]/.node-red/node_modules/bson/lib/bson/objectid.js:88:11)
    at ObjectID.toJSON (/home/[user]/.node-red/node_modules/bson/lib/bson/objectid.js:196:15)
    at Object.stringify (<anonymous>)
    at encodeAsString (/home/[user]/.node-red/node_modules/socket.io-parser/index.js:178:17)
    at writeEncoding (/home/[user]/.node-red/node_modules/socket.io-parser/index.js:199:16)
    at Object.exports.removeBlobs (/home/[user]/.node-red/node_modules/socket.io-parser/binary.js:139:5)
    at encodeAsBinary (/home/[user]/.node-red/node_modules/socket.io-parser/index.js:206:10)
    at Encoder.encode (/home/[user]/.node-red/node_modules/socket.io-parser/index.js:130:5)
    at Adapter.broadcast (/home/[user]/.node-red/node_modules/socket.io-adapter/index.js:121:16)
    at Namespace.emit (/home/[user]/.node-red/node_modules/socket.io/lib/namespace.js:222:18)
    at Server.(anonymous function) [as emit] (/home/[user]/.node-red/node_modules/socket.io/lib/index.js:425:29)
    at TemplateNode.<anonymous> (/home/[user]/.node-red/node_modules/node-red-dashboard/ui.js:242:16)
    at TemplateNode.emit (events.js:127:13)
    at TemplateNode.Node.receive (/usr/local/lib/node_modules/node-red/red/runtime/nodes/Node.js:215:14)
    at FunctionNode.Node.send (/usr/local/lib/node_modules/node-red/red/runtime/nodes/Node.js:202:14)
    at sendResults (/usr/local/lib/node_modules/node-red/nodes/core/core/80-function.js:52:18)`

biancode, 2018-04-03:
ui_template:valve message is more dashboard not Modbus ... please use the Slack group to get help for that

closed 2018-04-03 by biancode

gemini86, 2018-05-04:
Hello, just a follow up, I was able to solve this issue with the dashboard template node by deleting either msg.input or msg.responseBuffer from the message generated out of the ModbusRead node.