BiancoRoyal/node-red-contrib-modbus #93: FSM State On Reset INIT

biancode, 2018-04-29:
from @apanasara  

Please help to resolve error msg 'FSM State On Reset INIT'
https://user-images.githubusercontent.com/902578/38973172-f4dc8ab4-43c1-11e8-9207-6bbd4e5a00dc.jpg

biancode, 2018-04-29:
Hi, I need your [log from the console](https://github.com/biancode/node-red-contrib-modbus/wiki/DEBUG), please. What is the response from the device?

biancode, 2018-04-29:
What about the Modbus error code? I am sorry, but I need more details and information.

apanasara, 2018-05-04:
No error code... Written Only  'FSM State On Reset INIT'

biancode, 2018-05-04:
'FSM State On Reset INIT' is normaly if the connection settings are wrong

tinktonk, 2018-05-09:
We have the exact problem, usually working but it looks like we have some problem with the serial connection. If we reset the serialport it starts working again. We are running node-red on raspberry. 
Everytime we deploy we need to reset.

biancode, 2018-05-22:
Please, try to give some delay in time for the device to connect if you are using the inject node.

[image]

biancode, 2018-05-22:
[image]

... that is from the Modbus Read node

biancode, 2018-05-22:
reopen or open some new if you got more to analyze

closed 2018-05-22 by biancode