BiancoRoyal/node-red-contrib-modbus #95: More "FSM Reset On State INIT" messages

tmsound, 2018-05-21:
This is bizarre, I have a flow that talks to a device in the same class and it works.  I changed nothing in the code and plugged it into another device (nearly identical) and am hit with the dreaded reset messages.  I am converting floats and using flex getter, however I see you have mentioned if I must use more than 10 nodes (I think I must) ...that I should open multiple connections.  Do you simply mean I need to create additional modbus server connections to the same IP address? Should they all have the same timeout etc settings?  Splitting the difference with 2 working halves of the flow doesn't work. I have been working with this node for a few months now and just recently have been hitting it with too many queries.  Started to try to save the que by turning off queries per tab.  Any other advice?  I'd be happy to contribute financially to the project.  This was all going fine until I started trying to dump 'recipes' of floats while also reading them.

tmsound, 2018-05-22:
Ok, I saw you were asking for proper logging.  Here is my console log.  Today it is acting a little janky the original piece of hardware is now also timing out some.  I think I had it working when they were all using the same connection but swapping out queries per tab, was running out of ideas how to trouble shoot the 2nd system since they would never even initialize. Here is log for that system.
22 May 14:35:36 - [info] 

Welcome to Node-RED
===================

22 May 14:35:36 - [info] Node-RED version: v0.18.5
22 May 14:35:36 - [info] Node.js  version: v6.14.2
22 May 14:35:36 - [info] Linux 4.9.82-ti-r102 arm LE
22 May 14:35:39 - [info] Loading palette nodes
22 May 14:36:03 - [info] Dashboard version 2.9.3 started at /ui
22 May 14:36:05 - [warn] ------------------------------------------------------
22 May 14:36:05 - [warn] [node-red/rpi-gpio] Info : Ignoring Raspberry Pi specific node
22 May 14:36:05 - [warn] ------------------------------------------------------
22 May 14:36:05 - [info] Settings file  : /home/debian/.node-red/settings.js
22 May 14:36:05 - [info] User directory : /home/debian/.node-red
22 May 14:36:05 - [warn] Projects disabled : set editorTheme.projects.enabled=true to enable
22 May 14:36:05 - [info] Flows file     : /home/debian/.node-red/flows_beaglebone.json
22 May 14:36:05 - [info] Server now running at http://127.0.0.1:1880/
22 May 14:36:05 - [warn] 

---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.

If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.

You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------

22 May 14:36:07 - [info] Starting flows
{ id: 'd8c6516f.e50d2',
  type: 'sqldbsdatabase',
  z: '',
  host: '10.8.0.1',
  port: '3306',
  db: 'udc_3500',
  dialect: 'mysql',
  _users: [] }
{ id: 'd93ba84.1b3d8d8',
  type: 'sqldbsdatabase',
  z: '',
  host: '10.8.0.1',
  port: '3306',
  db: 'udc_3500',
  dialect: 'mysql',
  _users: [ 'e7595cf6.981958' ] }
22 May 14:36:07 - [warn] [modbus-client:udc] Client -> reconnect in 4000 ms TCP@10.0.0.2:502 default Unit-Id: 1
22 May 14:36:07 - [warn] [modbus-client:udc2] Client -> reconnect in 2000 ms TCP@10.0.0.2:502 default Unit-Id: 1
{ id: '662bf9cd.e5bce8',
  type: 'sqldbsdatabase',
  z: '839b3fb8.819ce',
  host: '127.0.0.1',
  port: '3306',
  db: 'udc_3200',
  dialect: 'mysql' }
dialect: mysql
Sequelize {
  options: 
   { dialect: 'mysql',
     dialectModulePath: null,
     host: '10.8.0.1',
     protocol: 'tcp',
     define: {},
     query: {},
     sync: {},
     timezone: '+00:00',
[... 862 lines cut ...]
22 May 14:37:18 - [warn] [modbus-flex-getter:flex getem] FSM Reset On State INIT
22 May 14:37:18 - [warn] [modbus-flex-getter:flex getem] FSM Reset On State INIT
22 May 14:37:18 - [warn] [modbus-flex-getter:flex getem] FSM Reset On State INIT
22 May 14:37:18 - [warn] [modbus-flex-getter:flex getem] FSM Reset On State INIT
22 May 14:37:18 - [warn] [modbus-flex-getter:flex getem] FSM Reset On State INIT
22 May 14:37:18 - [warn] [modbus-flex-getter:flex getem] FSM Reset On State INIT
22 May 14:37:18 - [warn] [modbus-flex-getter:flex getem] FSM Reset On State INIT
22 May 14:37:18 - [warn] [modbus-flex-getter:flex getem] FSM Reset On State INIT
22 May 14:37:18 - [warn] [modbus-getter:Modbus Get 6] FSM Reset On State INIT
22 May 14:37:18 - [warn] [modbus-getter:read Ramp Units] FSM Reset On State INIT
22 May 14:37:18 - [warn] [modbus-flex-getter:flex getem] FSM Reset On State INIT
22 May 14:37:18 - [warn] [modbus-client:udc2] Client -> try to reconnect by init in 2000 ms TCP@10.0.0.2:502 default Unit-Id: 1
22 May 14:37:18 - [info] [inject:poll button] stopped
22 May 14:37:18 - connection is closing
22 May 14:37:18 - [info] [inject:poll button] stopped
22 May 14:37:18 - [info] [inject:1 min Resetter] stopped
22 May 14:37:18 - [info] [inject:3 min Resetter] stopped
22 May 14:37:18 - [info] [inject:Query Modbus] stopped
22 May 14:37:18 - [info] [inject:Query Modbus] stopped
22 May 14:37:18 - [info] [inject:Query Modbus] stopped
22 May 14:37:18 - [info] [inject:Query Modbus] stopped
22 May 14:37:18 - [info] [inject:Query Modbus] stopped
22 May 14:37:18 - [info] [inject:Always run] stopped
22 May 14:37:18 - [warn] [modbus-client:udcmaybe2] Client -> try to reconnect by init in 6000 ms TCP@10.0.0.2:502 default Unit-Id: 1
22 May 14:37:19 - [info] Stopped flows

biancode, 2018-05-22:
[image]

biancode, 2018-05-22:
please, send me your Modbus flow parts via E-Mail ...

tmsound, 2018-05-22:
The flow in question is flow 2, however I included flow 1 and 3 for context
since they are all connected.   I have an iterator with delays for a dump
write that works most of the time.  I had tried chaining inject nodes with
triggers/delays.  What would be a good strategy for this? Do I have too
many queries?  Should I recode this in a differnent way?  In flow 2 in the
lower right where I start doing the floats and break them into arrays is
the section added that started hosing the system.  As I write this email it
is reading but writing stopped working agian, I think my que is just
backing up to much or buffers are flushing or something.  I built all the
parts individually before trying to combine them so they work on their
own.  I will try try delaying the inject nodes.  Thanks for the reply.
Trevor Morris

biancode, 2018-05-22:
.. and your flows doesn't work on my side - please send a JSON data exported from Node-RED via E-Mail ;)

biancode, 2018-05-28:
Reopen if you need more

closed 2018-05-28 by biancode