BiancoRoyal/node-red-contrib-modbus #97: modbus cause restart of nodered

nickomalley1, 2018-05-22:
Hello all,

I have started experimenting with the modbus flows but keep hitting a problem that crashes nodered,  I have copied the error below.  

another side effect of this is all my other contributed flows dissapear when this first occurs.

Anyway, hoping you can point me in the right direction to resolve this.

Thanks
Nick

22 May 20:40:40 - [info] Starting flows
INFO : ModbusTCPServer log level: info
INFO : ModbusTCPServer log level: info
INFO : ModbusTCPServer log level: info
22 May 20:40:41 - [info] Started flows
22 May 20:40:41 - [red] Uncaught Exception:
22 May 20:40:41 - Error: listen EADDRINUSE 0.0.0.0:10502
    at Object.exports._errnoException (util.js:1020:11)
    at exports._exceptionWithHostPort (util.js:1043:20)
    at Server._listen2 (net.js:1271:14)
    at listen (net.js:1307:10)
    at net.js:1417:9
    at _combinedTickCallback (internal/process/next_tick.js:83:11)
    at process._tickCallback (internal/process/next_tick.js:104:9)
nodered.service: Main process exited, code=exited, status=1/FAILURE
nodered.service: Unit entered failed state.
nodered.service: Failed with result 'exit-code'.
nodered.service: Service hold-off time over, scheduling restart.

biancode, 2018-05-22:
You could set ports to 10502, 11502, 12502 ... check every port to be unique

closed 2018-05-22 by biancode

nickomalley1, 2018-05-22:
Thank you ,  

that worked great.

regards
Nick

biancode, 2022-07-29:
With v3.0.x of node-red we decided to handle errors in our package from v5.22.x+