BiancoRoyal/node-red-contrib-modbus #99: [Question] Modbus Server and Input registers

Javierete, 2018-07-04:
Hello Biancode,

I want to preset some input registers to node-red own ModBus server, as there are no write function to do that. I'm figuring out how to do or if it's supported (assign a value to Input register to be read only value)

It's possible to do this? 
Thanks in advance!

biancode, 2018-07-04:
Hello Javier,

if you want to set initial values to the server, then you could that with the inject node of Node-RED and the once operation to inject only once to the server with a Modbus-Writer node.

Was that your question?
cya
Klaus

Javierete, 2018-07-04:
Hello Klaus, probably not explained well (excuse my English).
At the moment I'm able to write correctly (Funcions 05, 06, 15 and 16) to local modbus server with a Flex-Writer so that's okay.

My question is I want to write to read only registers, or maybe instead of writting with a modbus funcion (because in protocol isn't possible) preset them with some kind of flow:
- Discrete Inputs (Function 02 to read them)
- Input Registers (Function 04 to read them)

Does this possibiliy exists?

biancode, 2018-07-04:
If your read-only Inputs are electrical I/O's, then you could switch them via electrical signals like the PI could do. Otherwise read-only is read-only because of they are stateful  ;)
Do you have some flow for that to take a look at it?

Javierete, 2018-07-04:
I'll explain a little bit further my use case ;)

I'm trying to make some kind of mqtt to modbus gateway (with node-red-contrib-modbus). Using Modbus server inside node-red to make a master PLC read the modbus registers.

When an MQTT message come into node-red I parse the values and make them available to different modbus registers, as said before no problem at all with holding registers and coils. 

But I need to set Discrete inputs and input registers, the reason to do this is to avoid PLC writing to these registers (these are status registers).
My options are, writing (or presetting) to these read-only registers (hence this question) or if not supported write to another R/W area (Coils/Holding Regs) and do nothing with possible value changes.
The reason to do that is to keep things separated at MB Server side, but no problem for me if cannot be accomplished.

Javierete, 2018-07-04:
Here is my sample test flow, the mqtt receives a json string with info from a device, then parse status, battery and make available to modbus: 

`[
    {
        "id": "1e56b33.c84204d",
        "type": "mqtt in",
        "z": "c6235980.95ffe8",
        "name": "",
        "topic": "/device/10/rx",
        "qos": "0",
        "broker": "65f2568b.d65208",
        "x": 90,
        "y": 280,
        "wires": [
            [
                "88043801.ceb9c8",
                "693a6914.de4bd8"
            ]
        ]
    },
    {
        "id": "88043801.ceb9c8",
        "type": "function",
        "z": "c6235980.95ffe8",
        "name": "Decode Data",
        "func": "//1. Functions\nfunction uint32ToFloat64(l1,l2,l3,l4) {\n    var bits = (l1 << 24) | (l2 << 16) | (l3 << 8) | (l4);\n    var sign = ((bits >>> 31) === 0) ? 1.0 : -1.0;\n    var e = ((bits >>> 23) & 0xff);\n    var m = (e === 0) ? (bits & 0x7fffff) << 1 : (bits & 0x7fffff) | 0x800000;\n    var floatVal = sign * m * Math.pow(2, e - 150);\n    return floatVal;\n}\nfunction uint16ToFloat32(low, high) {\n  var buffer = new ArrayBuffer(4);\n  var intView = new Uint16Array(buffer);\n  var floatView = new Float32Array(buffer);\n\n  intView[0] = low;\n  intView[1] = high;\n  return floatView[0];\n}\nfunction float32ToUint16(value) {\n  var buffer = new ArrayBuffer(4);\n  var intView = new Uint16Array(buffer);\n  var floatView = new Float32Array(buffer);\n\n  floatView[0] = value;\n  return [intView[0], intView[1]];\n}\nfunction round(value, decimals) {\n  return Number(Math.round(value+'e'+decimals)+'e-'+decimals);\n}\n\n//Get incoming msg and convert to JSON\nvar incoming    = JSON.parse(msg.payload);\n\n//Get variables\nvar appname     = incoming.applicationName;\nvar from        = incoming.deviceName;\nvar deveui      = incoming.devEUI;\nvar data        = new Buffer(incoming.data,'base64'.toString('ascii'));\n\n//\nvar vbatt =     round(uint32ToFloat64(data[0],data[1],data[2],data[3]),3);\nvar vinput =    round(uint32ToFloat64(data[4],data[5],data[6],data[7]),3);\nvar Valve1 =    data[8];\nvar Valve2 =    data[9];\n\n//Compose Complete message\nvar msg = {\n    payload:\n        {\n            \"AppName\": appname,\n            \"From\" : from,\n            \"DevEui\" : deveui,\n            \"data\" : {\n                \"Battery\": vbatt,\n                \"Vinput\" : vinput,\n                \"Valve1\" : Valve1,\n                \"Valve2\" : Valve2\n                }\n        }\n};\n\n//Compose mg message\nvar mg = {\n        payload : {\n            \"Battery\"   : vbatt,\n            \"Vinput\"    : vinput,\n            \"Valve1\"    : Valve1,\n            \"Valve2\"    : Valve2\n        }\n};\n\n//Compose ModBus message (Input Registers) READ ONLY\n//TODO: Ask Klaus Lansdorf if it's possible to write from here to Input Registers\nvar modbusIR = {\n    payload : {\n        'value'     : [Valve1, Valve2],\n        'fc'        : 15,\n        'unitid'    : 1,\n        'address'   : 100,\n        'quantity'  : 2\n    }\n};\n\n//Compose ModBus message (Holding Registers) READ/WRITE\nvar modbusHR = {\n    payload : {\n        'value'     : [\n            float32ToUint16(vbatt)[0], \n            float32ToUint16(vbatt)[1],\n            float32ToUint16(vinput)[0], \n            float32ToUint16(vinput)[1],            \n            ],\n        'fc'        : 16,\n        'unitid'    : 1,\n        'address'   : 100,\n        'quantity'  : 4\n    }\n};\n\nnode.send([msg,mg,modbusIR,modbusHR]);",
        "outputs": 4,
        "noerr": 0,
        "x": 490,
        "y": 280,
        "wires": [
            [
                "36399d83.644d22"
            ],
            [
                "c5567fcb.f2a25"
            ],
            [
                "c1a22218.145cd"
            ],
            [
                "c1a22218.145cd"
            ]
        ],
        "inputLabels": [
            "Incoming msg"
        ],
        "outputLabels": [
            "Output",
            "Metragest",
            "ModBus",
            ""
        ]
    },
    {
        "id": "693a6914.de4bd8",
        "type": "debug",
        "z": "c6235980.95ffe8",
        "name": "",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "x": 490,
        "y": 220,
        "wires": []
    },
    {
        "id": "36399d83.644d22",
[... 79 lines cut ...]
    },
    {
        "id": "f7d7a6cc.cd1998",
        "type": "modbus-client",
        "z": "",
        "name": "Local Server",
        "clienttype": "tcp",
        "bufferCommands": true,
        "stateLogEnabled": false,
        "tcpHost": "127.0.0.1",
        "tcpPort": "10502",
        "tcpType": "DEFAULT",
        "serialPort": "/dev/ttyUSB",
        "serialType": "RTU-BUFFERD",
        "serialBaudrate": "9600",
        "serialDatabits": "8",
        "serialStopbits": "1",
        "serialParity": "none",
        "serialConnectionDelay": "100",
        "unit_id": "1",
        "commandDelay": "1",
        "clientTimeout": "1000",
        "reconnectTimeout": "2000"
    }
]`

biancode, 2018-07-04:
If you set up a Flex-Server, then you get a deeper access to provide your own functions or you could test the [open plc](https://www.npmjs.com/package/node-red-contrib-openplc)

That is all I can do for now ...

biancode, 2018-07-04:
[open plc](url)

closed 2018-07-04 by biancode