BiancoRoyal/node-red-contrib-modbus #100: Modbus read

Somavaram, 2018-07-10:
Hi,

I am presently working on Siemens S7-1500 PLC. I configured the PLC as modbus client. Now, I am trying to read the holding registers using Modbus read. But unfortunately, I am unable to establish the connection, The status on the node shows "Initialize", changes to "Error FSM not ready to read at state INIT" and then reconnecting. I have no clue how to solve this issue please help.

I have also tried using node S7 node to connect the PLC but as soon as I enter the IP address and deploy it the status on the node changes to disconnect.

Please give your valuable suggestions.

Regards,
Somavaram

biancode, 2018-07-10:
Hi Somavaram,

did you see issue #93 or does it help to give some delay before reading?

Regards,
Klaus

Somavaram, 2018-07-11:
Hi @biancode ,

I have tried both ways but couldn't succeed and yes there is only one node trying to establish the connection to the plc(Modbus read).

biancode, 2018-07-11:
can you sen a picture of your configuration nodes --->  Node-RED menu -> configuration -> all  from the right upper corner, please

Somavaram, 2018-07-12:
[image]
I see nothing like config

biancode, 2018-07-12:
right upper corner  --- the three white lines is the Node-RED menu button ;) ---  
Node-RED menu -> Configuration nodes

biancode, 2018-07-12:
Looks good. OK, and now the settings of the modbus-client node, please!

biancode, 2018-07-12:
and the settings of your read node, please

Somavaram, 2018-07-12:
[image]
[image]

biancode, 2018-07-12:
all OK - now I need detailed [DEBUG logs](https://github.com/biancode/node-red-contrib-modbus/wiki/DEBUG) - you could send them via E-Mail

biancode, 2018-07-12:
closed until we get more information

closed 2018-07-12 by biancode

Somavaram, 2018-07-12:
@biancode ,
 Sorry for the late reply, How do I get these debug logs or where do i find it.

Somavaram, 2018-07-12:
I see no error in the debug(NR page) as well as command prompt.

biancode, 2018-07-12:
[DEBUG logs](https://github.com/biancode/node-red-contrib-modbus/wiki/DEBUG)

biancode, 2018-07-12:
we have youtube, wiki and the Node-RED Slack channel, please use it