BiancoRoyal/node-red-contrib-modbus #101: CRC or data length errors

demetz, 2018-07-20:
**I'm submitting a bug report:
 
 **Current behavior**
read requests return crc errors or "Error: Data length error, expected y got x" with x < y. this problem does infrequently occur when requesting 1-2 registers, occurs about 50% of the time for 3-4 registers and almost always occurs when requesting more than 5 registers

```
`DEBUG=contribModbus*,modbus-serial node-red -v`

`20 Jul 22:17:35 - [info] Starting modified flows
20 Jul 22:17:35 - [info] Started modified flows
contribModbus:getter Add Message 5b524379bacbd8340b88b25f +0ms
contribModbus:getter {"topic":"213ce2cc.c9424e","payload":{"value":1532117881177,"unitid":"2","fc":3,"address":"696","quantity":"2","messageId":"5b524379bacbd8340b88b25f"},"_msgid":"11d6741e.5a34fc","queueUnitId":2} +87ms
contribModbus:core:client Client -> event: activate old: READING new: ACTIVATED Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 2 +2m
contribModbus:core Remove Message In:<5b524379bacbd8340b88b25f> Out:<5b524379bacbd8340b88b25f> +0ms
contribModbus:getter Data length error, expected 9 got 1 +95ms
20 Jul 22:18:01 - [error] [modbus-getter:213ce2cc.c9424e] Error: Data length error, expected 9 got 1
    at SerialPort.<anonymous> (/home/deploy/.node-red/node_modules/node-red-contrib-modbus/node_modules/modbus-serial/index.js:246:42)
    at emitOne (events.js:96:13)
    at SerialPort.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:176:18)
    at SerialPort.Readable.push (_stream_readable.js:134:10)
    at binding.read.then (/home/deploy/.node-red/node_modules/serialport/lib/serialport.js:378:10)
  modbus-serial modbusErrorHandling:Data length error, expected 9 got 1 +2m
  contribModbus:getter Add Message 5b524384bacbd8340b88b260 +11s
  contribModbus:getter {"topic":"213ce2cc.c9424e","payload":{"value":1532117892002,"unitid":"2","fc":3,"address":"696","quantity":"2","messageId":"5b524384bacbd8340b88b260"},"_msgid":"44647cb6.dbca14","queueUnitId":2} +30ms
  contribModbus:core:client Client -> event: activate old: READING new: ACTIVATED Serial@/dev/ttyUSB0:9600bit/s default Unit-Id: 2 +11s
  contribModbus:core Remove Message In:<5b524384bacbd8340b88b260> Out:<5b524384bacbd8340b88b260> +11s
```
`
 **Expected behavior**
node should return the requested register. modpoll on command line (same system) works as expected:
 
```
$ mbpoll -m rtu -a 2 -r 697 -c 20 -1 -o 5 -t 4 /dev/ttyUSB0 -b 9600 -d 8 -s 1 -P none -v
debug enabled
Set device=/dev/ttyUSB0
mbpoll 1.0-0 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright © 2015-2018 Pascal JEAN, https://github.com/epsilonrt/mbpoll
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'mbpoll -w' for details.
 
Opening /dev/ttyUSB0 at 9600 bauds (N, 8, 1)
Set response timeout to 5 sec, 0 us
Protocol configuration: Modbus RTU
Slave configuration...: address = [2]
                        start reference = 697, count = 20
Communication.........: /dev/ttyUSB0,       9600-8N1
                        t/o 5.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table
 
-- Polling slave 2...
[02][03][02][B8][00][14][C4][6B]
Waiting for a confirmation...
<02><03><28><00><B7><00><00><00><B7><00><00><80><00><00><00><00><A5><00><00><01><0F><00><00><02><02><00><00><05><02><00><00><4C><02><00><00><4C><04><00><00><4C><04><00><00><16><EF>
[697]:  183
[698]:  0
[699]:  183
[700]:  0
[701]:  32768 (-32768)
[702]:  0
[703]:  165
[704]:  0
[705]:  271
[706]:  0
[707]:  514
[708]:  0
[709]:  1282
[710]:  0
[711]:  19458
[712]:  0
[713]:  19460
[714]:  0
[715]:  19460
[716]:  0
```
`
## Your Environment
   
Node-RED version: v0.17.5
Node.js  version: v6.12.1
Linux 4.9.59+ arm LE
node-red-contrib-modbus: v2.5.0

upgrading also did not help:

Node-RED version: v0.18.7
Node.js  version: v8.11.3
Linux 4.9.59+ arm LE
node-red-contrib-modbus: v3.6.1   

* Description of the connecting Modbus system:

AC-System from Kampman. connecting to it from the cmd-line as well as from a different windows system works as expected.

demetz, 2018-07-27:
as a brief follow-up: problem seems to be due to node-red-contrib-modbus. Using node-red-contrib-modbus-serial, I was able to retrieve the values as expected.

biancode, 2018-08-01:
@demetz  could you please try with the new version of [node-modbus-serial](https://github.com/yaacov/node-modbus-serial), please? You'll find [here some examples](https://github.com/yaacov/node-modbus-serial/tree/master/examples) to work via scripts.

biancode, 2018-08-01:
If you got the same problem, then please [report it](https://github.com/yaacov/node-modbus-serial/issues) to the lib developer.

demetz, 2018-08-02:
@biancode 
thanks for the hint. this [thread](https://github.com/yaacov/node-modbus-serial/issues/25) pointed me in the right direction: the problem was solved by using RTU buffered... 

thanks again!

closed 2018-08-02 by demetz