BiancoRoyal/node-red-contrib-modbus #102: Modbus flex Write exception 4

esbayonat, 2018-07-26:
**I'm submitting a ...**  (check one with "x")

### Problem
 - ( ) bug report => please search github for a similar issue or PR before submitting
 - (x) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
I'm trying to do a commissioning/decommissioning procedure, for this, I start remote configuration with the sologate, add the device, wait to read it and the I start remote configuration again and remove it, I do this in a loop with 4 sologates at the same time (I commission the sensor for each sologate) an 2 sensors. The flow works for around an hour and suddenly the modbus flex write node sends "error modbus exception 4", this happens when I write the instruction to remove the device. As I am using 4 sologates, I change the IP constantly so I use the modbus flex connector node to change the server.
 
 **Expected behavior**
 
 I expect the program to run permanently 24/7.

 **Minimal reproduction of the problem with instructions**
 
   1. I start the flow which updates the information read from one sensor on the 4 sologates in a table every minutes.
   2. I let it run for around an hour (55 min)
   3. The flow stops due to modbus flex write error modbus exception 4.
   
 ## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 
 
 
 ## Support
 - support request => **Please do not submit support requests here**, instead see use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer)
 - (x) simple question => please visit the Wiki and Info tab by clicking on the node inside Node-RED before

## Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got modbus-serial installed with it
  * ( ) I have installed modbus-serial global from source (using git clone)
  * (x) I have installed modbus-serial as a global package (using npm i -g)
        
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - (x) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : ZBRN2
     - Version: 
     - Vendor : Schneider electric
     - Link   : https://www.schneider-electric.us/en/product/ZBRN2/access-point---2-rj45-24..240v-ac-dc-4-displays-5-leds-modbus-wireless-radio/
     
    
* **node-red-contrib-modbus version:** ``

* **Node:** (min. 8.x)
   `node --version` = 8.11.3

sometimes I'm using v6.11.2

biancode, 2018-08-01:
@esbayonat here the same: could you please try with the new version of [node-modbus-serial](https://github.com/yaacov/node-modbus-serial), please? You'll find [here some examples](https://github.com/yaacov/node-modbus-serial/tree/master/examples) to work via scripts.

esbayonat, 2018-08-02:
@biancode I forgot to mention, I'm using ethernet to communicate with the modbus

biancode, 2018-08-06:
reopen with an answer, please

closed 2018-08-06 by biancode

esbayonat, 2018-08-07:
Hello, sorry I took my time to answer. I did the implementation with node-modbus-serial and it works perfectly! thank you