BiancoRoyal/node-red-contrib-modbus #103: Failed to load Modbus components. SyntaxError Block-scoped declarations not yet supported outside strict mode

cfb95, 2018-07-29:
### Problem
 - ( ) bug report => please search github for a similar issue or PR before submitting
 - (x) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 Node-Red show error messages: 
[image]
And components (nodes) are not loaded into palette

node-red-contrib-modbus@3.6.1
Node-RED version: v0.18.7
Node.js  version: v8.11.3
Ubuntu: Linux 4.4.0-124-generic x64 LE

Installed with:
sudo npm install -g node-red-contrib-modbus --unsafe-perm --build-from-source

Thanks.

biancode, 2018-08-01:
I had the problem too, but now after updating npm 6.2.0 and nodejs v8.11.3, a clean install of node-red, and delete of old folders, it works well again. Looks like an upgrade problem of Node-RED v0.18.7 for me.

closed 2018-08-01 by biancode

cfb95, 2018-08-02:
Thanks! 
I've uninstalled, deleted folders and reinstalled, this time with npm 6.3.0
but problem persists on my installation.
I will try with a fresh virtualmachine next time to see what happen. meanwhile, the serial-modbus is ok for me.