BiancoRoyal/node-red-contrib-modbus #104: Feature request: server mode for solving CGNAT issues on GPRS IoT devices

mainakae, 2018-08-02:
**I'm submitting a ...**  (check one with "x")

### Problem
 - ( ) bug report => please search github for a similar issue or PR before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 
 **Expected behavior**
 
 
 **Minimal reproduction of the problem with instructions**
 
   1. 
   2. 
   3. 
   
 ## Feature
 - (X) feature request
 
I'm currently working on a IoT project involving ModBus TCP communications with remote low power devices connected via GPRS. Those devices are in a CGNAT (Carrier Grade NAT) network, which means that the assigned IP of the device is not accessible from the internet. The only solution (and the one we've previously used and is more or less the standard solution in related forums) we have come up with, is to start the communications from the device (as if it were a client interrogating our central server) and then re-use that channel from the server to interrogate the device.

It's doable from regular code, and (I think) a common enough problem to be considered a nice feature to have in this project.

The main caveats are:
- non standard solution (albeit commonly used)
- keeping alive the connection: forces us to send regular "heartbeats" via exceptions (whose codes may even be used to notify changes in the device, and thus trigger an update from the service)
 
In case you are interested in adding this functionality to the project, I can devote some of my time to do so (If, again, you are so inclined too) and publish it via pull-request.

Thanks for your time and for such a great node-red module!!
Pedro.
 
 ## Support
 - support request => **Please do not submit support requests here**, instead see use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer)
 - ( ) simple question => please visit the Wiki and Info tab by clicking on the node inside Node-RED before

## Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got modbus-serial installed with it
  * ( ) I have installed modbus-serial global from source (using git clone)
  * ( ) I have installed modbus-serial as a global package (using npm i -g)
        
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
     
    
* **node-red-contrib-modbus version:** ``

* **Node:** (min. 8.x)
   `node --version` =

biancode, 2018-08-06:
Hi Pedro,

you're welcome to send your pull request.
please read the [contributing.md](https://github.com/biancode/node-red-contrib-modbus/blob/master/.github/CONTRIBUTING.md)

biancode, 2018-08-06:
please, try to write tests for the new code ...

biancode, 2018-08-10:
please, work on branch develop - many changes are there - to fit into later releases

biancode, 2018-08-14:
reopen with your pull request

closed 2018-08-14 by biancode