BiancoRoyal/node-red-contrib-modbus #105: Don't connect client that are on disabled nodes

robertsLando, 2018-08-02:
If there aren't any node (or even if there are node but them are in a disabled flow) using a specific client configuration that client configuration should not try to connect to modbus.

This has caused me lot issues because I was using other nodes connecting to the same serial port and I was not able to understand why the connection couldn't be established after I checked there was a modbus client node in a disabled flow that was tring to connect to the same serial port.

biancode, 2018-08-06:
@robertsLando did you discuss that in the Node-RED community?

hardillb, 2018-08-06:
This is not a Node-RED problem.

It is up to the config node to keep track of if it is being used and react appropriately

biancode, 2018-08-06:
@hardillb thank you!

biancode, 2018-08-06:
@hardillb  maybe the community could add some hints about that in the [dev docs](https://nodered.org/docs/creating-nodes/config-nodes), please!

biancode, 2018-08-06:
@robertsLando was the flow just to use with the disabled flow or global?

robertsLando, 2018-08-06:
I have discuss this on slack channel because I have developed a node that handles serial and TCP connections like the modbus node. They only told me that this should be self checked by nodes developers because some configuration nodes may need to be inited even if there are no nodes using them, they also said that the mqtt node works in this way but I don't know how to check if a node is in a disabled flow or not

biancode, 2018-08-06:
[Slack protocol](https://node-red.slack.com/archives/C03M2TAQ8/p1533578471000276)

biancode [8:01 PM]
Hi, hardillb posted here https://github.com/biancode/node-red-contrib-modbus/issues/105, that I've to handle the enable or disable of config nodes - but what is the best way to handle? We have three scenarios:
1) config node is global for all flows - some are enabled and some disabled
2) config node is addressed to the flow and the flow is disabled
3) config node is unused (edited)

knolleary [8:09 PM]
There is really only two scenarios: the config node has active users or it doesn't.
If it has no active users, it shouldn't do anything

biancode [8:13 PM]
OK, is there some API like inUse() to get this information inside a config node?

knolleary [8:14 PM]
No. The model we've used elsewhere, such as the mqtt nodes, is for the config node to expose a register/unregister pair of functions that flow nodes can call when they are instantiated and closed.
So the config node can track nodes coming and going.
It's an area we could do more to help in the platform, but this is what we have now

biancode [8:18 PM]
that means - config nodes are not to switch off or to get the information to switch off, doesn't it? (edited)

knolleary [8:19 PM]
Not sure what you mean. They can 'switch off' when the last node unregisters itself.
As I mentioned, the mqtt nodes do this; the broker config node only connects when there are in/out nodes actively using it.

biancode [8:21 PM]
I got it. :wink: Thank you!
I see register and deregister

biancode, 2018-08-14:
working with coming v4.0.0 and above

closed 2018-08-14 by biancode