BiancoRoyal/node-red-contrib-modbus #106: modbus-write memory / garbage collector problems

jinx17, 2018-08-04:
### Problem
 - (X) bug report => please search github for a similar issue or PR before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
I am using: 
- Raspberry PI Zero W
- connected to Modbus TCP (WIFI) Relay modules.
- DietPI SW package is installed and updated
- I tried with Node.js version 10.8.0 (latest) and 8.11.3. (LTS) --> same behaviour
- NPM version 5.6.0
- Node-red version v0.18.4 (latest)

**A) - PROBLEM WITH MODBUS-WRITE** 
I experience severe memory leak with modbus-write (maybe also modbus-read - not confirmed), which leads to node-red crash (in about 12 hours, using 2 modbus-write clients to 2 different modbus servers). I tried also with startup options "--max_old_space_size=128". 
I tried with startup command node-red and with node-red-pi.
At node-red start there is around 60-70MB memory used, after some hours memory usage is maxed out (481MB = max). Memory rises consistantly. Garbage collector tries to free memory but unsuccessful.

Latest node-red debug info:
[16891:0x19530a0]   25618339 ms: Mark-sweep 121.3 (165.6) -> 121.3 (165.6) MB, 3628.6 / 0.1   ms  allocation failure GC in old space requested
[16891:0x19530a0] 25625176 ms: Mark-sweep   121.3 (165.6) -> 121.3 (132.6) MB, 6836.4 / 0.1 ms  last resort GC in old space requested
[16891:0x19530a0] 25630109 ms: Mark-sweep   121.3 (132.6) -> 121.3 (132.6) MB, 4933.1 / 0.1 ms  last resort GC in old space requested

<--- JS stacktrace   --->--
 ==== JS stack trace   =========================================
 Security context: 0x397161d9   <JSObject>
1: set(this=0x3f5c1bf9 <Map map = 0x45d0546d>,0x2b21c1a1   <ObjectID map = 0x55068ee9>,0x2b21c355 <Object map = 0x5505bd91>)
2: /* anonymous */   [/usr/local/lib/node_modules/node-red-contrib-modbus/modbus/modbus-flex-getter.js:~1]   [pc=0xae70d958](this=0xb1210199 <JSObject>,o=0x2b21c355 <Object map   = 0x5505bd91>)
3: emitOne(aka emitOne) [events.js:~114]   [pc=0xae71b870](this=0x2d70417d <undef...

FATAL ERROR: CALL_AND_RETRY_LAST   Allocation failed - JavaScript heap out of memory

**B) NO PROBLEMS WITH MEMORY WHEN USING MODBUS-FLEX-WRITE**
I changed / replaced all modbus-read and modbus-write nodes with -flex- nodes. After 12 hours, memory is still below 128MB (I beleive garbage collector is doing its work - I saw memory around 120MB, and in 20 minutes, memory is released and now only around 80MB is used).

**C) PROBLEM? WHY I TRY TO USE MODBUS-WRITE INSTEAD OF MODBUS-FLEX-WRITE**
I found out modbus-read and modbus-write are much faster and much less errors appear to happen, and much less timeout also! 
With modbus wifi relay module I am using, modbus-flex-write is not consistent in writing commands to module,. Sometimes there is a delay in writing from node-red or accepting commands in the module. With modbus-write I see no such delays, and timeout happens rarely!
FOR EACH SERVER (WIFI RELAY MODULE) THERE IS ONLY ONE NODE TO READ STATES OF COILS (Fn01), AND ONE NODE TO WRITE COILS TO THE MODULE (Fn15). Both using 16 bits.

 **Expected behavior**
No memory leaks, or garabge collector proper functioning.
 
 **Minimal reproduction of the problem with instructions**
 
   1. I sent two files / flows for revision.
   2. I can send printscreens of memory behaviour
[Modbus-write FLOW - MEMORY PROBLEMS.txt](https://github.com/biancode/node-red-contrib-modbus/files/2259290/Modbus-write.FLOW.-.MEMORY.PROBLEMS.txt)
[Modbus-flex-write FLOW - no memory problems.txt](https://github.com/biancode/node-red-contrib-modbus/files/2259291/Modbus-flex-write.FLOW.-.no.memory.problems.txt)

   3. I can help debugging - you need to lead me how to debug
   
 ## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 
 
 
 ## Support
 - support request => **Please do not submit support requests here**, instead see use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer)
 - ( ) simple question => please visit the Wiki and Info tab by clicking on the node inside Node-RED before

## Your Environment
**Please tell us about your environment:**

  * (X) I am using just the Node-RED package and got modbus-serial installed with it
  * ( ) I have installed modbus-serial global from source (using git clone)
  * ( ) I have installed modbus-serial as a global package (using npm i -g)
        
  * Device :
    
    - ( ) macOS: v 
    - () Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - (X) Other: Raspberry PI Zero W
         - Name: DietPI (based on Raspbian Stretch if I understand correctly)
         - Version: latest, v6.12
   
   * Description of the connecting Modbus system:
   
     - Name   : Denkovi Modbus WIFI Relay board
     - Version: latest
     - Vendor : Denkovi
     - Link   : https://
     
    
* **node-red-contrib-modbus version:** ``
     - latest = 3.6.1

* **Node:** (min. 8.x)
   `node --version` =tried with v10.8.0 and v8.11.3 (LTS)

biancode, 2018-08-06:
Please, check the node.bufferMessageList - if you did not get any response for messages the list will grow up on errors

jinx17, 2018-08-07:
Currently I am using only -flex- nodes. How / where can I check this?
I looked for a file with this name but there is none found in my RPI.

jinx17, 2018-08-09:
I am sorry, accidentaly closed this issue. Reopened again.

biancode, 2018-08-09:
Please, read the Wiki and send me detailed logs from that nodes! I'm running since the last release with those nodes on PI1, PI2 and PI3 without any problem or memory leaks.

biancode, 2018-08-09:
Please, read other issues of [memory leaks](https://github.com/biancode/node-red-contrib-modbus/issues?utf8=%E2%9C%93&q=label%3A%22memory+leak%22+)

biancode, 2018-08-14:
Please, reopen with more details.

closed 2018-08-14 by biancode