BiancoRoyal/node-red-contrib-modbus #107: Help on reading multiple holding registers

abhilashkm1988, 2018-08-08:
Hai , 

 Am new to modbus programming. I have an issue regarding values from multiple holding registers. 
My idea is get the starting address and length to read from db. And pass it to a function node for iterate through all the rows, then send the payload to modbus flex getter using node.send(payload). Everything to this part is working well. But in my db I have around 50 -64 rows of address details. When i use node.send method from my function it reads the data correctly , but i cant track which data belongs to which address. Am I using the wrong way ? Please help a to solve this problem in a better way.

Thanks in advance.

biancode, 2018-08-08:
Hi @abhilashkm1988 ,

I'm sorry, but I did not see your problem.
Please, explain more in detail what the meaning of that little sentence is!
_
> When I use node.send method from my function it reads the data correctly , but i cant track which data belongs to which address.
_

biancode, 2018-08-10:
reopen with more information, please

closed 2018-08-10 by biancode