BiancoRoyal/node-red-contrib-modbus #109: CPU 100% after upgrade to v 4.0.0

jinx17, 2018-08-23:
Problem

    (X) bug report => please search github for a similar issue or PR before submitting
    ( ) unexpected behavior => please visit the Wiki before

Current behavior
I am using:

    Raspberry PI Zero W
    connected to Modbus TCP (WIFI) Relay modules.
    DietPI SW package is installed and updated
    NPM version 5.6.0
    Node-red version v0.18.4 (latest)

CPU USAGE 100%
After upgrading to version v4.0.0 I get allmost 100% CPU usage all the time. 
I reinstalled version v3.6.1 and CPU usage is around 25%.

Feature

    ( ) feature request

What is the motivation / use case for changing the behavior?
Support

    support request => Please do not submit support requests here, instead see use the link to our support page
    ( ) simple question => please visit the Wiki and Info tab by clicking on the node inside Node-RED before

Your Environment

Please tell us about your environment:

    (X) I am using just the Node-RED package and got modbus-serial installed with it

    ( ) I have installed modbus-serial global from source (using git clone)

    ( ) I have installed modbus-serial as a global package (using npm i -g)

    Device :
        ( ) macOS: v
        () Linux: v
        ( ) Windows: v
        ( ) Raspbian: v
        (X) Other: Raspberry PI Zero W
            Name: DietPI (based on Raspbian Stretch if I understand correctly)
            Version:  v6.12

    Description of the connecting Modbus system:
        Name : Denkovi Modbus WIFI Relay board
        Version: latest
        Vendor : Denkovi
        Link : https://

    node-red-contrib-modbus version: ``
        latest = 4.0.0.

    Node: (min. 8.x)
    node --version = v8.11.3 (LTS)

biancode, 2018-08-23:
It is a major version. Please, check your flows and maybe you have to update or replace some nodes. Node-RED did not update all changes automatically. Otherwise, I need some logs or information what is the memory consumption about, please.

biancode, 2018-08-23:
It could help to get the flow.

biancode, 2018-08-23:
Please, test with nodejs v10 and node-RED v0.19!

jinx17, 2018-08-25:
Hi,
**I upgraded everything to latest:
node-red: v0.19.2
node.js: v10.9.0
node-red-contrib-modbus: v4.0.0**

My findings are:
1. I started fresh without any nodes. Created ONLY modbus-server node (both flex and simple), and everytime I create a modbus server on port 502 node-red crashes! 
When creating modbus server with port 10502 is working OK (no crash).

2. CPU usage is OK, WHEN NOT USING node-red modbus server.

3. when using modbus server in node-red and I am doing also reading/writing modbus from within node-red, CPU usage is high (100%) + memory leaks are high.

4. when creating modbus server on different - dislocated machine (not on RPI Zero, where node-red is operating), and I am using only reading/writing in node-red to dislocated modbus server, CPU usage is OK (avg 15%), also I dont see high memory leaks at the moment.

All in all, I think there is a problem in node-red handling modbus server node (both flex and simple).
Best regards.

jinx17, 2018-08-25:
Just an update with running bash command: `netstat --tcp --listening --programs --numeric`
Checking to see if port 502 is reserved or used by another process. 

Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:1880            0.0.0.0:*               LISTEN      13104/node-red
tcp        0      0 0.0.0.0:1883            0.0.0.0:*               LISTEN      1074/mosquitto
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      474/vpnserver
tcp        0      0 0.0.0.0:222             0.0.0.0:*               LISTEN      314/sshd
tcp        0      0 0.0.0.0:992             0.0.0.0:*               LISTEN      474/vpnserver
tcp        0      0 0.0.0.0:1443            0.0.0.0:*               LISTEN      474/vpnserver
tcp        0      0 0.0.0.0:1194            0.0.0.0:*               LISTEN      474/vpnserver
tcp        0      0 0.0.0.0:5555            0.0.0.0:*               LISTEN      474/vpnserver
tcp6       0      0 :::1883                 :::*                    LISTEN      1074/mosquitto
tcp6       0      0 :::443                  :::*                    LISTEN      474/vpnserver
tcp6       0      0 :::222                  :::*                    LISTEN      314/sshd
tcp6       0      0 :::992                  :::*                    LISTEN      474/vpnserver
tcp6       0      0 :::1443                 :::*                    LISTEN      474/vpnserver
tcp6       0      0 :::1194                 :::*                    LISTEN      474/vpnserver
tcp6       0      0 :::5555                 :::*                    LISTEN      474/vpnserver

biancode, 2018-08-25:
502 is a well known port on Unix / Linux and is just to use with root or sudo running node-red - to use sudo for node-red means every user on website of node-red is sudo/root

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

biancode, 2018-08-25:
Is the point 4 also v4.0.0 in you listing?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

biancode, 2018-08-25:
I’ll test it with PIW in the next few days.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

jinx17, 2018-08-25:
Yes, everything on v4.0.0

Before upgrade I could use port TCP 502 for modbus server without any problem!

biancode, 2018-08-25:
25 Aug 11:35:42 - [red] Uncaught Exception:
25 Aug 11:35:42 - Error: listen [EACCES](http://man7.org/linux/man-pages/man3/errno.3.html) 0.0.0.0:502

biancode, 2018-08-25:
> when using modbus server in node-red and I am doing also reading/writing modbus from within node-red, CPU usage is high (100%) + memory leaks are high.

@jinx17 could you find a memory leak or is it just a guess?

jinx17, 2018-08-25:
Error when crash occurs:
24 Aug 20:41:30 - [red] Uncaught Exception:
24 Aug 20:41:30 - Error: listen EACCES 0.0.0.0:502
    at Server.setupListenHandle [as _listen2] (net.js:1317:19)
    at listenInCluster (net.js:1382:12)
node-red.service: Main process exited, code=exited, status=1/FAILURE

jinx17, 2018-08-25:
Memory leak is educated guess. I find memory rises very quickly over an hour or two from around 100MB to 160 or more. The same behaviour where memory rises from 100MB to 450MB in about 12 hours was observed all the time this week (so it is experience with same settings but before upgrade).

Now, when I am using dislocated modbus server, memory usage is pretty stable around 110MB for 3 hours now.

biancode, 2018-11-08:
@jinx17 some more news here?

closed 2018-11-08 by biancode

jinx17, 2018-11-09:
Hi Klaus,
I had to fund another sollution - using modbus server on another device. It
was simply too many problems and instabillity with CPU usage and memory
leaks. I am only using only modbus client on PI-zero now, and its working
good.

BR,
Gregor

-- 
Lep pozdrav,
Gregor Jenko
031/77.55.33
-----------------------------
TechnoSoft d.o.o.
Lipa 22
3320 Velenje
t: 03/897.5000
www.technosoft.si