BiancoRoyal/node-red-contrib-modbus #118: Node red Docker doesn't connect

rjbaat, 2018-12-17:
Hi, 
I am trying to get node-red-contrib-modbus to work with /dev/ttyAMA0 on a Rpi.
I got the connection working with version: npm install node-red-contrib-modbus@4.0.0 on node red directly on the Rpi. But not with the node red docker image. 

Then i get messages like:
"FSM Reset On State INIT"
"Client -> reconnect in 2000 ms Serial@/dev/ttyAMA0:19200bit/s default Unit-Id: 1"
 I started the node red docker container with:

docker run -d \
--name node-red \
--privileged \
-p 1880:1880 \
-v /home/pi/apps/node-red:/data \
--restart unless-stopped \
nodered/node-red-docker:rpi

after this i tried to install the packege both via cmd line (npm install node-red-contrib-modbus@4.0.0) and also the latest version from the manage pallete. 

Can you help me out?

rjbaat, 2018-12-17:
I have also tried to get acces to the serial with by adding:
--device /dev/ttyAMA0:/dev/ttyAMA0

```
docker run -d 
--name node-red 
--device /dev/ttyAMA0:/dev/ttyAMA0
--privileged 
-p 1880:1880 
-v /home/pi/apps/node-red:/data 
--restart unless-stopped 
nodered/node-red-docker:rpi
```

biancode, 2018-12-17:
Hi, you start at a very low level to test with the [lib](https://github.com/yaacov/node-modbus-serial) first. Use the examples from the lib to test your port settings. If that is able to connect, then Node-RED will work too. Bes ure to open just one connection to your serial port ;)

Happy connecting!
Klaus

closed 2018-12-17 by biancode