BiancoRoyal/node-red-contrib-modbus #120: read/write too many registers crashes red from Modbus Server node (buffer overrun) #35v2

joshuatreenl, 2018-12-26:
## Problem
Still have the same problem as described in issue #35 / #46.  
 
 **Current behavior**
see issue #35 or #46 
- Used Node red code: 
[{"id":"1704df9a.25c2f","type":"tab","label":"Flow 1"},{"id":"739c2f41.070e38","type":"debug","z":"1704df9a.25c2f","name":"","active":false,"console":"false","complete":"true","x":570,"y":140,"wires":[]},{"id":"896a1e0d.eb9b38","type":"modbus-server","z":"1704df9a.25c2f","name":"ModbusServer_Local502","logEnabled":false,"serverPort":"502","responseDelay":"100","delayUnit":"ms","coilsBufferSize":"0","holdingBufferSize":1024,"inputBufferSize":"0","x":327,"y":142,"wires":[["739c2f41.070e38"],["739c2f41.070e38"],["739c2f41.070e38"]]},{"id":"36c4a054.3748f","type":"inject","z":"1704df9a.25c2f","name":"","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":49.99998474121094,"y":144,"wires":[["896a1e0d.eb9b38"]]}]
    
More info: I installed version 2.5.0 of the modbus package because the node version I am using is 6.12.3. 

### Your Environment
**Please tell us about your environment:**

  * Device :
    - ( ) Other:
         - Name: Siemens
         - Version: IOT2040 - image v2.2.0
   
   * Description of the connecting Modbus system:
   
     - Name   : Modbus Doctor
     - Version: 2.6
     - Vendor : KScada
   
* **node-red-contrib-modbus version:** ``

* **Node:** (min. 6.x)
   `node --version` = 6.12.3
   `node red --version` = 0.16.2
   `modbus package` = v2.5.0 (npm install node-red-contrib-modbus@2.5.0) 

Regards,
Rene

biancode, 2018-12-26:
Hi Rene, feel free to send a pull request with the fix.

joshuatreenl, 2018-12-26:
> 
> 
> Hi Rene, feel free to send a pull request with the fix.

Hi biancode, thanks for the quick response. But... what do you mean with 'a pull 'request'

biancode, 2018-12-26:
https://github.com/yaacov/node-modbus-serial/issues?q=is%3Aissue+is%3Aclosed

biancode, 2018-12-26:
client:  https://github.com/yaacov/node-modbus-serial
server: https://github.com/Cloud-Automation/node-modbus

biancode, 2018-12-26:
old server issue to https://github.com/Cloud-Automation/node-modbus

joshuatreenl, 2018-12-26:
In the server I defined for example 1024 holding registers. The reading part of the software is Modbus Doctor, in that software I read more than 127 registers. Of course this is not the real software I am using, but I want to check if the software is ruggedized. 

It seems to be solved at version >v4.1.0 of the modbus package. Unfortunately I can not use that version, because I then encounter other problems in combination with version of node I use. So I have to find a solution to make an image myself with the new node software.

biancode, 2018-12-26:
same at the actual state ...

26 Dec 12:56:12 - [red] Uncaught Exception:
26 Dec 12:56:12 - RangeError [ERR_OUT_OF_RANGE]: The value of "value" is out of range. It must be >= 0 and <= 255. Received 260
    at writeU_Int8 (internal/buffer.js:564:11)
    at Buffer.writeUInt8 (internal/buffer.js:574:10)
    at ReadHoldingRegistersResponseBody.createPayload (/Users/klaus/node_modules/jsmodbus/src/response/read-holding-registers.js:89:15)
    at ModbusTCPResponse.createPayload (/Users/klaus/node_modules/jsmodbus/src/tcp-response.js:113:16)
    at ModbusServerResponseHandler.handle (/Users/klaus/node_modules/jsmodbus/src/modbus-server-response-handler.js:69:30)
    at ModbusServerClient._onData (/Users/klaus/node_modules/jsmodbus/src/modbus-server-client.js:39:29)
    at Socket.emit (events.js:189:13)
    at addChunk (_stream_readable.js:288:12)
    at readableAddChunk (_stream_readable.js:269:11)
    at Socket.Readable.push (_stream_readable.js:224:10)

joshuatreenl, 2018-12-26:
> same at the actual state ...
>
> 26 Dec 12:56:12 - [red] Uncaught Exception:
> 26 Dec 12:56:12 - RangeError [ERR_OUT_OF_RANGE]: The value of "value" is out of range. It must be >= 0 and <= 255. Received [...]

I have: 
26 Dec 09:49:49 - [red] Uncaught Exception:
26 Dec 09:49:49 - TypeError: "value" argument is out of bounds
    at checkInt (buffer.js:1041:11)
    at Buffer.writeUInt8 (buffer.js:1089:5)
    at Object.<anonymous> (/usr/lib/node_modules/node-red-contrib-modbus/node_modules/node-modbus/node-modbus/handler/server/ReadHoldingRegisters.js:58:14)
    at ontimeout (timers.js:386:11)
    at tryOnTimeout (timers.js:250:5)
    at Timer.listOnTimeout (timers.js:214:5)

biancode, 2018-12-26:
1024 (buffer size) / 8 = 128  ( 0 ... 127 registers) in the old version is what you get

joshuatreenl, 2018-12-26:
> 
> 
> 1024 (buffer size) / 8 = 128 ( 0 ... 127 registers) in the old version is what you get

Oh boy... I didn't realize that I should divide it by 8...

biancode, 2018-12-26:
TypeError is not to catch in Node-RED - let me know if you find a way to do so ;)

biancode, 2018-12-26:
If you read the source code, then you will see a whole try-catch block but it does not catch TypeError.

joshuatreenl, 2018-12-26:
I am installing version 4.1.1 now. I will see what will happen if I do the same actions.

joshuatreenl, 2018-12-26:
serial module v7.4.2 is installed. 

`-- node-red-contrib-modbus@4.1.1
  +-- bson@3.0.2
  +-- debug@4.1.0
  | `-- ms@2.1.1
  +-- jsmodbus@3.1.0
  | +-- crc@3.4.0
  | `-- debug@3.2.6
  +-- line-by-line@0.1.6
  +-- modbus-serial@7.4.2
  | +-- debug@3.2.6
  | `-- serialport@6.2.2
  |   +-- @serialport/parser-byte-length@1.0.5
  |   +-- @serialport/parser-cctalk@1.0.5
  |   +-- @serialport/parser-delimiter@1.0.5
  |   +-- @serialport/parser-readline@1.0.5
  |   +-- @serialport/parser-ready@1.0.5
  |   +-- @serialport/parser-regex@1.0.5
  |   `-- prebuild-install@4.0.0
  +-- serialport@7.0.2
  | +-- @serialport/binding-mock@2.0.3
  | | `-- @serialport/binding-abstract@2.0.3
  | +-- @serialport/bindings@2.0.4
  | | +-- @serialport/parser-readline@2.0.2
  | | | `-- @serialport/parser-delimiter@2.0.2
  | | `-- prebuild-install@5.2.2
  | |   +-- expand-template@2.0.3
  | |   `-- napi-build-utils@1.0.1
  | +-- @serialport/parser-byte-length@2.0.2
  | +-- @serialport/parser-cctalk@2.0.2
  | +-- @serialport/parser-delimiter@2.0.2
  | +-- @serialport/parser-readline@2.0.2
  | +-- @serialport/parser-ready@2.0.2
  | +-- @serialport/parser-regex@2.0.2
  | +-- @serialport/stream@2.0.3
  | +-- debug@3.2.6
  | `-- prompt-list@3.2.0
  |   +-- ansi-cyan@0.1.1
  |   | `-- ansi-wrap@0.1.0
  |   +-- ansi-dim@0.1.1
  |   `-- prompt-radio@1.2.1
  |     +-- debug@2.6.9
  |     `-- prompt-checkbox@2.2.0
  |       +-- debug@2.6.9
  |       | `-- ms@2.0.0
  |       `-- prompt-base@4.1.0
  |         +-- component-emitter@1.2.1
  |         +-- debug@3.2.6
  |         +-- koalas@1.0.2
  |         +-- log-utils@0.2.1
  |         | +-- ansi-colors@0.2.0
  |         | | +-- ansi-bgblack@0.1.1
  |         | | +-- ansi-bgblue@0.1.1
  |         | | +-- ansi-bgcyan@0.1.1
  |         | | +-- ansi-bggreen@0.1.1
  |         | | +-- ansi-bgmagenta@0.1.1
  |         | | +-- ansi-bgred@0.1.1
  |         | | +-- ansi-bgwhite@0.1.1
  |         | | +-- ansi-bgyellow@0.1.1
  |         | | +-- ansi-black@0.1.1
  |         | | +-- ansi-blue@0.1.1
  |         | | +-- ansi-bold@0.1.1
  |         | | +-- ansi-gray@0.1.1
  |         | | +-- ansi-green@0.1.1
  |         | | +-- ansi-grey@0.1.1
  |         | | +-- ansi-hidden@0.1.1
  |         | | +-- ansi-inverse@0.1.1
  |         | | +-- ansi-italic@0.1.1
  |         | | +-- ansi-magenta@0.1.1
  |         | | +-- ansi-red@0.1.1
  |         | | +-- ansi-reset@0.1.1
  |         | | +-- ansi-strikethrough@0.1.1
  |         | | +-- ansi-underline@0.1.1
  |         | | +-- ansi-white@0.1.1
  |         | | +-- ansi-yellow@0.1.1
  |         | | `-- lazy-cache@2.0.2
  |         | |   `-- set-getter@0.1.0
  |         | |     `-- to-object-path@0.3.0
  |         | +-- error-symbol@0.1.0
  |         | +-- info-symbol@0.1.0
  |         | +-- log-ok@0.1.1
  |         | +-- success-symbol@0.1.0
  |         | +-- time-stamp@1.1.0
  |         | `-- warning-symbol@0.1.0
  |         +-- prompt-actions@3.0.2
  |         | `-- debug@2.6.9
  |         |   `-- ms@2.0.0
  |         +-- prompt-question@5.0.2
  |         | +-- clone-deep@1.0.0
  |         | | +-- for-own@1.0.0
  |         | | | `-- for-in@1.0.2
  |         | | +-- is-plain-object@2.0.4
  |         | | +-- kind-of@5.1.0
  |         | | `-- shallow-clone@1.0.0
  |         | |   +-- kind-of@5.1.0
  |         | |   `-- mixin-object@2.0.1
  |         | |     `-- for-in@0.1.8
  |         | +-- debug@3.2.6
  |         | +-- define-property@1.0.0
  |         | | `-- is-descriptor@1.0.2
  |         | |   +-- is-accessor-descriptor@1.0.0
  |         | |   | `-- kind-of@6.0.2
  |         | |   +-- is-data-descriptor@1.0.0
  |         | |   | `-- kind-of@6.0.2
  |         | |   `-- kind-of@6.0.2
  |         | +-- isobject@3.0.1
  |         | +-- kind-of@5.1.0
  |         | `-- prompt-choices@4.1.0
  |         |   +-- arr-swap@1.0.1
  |         |   | `-- is-number@3.0.0
  |         |   +-- choices-separator@2.0.0
  |         |   | `-- debug@2.6.9
  |         |   |   `-- ms@2.0.0
  |         |   +-- clone-deep@4.0.1
  |         |   | `-- shallow-clone@3.0.0
  |         |   +-- collection-visit@1.0.0
  |         |   | +-- map-visit@1.0.0
  |         |   | `-- object-visit@1.0.1
  |         |   +-- define-property@2.0.2
  |         |   +-- is-number@6.0.0
  |         |   +-- kind-of@6.0.2
  |         |   +-- pointer-symbol@1.0.0
  |         |   +-- radio-symbol@2.0.0
  |         |   +-- set-value@3.0.0
  |         |   +-- terminal-paginator@2.0.2
  |         |   | `-- debug@2.6.9
  |         |   |   `-- ms@2.0.0
  |         |   `-- toggle-array@1.0.1
  |         +-- readline-ui@2.2.3
  |         | +-- debug@2.6.9
  |         | | `-- ms@2.0.0
  |         | `-- string-width@2.1.1
  |         |   +-- is-fullwidth-code-point@2.0.0
  |         |   `-- strip-ansi@4.0.0
  |         |     `-- ansi-regex@3.0.0
  |         +-- readline-utils@2.2.3
  |         | +-- arr-flatten@1.1.0
  |         | +-- extend-shallow@2.0.1
  |         | | `-- is-extendable@0.1.1
  |         | +-- is-buffer@1.1.6
  |         | +-- is-number@3.0.0
  |         | | `-- kind-of@3.2.2
  |         | +-- is-windows@1.0.2
  |         | +-- mute-stream@0.0.7
  |         | +-- strip-color@0.1.0
  |         | `-- window-size@1.1.1
  |         |   `-- is-number@3.0.0
  |         `-- static-extend@0.1.2
  |           +-- define-property@0.2.5
  |           | `-- is-descriptor@0.1.6
  |           |   +-- is-accessor-descriptor@0.1.6
  |           |   | `-- kind-of@3.2.2
  |           |   +-- is-data-descriptor@0.1.4
  |           |   | `-- kind-of@3.2.2
  |           |   `-- kind-of@5.1.0
  |           `-- object-copy@0.1.0
  |             +-- copy-descriptor@0.1.1
  |             `-- define-property@0.2.5
  |               `-- is-descriptor@0.1.6
  |                 +-- is-accessor-descriptor@0.1.6
  |                 +-- is-data-descriptor@0.1.4
  |                 `-- kind-of@5.1.0
  +-- source-map-support@0.5.9
  | +-- buffer-from@1.1.1
  | `-- source-map@0.6.1
  `-- vm2@3.6.4

biancode, 2018-12-26:
https://github.com/node-red/node-red/issues?utf8=%E2%9C%93&q=TypeError+crash

joshuatreenl, 2018-12-26:
Indeed I see a lot of typeerror crashes. If I change the number of 'registers' in Node Red 1024 -> 8192. I've got the same error although the size of the buffer is Ok. I have to think about it what to do next. Thanks for your help!

joshuatreenl, 2018-12-26:
I was wondering why I didn't see this behaviour before. I realize that I was using the qmodModmaster software instead of the Modbus Doctor software. In qmodMaster it is not allowed to get more than 125 registers... Interesting.

biancode, 2018-12-26:
maybe ModbusDoctor did not check that and send a request ?

biancode, 2018-12-26:
I asked the core-team of node-red:

> Hi core-team, if there is some [red] Uncaught Exception from a third library - how could I handle this in a contrib node to stay alive and do not crash node-red? I have tried a try-catch around the third-lib-call and around the whole node, but the red.js event process.on(‘uncaughtException’ do an exit(1) …
> https://github.com/node-red/node-red/issues?utf8=%E2%9C%93&q=TypeError+crash
> every TypeError is a node-red crash, right?
> or exit(1)
> the third lib does not catch => uncaughtException - I do but never get it because of the exit(1) from the uncaughtException event
> it works as expected if I remove process.exit(1); - maybe it could check for TypeError or on which specific Error it has to exit(1) (edited)

biancode, 2018-12-26:
> hardillb [3:42 PM]
> Node-RED exits on uncaught exception because we have no way of knowing if it is possible to continue when a 3rd party library throws one. The correct thing to do is to fix the library to catch it's own errors and return an error (edited)

joshuatreenl, 2018-12-26:
I tried and it works. It's a workaround. I'll hope they will modify the code. 

26 Dec 16:40:06 - [info] Starting flows
26 Dec 16:40:06 - [info] Started flows
26 Dec 16:40:12 - [red] Uncaught Exception:
26 Dec 16:40:12 - TypeError: "value" argument is out of bounds
    at checkInt (buffer.js:1041:11)
    at Buffer.writeUInt8 (buffer.js:1089:5)
    at ReadHoldingRegistersResponseBody.createPayload (/usr/lib/node_modules/node-red-contrib-modbus/node_modules/jsmodbus/src/response/read-holding-registers.js:89:15)
    at ModbusTCPResponse.createPayload (/usr/lib/node_modules/node-red-contrib-modbus/node_modules/jsmodbus/src/tcp-response.js:113:16)
    at ModbusServerResponseHandler.handle (/usr/lib/node_modules/node-red-contrib-modbus/node_modules/jsmodbus/src/modbus-server-response-handler.js:69:30)
    at ModbusServerClient._onData (/usr/lib/node_modules/node-red-contrib-modbus/node_modules/jsmodbus/src/modbus-server-client.js:39:29)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:176:18)
    at Socket.Readable.push (_stream_readable.js:134:10)
26 Dec 16:40:13 - [red] Uncaught Exception:
26 Dec 16:40:13 - TypeError: "value" argument is out of bounds
    at checkInt (buffer.js:1041:11)
    at Buffer.writeUInt8 (buffer.js:1089:5)
    at ReadHoldingRegistersResponseBody.createPayload (/usr/lib/node_modules/node-red-contrib-modbus/node_modules/jsmodbus/src/response/read-holding-registers.js:89:15)
    at ModbusTCPResponse.createPayload (/usr/lib/node_modules/node-red-contrib-modbus/node_modules/jsmodbus/src/tcp-response.js:113:16)
    at ModbusServerResponseHandler.handle (/usr/lib/node_modules/node-red-contrib-modbus/node_modules/jsmodbus/src/modbus-server-response-handler.js:69:30)
    at ModbusServerClient._onData (/usr/lib/node_modules/node-red-contrib-modbus/node_modules/jsmodbus/src/modbus-server-client.js:39:29)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:176:18)
    at Socket.Readable.push (_stream_readable.js:134:10)

biancode, 2018-12-26:
https://node-red.slack.com/archives/C03M2TAQ8/p1545840558103700

biancode, 2018-12-26:
We could try to get the point to listen for errors of the library.

joshuatreenl, 2018-12-26:
> 
> 
> https://node-red.slack.com/archives/C03M2TAQ8/p1545840558103700

I think I have to be invited to join the node-red group? There is no "button" to contact the Workspace Administrator.

biancode, 2018-12-27:
https://stackoverflow.com/questions/50202507/nodejs-uncaught-exception-handling

biancode, 2018-12-27:
https://flaviocopes.com/node-exceptions/

joshuatreenl, 2018-12-27:
Hi Klaus,
Thanks for the links. I will examine what I can change to make the software robust. 
Rene