BiancoRoyal/node-red-contrib-modbus #122: modbus-noserial

emes, 2019-01-19:
## Feature
 - (X) feature request
 
 **What is the motivation / use case for changing the behavior?**
 
Avoiding the dreaded serialport dependancy.  Doing so allows for simpler, and much smaller images (docker, etc) since most of the smaller linux distros need to build serialport from source.  Many of us use modbus tcp-only.

I understand that the serialport dep is coming from modbus-serial.

However, yaacov is now providing versions without that dep.

Please see:

https://github.com/yaacov/node-modbus-serial/issues/249

and

https://www.npmjs.com/package/modbus-serial/v/7.5.1-no-serial-port

Any chance you can think of a way to provide the wonderful node-red-contrib-modbus while not inheriting the dependency?

biancode, 2019-01-24:
A CLI question answer while run install would be great ...

biancode, 2019-11-17:
The serialport package is optional, that should be enough to do not get in trouble on installation.

closed 2019-11-17 by biancode