BiancoRoyal/node-red-contrib-modbus #123: Modbus serial RTS support

Aleshus, 2019-01-20:
## Feature
 - ( x) feature request
 
 **What is the motivation / use case for changing the behavior?**
some RPi HATs need to change RTS pin when read or write to the RS485 bus. Can you add configuration support for that? Similar as the node-red-contrib-serialport-rs485 where autswitch RTS/DTR is implemented.

Thanks,
Aleš

wazza-b, 2019-07-19:
Encountered a need for this when using MAX485 based boards and also a Beaglebone 'RS485 CAN CAPE'. Ideally be able to define which gpio are used for DE/RE, and it's also possible to tie them together and use one pin. Happy to sponsor if I can figure out how...

biancode, 2019-11-17:
@wazza-b sponsoring could be done by lines of code the best, here on GitHub Sponsoring or here: [Sponsoring](https://bianco-royal.com/supporter.html)

closed 2022-09-29 by github-actions[bot]