BiancoRoyal/node-red-contrib-modbus #125: Queue messages by server rather than by unit-id

diogodtway, 2019-02-08:
**I'm submitting a ...**  (check one with "x")

## Feature
 - ( x) feature request
 
 **What is the motivation / use case for changing the behavior?**
 The current queue for a modbus server (either serial or tcp) only queues messages separately for each unit-id. While in tcp I think this is not an issue, in serial the channel is occupied regardless of the unit-id communicating.
The desired behavior would be for the requests to be queued by server rather than by unit-id. This way if I am pooling several addresses too fast the queue handles it and softens any peak concurrent requests.

Hope the text is clear enough. If not please reply with any questions.

Thank you in advance.

biancode, 2019-02-08:
I'm sorry it is not clear enough. Is there some real use case, please? Which "modbus server" (Node-RED, external, device ...)? Why should we organize the queue by server if every server has its own queue or client with a queue per unit-id inside node-red?

diogodtway, 2019-02-08:
Let me describe my setup:
I have a raspberry PI with a serial connection to a modbus bus.
In this bus I have 9 devices - with addresses 001-005 and 011-014
Every minute I make 3 requests for each address through a flex getter. This gives me a total of 27 requests per minute. 
The problem is that when I send a request to a different address the queue is ignored. A typical sequence would be:
 - Request 1 - Poll address 001 FC4 start at 0
 - Request 2 - Poll address 001 FC4 start at 300
 - Request 3 - Poll address 001 FC4 start at 400
At this point we have 1 request sent and 2 requests in queue. The problem starts here, when the following requests come in they ignore the 2 requests in queue because the address is different:
 - Request 4 - Poll address 002 FC4 start at 0
 - Request 5 - Poll address 002 FC4 start at 300
 - Request 6 - Poll address 002 FC4 start at 400
When request 4 goes in it doesn't wait for requests 2 and 3 to go first. It even ignores if the reply for request 1 as arrived or not.
What I would like is for the queue to be for the server (serial connection) and not a different queue for each address of slaves.

thatcadguy, 2019-03-19:
I haven't had a chance to use this with multiple devices yet (I will soon), but if this is true, this should be marked as a bug/defect. The messages have to be queued per serial port and not by anything else since only one device is allowed to talk at any one time.

thatcadguy, 2019-05-31:
@biancode What files govern the queuing behavior for serial? I may try to do a pull request as I'm about to need a fix for this. As I said, only one device can be speaking at a time on a serial line, so we have to queue for the whole serial port.

biancode, 2019-05-31:
@thatcadguy it is all inside connector based on an FSM

joearkay, 2019-12-04:
FYI - I too have come across this problem whilst deploying two Modbus RTU slave devices. I was hesitant to open an issue until some further feedback and research, but here's mine detailed: https://discourse.nodered.org/t/multiple-modbus-rtu-slaves-causes-error/18783

joearkay, 2019-12-05:
I’m considering at attempt at resolving this, as it looks like the assignee @biancode may be away.

biancode, 2019-12-08:
please, test with v5.2.x
new option on clients
[image]

biancode, 2019-12-14:
reopen or create a new issue if there is more

closed 2019-12-14 by biancode