BiancoRoyal/node-red-contrib-modbus #126: Stability issues, dev/tty/usb,  AMA0 not showing in the list

smithtekIOT, 2019-02-09:
Hello.

We have just updated to the latest version 4.1, When we first installed we could access the list of available serial devices in the drop down. For no unknown reason this list has vanished.
If i go into the configuration of the Node red serial node i can see the list of available devices, /dev/tty/USB, AMA0.

Additional to this there is also stability issues. Every couple of minutes the communications times out, this never used to happen before.

I have rolled back to version 3.6.1, this made no change.  
Andy advice or checks we can do ?
Thank you

 

**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 
 **Expected behavior**
 
 
 **Minimal reproduction of the problem with instructions**
 
   1. 
   2. 
   3. 
   
### Your Environment
**Please tell us about your environment:**

  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
     
    
* **node-red-contrib-modbus version:** ``

* **Node:** (min. 6.x)
   `node --version` =

smithtekIOT, 2019-02-10:
Hello Mate.
This is what i have discovered so far.

Not working version.
Node.js V11
Node red V0.19.5
Npm V3.10.10

Working version
Node.js 6.11.0
Node-red 0.19.5
Npm V3.10.10
serialport version 3.10.10
Working version matches that i have on a windows PC..
I'm reading registers from 3 devices using the same serial port dev/ttyAMA0

I'm using a round Robbin sequencer that polls 9 registers, grouped into bunches of 3, poll rate is 500ms
Time out setting is :1000ms
Reconnect setting is :2000ms

On the working version PC and RPIB+ they work no problems.
On the RPI3+ with node 11 it constantly;y has time out errors.

Im going to try with Node 8 as this is the recommended version to upgrade to.

Any ideas ?

biancode, 2019-12-08:
please, test with v5.2.x

closed 2019-12-14 by biancode