BiancoRoyal/node-red-contrib-modbus #128: Modbus Response node wont change status unless you redeploy the flow

smithtekIOT, 2019-02-17:
Hi 

I'm not sure if this is a bug or beyond my expectations.

**## Problem**
_Using Modbus Serial over LoRa._
I 'm using a  Core Node red "Status Node"  checking the status of a modbus response node.
The response node is connected to the output pin of Modbus getter node, Getter node is connected to Modbus read node, device "1"
If the modbus device "1" is disconnected from the bus the response node will display status "Initialized"
This is reflected in the status node as an object,  {text:"closed"} see below. However it will not update the status until you re-deploy the flow, Not practical in a unmanned scenario. I have left it for over an hour and it didn't pick up the changes to the modbus device, it did however reduce efficiency of the Modbus Bus causing the other devices to respond chaotically 

_msg : Object
object
status: object
text: "closed"
source: object
id: "abae7ff5.f68b88"
type: "modbus-response"
name: "Device 1"
_msgid: "3af8c38d.3a891c"_

If i re-connect the disconnected Modbus device, the response node detects this and instantly changes  its object.text  to the register value it is reading "[0]'" and also the populates Status node. 
It also updates every time it polls the Modbus register. This is great

_msg : Object
object
status: object
text: "'[0]'"
source: object
id: "abae7ff5.f68b88"
type: "modbus-response"
name: "Device 1"
_msgid: "6dc7d812.25b068"
payload: "open"_

 
 **Current behavior**
As explained above, Modbus response is not providing changing its status unless restarted.
 
 **Expected behavior**
 Modbus response node to change its status without the need to re-deploy the flow.
This feature is very useful and practical when you have many devices on the network.
If one device drops out due to "Power failure",  "Radio interference", "Vandalism" The status node can detect this. Using a Bool gate you can use the status of the response node to stop this particular device being read, this would ensure the rest of the modbus devices don't suffer and reduce there working efficiency.

 
 **Minimal reproduction of the problem with instructions**

1.   connect a modbus serial device to node-red
2.  insert 1 x Modbus serial read, 1 x modbus response node, 1 x status node., "connect in this order"
3.  deploy all 3 nodes into the flow editor
Disconnect the modbus device from the serial bus, monitor the response node. Reconnect the modbus device monitor the response node.
   
### Your Environment

Node red 19.5
Modbus contrib 4.1.1
serial port 7.1.4
Modbus devices.  N/A
Raspian Latest version  17/2/19
Node.js 6.11.0
Npm V3.10.10

  
Hope you can help with this. If it is not a bug, could it please be made a new feature.

Thanks

biancode, 2019-12-08:
please, test with v5.2.x or above

biancode, 2019-12-14:
like #162 - reopen if there is more

closed 2019-12-14 by biancode