BiancoRoyal/node-red-contrib-modbus #132: Modbus TCP Server trigger on write

obstler42, 2019-03-22:
- (X) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

I have a modbus device (heating controller) that periodically pushes out new data to a Modbus TCP Server -- I set it up to connect to the node-red-contrib-modbus Server and that basically works fine. Now I would like to act upon data received by the node-red tcp-modbus server, By I can't seem to get it to do anything every time a new value is written.

I don't want to poll the server -- I want it to react immediately a new holding register value is written by the external modbus device.

Is there a way to get the server node to start pushing out msgs/flows upon every WRITE?

biancode, 2019-03-28:
I would say you could do this with the Flex Server.

closed 2019-04-17 by biancode

CTGControls, 2023-12-21:
I would like to do something similar to [obstler42](https://github.com/obstler42) but I am not getting any output from the "Modbus-Flex Server" or the Modbus-Server when a successful right happens. I have a debug node on each of the 5 outputs.
[image]