BiancoRoyal/node-red-contrib-modbus #133: FSM Reset On State QUEUEING

PierreScerri, 2019-03-23:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - (X) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 Frequent 'FSM Reset On State QUEUEING' message

 **Expected behavior**
 
 
 **Minimal reproduction of the problem with instructions**
 
   1. 
   2. 
   3. 
   
### Your Environment
**Please tell us about your environment:**

  * (X) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - (X) Other:
         - Name: HassOS
         - Version: 2.10
   
   * Description of the connecting Modbus system:
   
     - Name   : SDM120/SDM230 via RS485 to USB converter to Raspberry Pi3 B
     - Version:
     - Vendor :
     - Link   : https://
     
    
* **node-red-contrib-modbus version:** ``

4.1.3

* **Node:** (min. 6.x)
   `node --version` = 9.11.1

biancode, 2019-03-28:
There should exist a problem in communication. The node is on queuing and gets a reset of the network or serial port. Maybe you are polling the device to fast? Check your flow to communicate less or more slowly.

PierreScerri, 2019-03-28:
Polling is every 15 seconds.

PierreScerri, 2019-03-28:
No errors with 30s.  Readjusted to 20s.  Seems to be working fine.

Thank you.

biancode, 2019-03-28:
If you get that problem again let me review your flow to help, please.

closed 2019-03-28 by biancode

PierreScerri, 2019-03-28:
I have included my flow.

Every 1m32 (sometimes randomly 2m32) I get this error:  'TransactionTimedOutError: Timed out' immediately followed by: 'FSM Reset On State QUEUEING' for each modbus read node (ie 9 messages).

Then I get:
[image]
followed by  
[image]
and it restarts.

Also, whenever I deploy the flow it gets stuck on 'waiting' and I have to restart node-red.

Any suggestions would be most welcome

`[{"id":"df73e07d.dc5bb","type":"tab","label":"SDM","disabled":false,"info":""},{"id":"3c3eaa46.8311e6","type":"comment","z":"df73e07d.dc5bb","name":"SDM230 Energy Meter","info":"","x":160,"y":80,"wires":[]},{"id":"5444ad0a.946dc4","type":"modbus-read","z":"df73e07d.dc5bb","name":"SDM 230 Voltage","topic":"","showStatusActivities":true,"logIOActivities":false,"showErrors":true,"unitid":"1","dataType":"InputRegister","adr":"0","quantity":"2","rate":"20","rateUnit":"s","delayOnStart":false,"startDelayTime":"","server":"fe372aa0.f8c9d8","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":160,"y":140,"wires":[["478fd4f6.3a0b2c"],[]]},{"id":"478fd4f6.3a0b2c","type":"function","z":"df73e07d.dc5bb","name":"Voltage","func":"var rawData = new ArrayBuffer(4);\nvar intView = new Uint16Array(rawData);\nvar fltView = new Float32Array(rawData);\n\nintView[0] = msg.payload[1]; //low\nintView[1] = msg.payload[0]; //high\n\nmsg.payload = parseFloat(fltView[0].toFixed(0));\nmsg.topic = \"voltage\";\n\nnode.status({fill:\"blue\",shape:\"ring\",text:msg.topic + \":\" + msg.payload});    \n\nreturn msg;","outputs":1,"noerr":0,"x":380,"y":140,"wires":[["a2b9b965.1b5818","7eb54e40.eee44"]]},{"id":"96d56a4f.ed4b6","type":"modbus-read","z":"df73e07d.dc5bb","name":"SDM 230 Current","topic":"","showStatusActivities":true,"logIOActivities":false,"showErrors":true,"unitid":"1","dataType":"InputRegister","adr":"6","quantity":"2","rate":"20","rateUnit":"s","delayOnStart":false,"startDelayTime":"","server":"fe372aa0.f8c9d8","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":160,"y":200,"wires":[["6fd7a085.f4d9f"],[]]},{"id":"6fd7a085.f4d9f","type":"function","z":"df73e07d.dc5bb","name":"Current","func":"var rawData = new ArrayBuffer(4);\nvar intView = new Uint16Array(rawData);\nvar fltView = new Float32Array(rawData);\n\nintView[0] = msg.payload[1]; //low\nintView[1] = msg.payload[0]; //high\n\nmsg.payload = parseFloat(fltView[0].toFixed(1));\nmsg.topic = \"current\";\n\nnode.status({fill:\"blue\",shape:\"ring\",text:msg.topic + \":\" + msg.payload});    \n\nreturn msg;","outputs":1,"noerr":0,"x":380,"y":200,"wires":[["1164b3a0.98d95c","7eb54e40.eee44"]]},{"id":"5e6d7ebc.925c1","type":"modbus-read","z":"df73e07d.dc5bb","name":"SDM 230 Power","topic":"","showStatusActivities":true,"logIOActivities":false,"showErrors":true,"unitid":"1","dataType":"InputRegister","adr":"12","quantity":"2","rate":"20","rateUnit":"s","delayOnStart":false,"startDelayTime":"","server":"fe372aa0.f8c9d8","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":160,"y":260,"wires":[["bc76e392.d54248"],[]]},{"id":"bc76e392.d54248","type":"function","z":"df73e07d.dc5bb","name":"Power","func":"var rawData = new ArrayBuffer(4);\nvar intView = new Uint16Array(rawData);\nvar fltView = new Float32Array(rawData);\n\nintView[0] = msg.payload[1]; //low\nintView[1] = msg.payload[0]; //high\n\nmsg.payload = parseFloat(fltView[0].toFixed(1));\nmsg.topic = \"power\";\n\nnode.status({fill:\"blue\",shape:\"ring\",text:msg.topic + \":\" + msg.payload});    \n\nreturn msg;","outputs":1,"noerr":0,"x":370,"y":260,"wires":[["72babe22.f2369","7eb54e40.eee44","89f9ab77.79789"]]},{"id":"ec883c2d.3cc0a8","type":"modbus-read","z":"df73e07d.dc5bb","name":"SDM 230 Energy Imp","topic":"","showStatusActivities":true,"logIOActivities":false,"showErrors":true,"unitid":"1","dataType":"InputRegister","adr":"72","quantity":"2","rate":"20","rateUnit":"s","delayOnStart":false,"startDelayTime":"","server":"fe372aa0.f8c9d8","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":180,"y":320,"wires":[["629847da.5e933"],[]]},{"id":"629847da.5e933","type":"function","z":"df73e07d.dc5bb","name":"Import","func":"var rawData = new ArrayBuffer(4);\nvar intView = new Uint16Array(rawData);\nvar fltView = new Float32Array(rawData);\n\nintView[0] = msg.payload[1]; //low\nintView[1] = msg.payload[0]; //high\n\nmsg.payload = (parseFloat(fltView[0]) * 1.00195 + 36481.26).toFixed(2) ;\nmsg.topic = \"import\";\n\nnode.status({fill:\"blue\",shape:\"ring\",text:msg.topic + \":\" + msg.payload});    \n\nreturn msg;","outputs":1,"noerr":0,"x":370,"y":320,"wires":[["5ea597c8.757f4","7eb54e40.eee44"]]},{"id":"b35fa8cf.93ce38","type":"modbus-read","z":"df73e07d.dc5bb","name":"SDM 230 Energy Exp","topic":"","showStatusActivities":true,"logIOActivities":false,"showErrors":true,"unitid":"1","dataType":"InputRegister","adr":"74","quantity":"2","rate":"20","rateUnit":"s","delayOnStart":false,"startDelayTime":"","server":"fe372aa0.f8c9d8","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":180,"y":380,"wires":[["681bf682.e9f2b"],[]]},{"id":"681bf682.e9f2b","type":"function","z":"df73e07d.dc5bb","name":"Export","func":"var rawData = new ArrayBuffer(4);\nvar intView = new Uint16Array(rawData);\nvar fltView = new Float32Array(rawData);\n\nintView[0] = msg.payload[1]; //low\nintView[1] = msg.payload[0]; //high\n\nmsg.payload = (parseFloat(fltView[0]) * 1.00238 + 10274.41).toFixed(2);\nmsg.topic = \"export\";\n\nnode.status({fill:\"blue\",shape:\"ring\",text:msg.topic + \":\" + msg.payload});    \n\nre [... cut ...] :"","name":"SDM230","tab":"db77dc69.ae668","order":1,"disp":true,"width":"6","collapse":false},{"id":"5a3ddb88.1e6714","type":"ui_group","z":"","name":"SDM120","tab":"db77dc69.ae668","order":2,"disp":true,"width":"6","collapse":false},{"id":"d6595a1d.63d978","type":"ui_group","z":"","name":"Graphs","tab":"db77dc69.ae668","order":3,"disp":true,"width":"6","collapse":false},{"id":"db77dc69.ae668","type":"ui_tab","z":"","name":"Modbus","icon":"dashboard","order":1,"disabled":false,"hidden":false}]`

biancode, 2019-03-28:
it is not the time in requesting - it is the queue delay of just 30ms - try to set 100 ms or more in the connector

biancode, 2019-03-28:
you could request every 2 seconds by 200 ms queue delay  ( 200ms * 5values = 1000ms - serial request of 5 needs 1 sec.) - or - request poll per 1 sec by 120 ms ( 120 ms * 5 = 600ms) ... and so on - you are requesting 5 serial commands per poll if you insert more nodes think about your time to handle all request between node-red and your device ;)

PierreScerri, 2019-03-28:
Ok I think I've got it.  
I have 9 read nodes. 
So setting queue delay of 200ms gives 1800ms.  

If I set polling to 3s it should be fine.
Correct me if I'm wrong.

Thank you for your help

marcu5au, 2019-06-16:
Hi! I came here with the same issue "FSM Reset On State QUEUEING" and after reading the above have set my Queue Delay to 200ms to solve most of my issue.
I pole my SDM230 every 30 seconds (reading five registers) and I'm using the USR-W610 (RS485 to WiFi converter). 
I actually suspect the root cause is the USR-W610 as it is terrible with handling the TCP stack over WiFi compared to the older (now discontinued) USR-WIFI232-604 unit.
Thanks for the comments above to help me out!