BiancoRoyal/node-red-contrib-modbus #134: TypeError: Cannot read property `registerForModbus` of null

AndrewCarterUK, 2019-03-29:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( x) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
When we add a modbus server and deploy with it embedded in the flex getter node, we get the error below and it doesn't work.

There are no errors in the configuration highlighted and the only time an error becomes apparent is when we hit the deploy button and it appears in `node-red-log` and the debug messages.

```
29 Mar 13:33:51 - [error] [modbus-client:cf998c67.2dc71] SyntaxError: Unexpected token function
29 Mar 13:33:51 - [error] [modbus-flex-getter:2e593518.41214a] TypeError: Cannot read property 'registerForModbus' of null
```

 **Expected behavior**
It would work.
 
 **Minimal reproduction of the problem with instructions**
 
   1. Install `node-red` on a Raspberry Pi
   2. Install `node-red-contrib-modbus` in the `~/.node-red` directory on a Raspberry Pi
   3. Try and make a modbus server node 
   4. Deploy
   
### Your Environment
**Please tell us about your environment:**
Raspberry Pi

  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
     
    
* **node-red-contrib-modbus version:** ``

* **Node:** (min. 6.x)
   `node --version` =

AndrewCarterUK, 2019-03-29:
The member of my team investigating this found that downgrading the `node-red-contrib-modbus` version by 1 fixed the problem.

koraysels, 2019-08-28:
> The member of my team investigating this found that downgrading the `node-red-contrib-modbus` version by 1 fixed the problem.

a complete version or what do you mean ? I have this issue as well. do you mean 3.1.3 or 4.1.2 or 4.0.3 ... ?

AndrewCarterUK, 2019-08-29:
We downgraded to 3.6.1 and that fixed it for us.

Our manual has installation instructions as:

`npm install node-red-contrib-modbus@3.6.1`

biancode, 2019-11-17:
hm on my side I can not reproduce it - nodejs should be 8 or 10

[image]

biancode, 2019-12-08:
please, test with v5.2.x

biancode, 2019-12-14:
reopen if it happens with v.5.2 or above again and send me more logs, please

`DEBUG=contribModbus*,modbus-serial node-red -v`

closed 2019-12-14 by biancode