BiancoRoyal/node-red-contrib-modbus #137: Node-RED log full of WriteMultipleRegistersRequestBody messages

aapiskukko, 2019-04-09:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - (x ) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
When data is written to Modbus Server node, node-RED log gets spammed with this kind of messages:

WriteMultipleRegistersRequestBody {
  _fc: 16,
  _address: 68,
  _values: <Buffer 00 01>,
  _byteCount: 8,
  _numberOfBytes: 2,
  _quantity: 1,
  _valuesAsBuffer: <Buffer 00 01>,
  _valuesAsArray: [ 1 ] }
WriteMultipleRegistersRequestBody {
  _fc: 16,
  _address: 69,
  _values: <Buffer ff fe>,
  _byteCount: 8,
  _numberOfBytes: 2,
  _quantity: 1,
  _valuesAsBuffer: <Buffer ff fe>,
  _valuesAsArray: [ 65534 ] }
WriteMultipleRegistersRequestBody {
  _fc: 16,
  _address: 65,
  _values: <Buffer 02 6c>,
  _byteCount: 8,
  _numberOfBytes: 2,
  _quantity: 1,
  _valuesAsBuffer: <Buffer 02 6c>,
  _valuesAsArray: [ 620 ] }

 **Expected behavior**
 No this kind of messages in Node-RED log. Is there a way to hide these messages?
 
 **Minimal reproduction of the problem with instructions**
 
   1. 
   2. 
   3. 
   
### Your Environment
**Please tell us about your environment:**

node-red-contrib-modbus@4.1.3, Linux

  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * (x ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - (x) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   :
     - Version:
     - Vendor :
     - Link   : https://
     
    
* **node-red-contrib-modbus version:** ``
node-red-contrib-modbus@4.1.3

* **Node:** (min. 6.x)
   `node --version` = 8.12.0

biancode, 2019-04-17:
https://github.com/biancode/node-red-contrib-modbus/wiki/DEBUG

jan-kiszka, 2020-03-04:
Moin,
@biancode, am I interpreting your reply correctly that the reported log behavior is the default and the user is suppose to tune it down via that DEBUG env variable? If yes, shouldn't the default be less verbose?

I'm following up on this because of https://support.industry.siemens.com/tf/ww/en/posts/modbus-issue-image-v2-6-0/230028/?page=0&pageSize=10#post910369, where a user reported the same issue from our https://github.com/siemens/meta-iot2000 image (we started to preintegrate node-red-contrib-modbus).

biancode, 2020-03-04:
@jan-kiszka that interpretation is wrong, sorry. There was no modbus read at all and I had no more time to answer.

jan-kiszka, 2020-03-04:
Ah, ok, you wanted to the debug log from the user. Let's see what we can do this time.

aapiskukko, 2020-03-04:
I am sorry, I didn't know that I was explicitly expected more info. By just providing a link to wiki, it seemed that project author didn't seem to be interested in this issue. That is why I didn't come back. I debugged the root cause myself and created another issue to other project: https://github.com/Cloud-Automation/node-modbus/issues/236.

jan-kiszka, 2020-03-04:
But is node-modbus actually pulled in here?

jan-kiszka, 2020-03-04:
Ah, node-modbus == jsmodbus. OK, makes sense now. And upstream is open for PR as well.

biancode, 2020-03-04:
Jsmodbus is just the server modbus-serial is the client part also for tcp

alexbuczynsky, 2020-03-04:
See full comments on resolved issue here: https://github.com/Cloud-Automation/node-modbus/issues/236

TL;DR;
`node-red-contrib-modbus` greater than 5.0.1 [has already upgraded to the jsmodbus@4.0.0 version](https://github.com/BiancoRoyal/node-red-contrib-modbus/blob/afe657b052c15e5addcc7b9b2a31896c311a0f55/package.json#L8) of the library and where the console.log statement has been removed already.

So if you are using `node-red-contrib-modbus: 4.x.x` make sure you are using [jsmodbus@3.1.6](https://www.npmjs.com/package/jsmodbus/v/3.1.6) but if you are using `node-red-contrib-modbus: 5.x.x` then this issue should already not be a problem.

biancode, 2020-04-01:
reopen if it has issues with v5.x of the package, please

closed 2020-04-01 by biancode