BiancoRoyal/node-red-contrib-modbus #138: Modbus TCP server with different Unit IDs

ABatGitHub, 2019-04-24:
I'm submitting a question to get some help
I have a Modbus TCP server acting as a gateway for slave devices.
The server has an IP address and each slave has a Unit ID. 

I am using node-red-contrib-modbus. I want to set up a test server that can simulate my real-world device and test the coils in the slave devices independently.

I have set up a write to UNIT IDs coils 1-4 and set up Reads from coils 1-4

When I write to one of the Unit IDs the value appears on all the Unit IDs not just the one, I wrote to.

When I read from the server the buffers read the same values on all Unit IDs.

I cannot get the coils on each UNIT ID to act independently.
I have tried both TCP and RTU-BUFFERED types on the client with no difference.
I would appreciate any advice to be able to read and write to the server and see the different Unit IDs with different values.

Thanks for your help.
 
Below is a flow to reproduce what I am doing.

[{"id":"3e64b5bd.d952ea","type":"modbus-server","z":"81ec175d.437d98","name":"BMS","logEnabled":false,"hostname":"10.0.0.1","serverPort":"502","responseDelay":100,"delayUnit":"ms","coilsBufferSize":10000,"holdingBufferSize":10000,"inputBufferSize":10000,"discreteBufferSize":10000,"showErrors":false,"x":270,"y":40,"wires":[[],[],[],["7cd35ce0.0c4634"]]},{"id":"11d44fc6.f22bd","type":"inject","z":"81ec175d.437d98","name":"","topic":"","payload":"","payloadType":"date","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":120,"y":40,"wires":[["3e64b5bd.d952ea"]]},{"id":"7cd35ce0.0c4634","type":"debug","z":"81ec175d.437d98","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":490,"y":60,"wires":[]},{"id":"9cce7256.a8396","type":"modbus-write","z":"81ec175d.437d98","name":"UNIT ID 1 Fault","showStatusActivities":true,"showErrors":true,"unitid":"1","dataType":"Coil","adr":"0","quantity":"1","server":"bd31c2d6.405ec","x":280,"y":140,"wires":[[],["e1fa5c1d.e0c5e"]]},{"id":"73d539fc.034288","type":"inject","z":"81ec175d.437d98","name":"True","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":110,"y":120,"wires":[["9cce7256.a8396"]]},{"id":"e1fa5c1d.e0c5e","type":"debug","z":"81ec175d.437d98","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":490,"y":140,"wires":[]},{"id":"2ae3012.7f829fe","type":"modbus-write","z":"81ec175d.437d98","name":"UNIT ID 1 CMD","showStatusActivities":false,"showErrors":false,"unitid":"1","dataType":"Coil","adr":"1","quantity":"1","server":"bd31c2d6.405ec","x":280,"y":260,"wires":[[],["b23f5f1a.3ad05"]]},{"id":"3c67f7be.edb8a8","type":"inject","z":"81ec175d.437d98","name":"True","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":110,"y":240,"wires":[["2ae3012.7f829fe"]]},{"id":"b23f5f1a.3ad05","type":"debug","z":"81ec175d.437d98","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":490,"y":260,"wires":[]},{"id":"4fd62eed.4cf9d","type":"modbus-write","z":"81ec175d.437d98","name":"UNIT ID 2 Fault","showStatusActivities":false,"showErrors":false,"unitid":"2","dataType":"Coil","adr":"0","quantity":"1","server":"bd31c2d6.405ec","x":280,"y":380,"wires":[[],["c506517b.f2eab"]]},{"id":"370bbb70.cda454","type":"inject","z":"81ec175d.437d98","name":"True","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":110,"y":360,"wires":[["4fd62eed.4cf9d"]]},{"id":"c506517b.f2eab","type":"debug","z":"81ec175d.437d98","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":490,"y":380,"wires":[]},{"id":"5c68dc43.dba214","type":"modbus-write","z":"81ec175d.437d98","name":"UNIT ID 2 CMD","showStatusActivities":false,"showErrors":false,"unitid":"2","dataType":"Coil","adr":"1","quantity":"1","server":"bd31c2d6.405ec","x":280,"y":500,"wires":[[],["26a327dd.d61b38"]]},{"id":"94b5c0e0.cafb1","type":"inject","z":"81ec175d.437d98","name":"True","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":110,"y":480,"wires":[["5c68dc43.dba214"]]},{"id":"26a327dd.d61b38","type":"debug","z":"81ec175d.437d98","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":490,"y":500,"wires":[]},{"id":"edb5d773.ba8318","type":"modbus-write","z":"81ec175d.437d98","name":"UNIT ID 3 Fault","showStatusActivities":false,"showErrors":false,"unitid":"3","dataType":"Coil","adr":"0","quantity":"1","server":"bd31c2d6.405ec","x":900,"y":140,"wires":[[],["ed152390.37192"]]},{"id":"c8b2f175.343ea","type":"inject","z":"81ec175d.437d98","name":"True","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":730,"y":120,"wires":[["edb5d773.ba8318"]]},{"id":"ed152390.37192","type":"debug","z":"81ec175d.437d98","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":1110,"y":140,"wires":[]},{"id":"60cb488b.7845b8","type":"inject","z":"81ec175d.437d98","name":"False","topic":"","payload":"false","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":730,"y":180,"wires":[["edb5d773.ba8318"]]},{"id":"988d202b.0d8e6","type":"modbus-write","z":"81ec175d.437d98","name":"UNIT ID 3 CMD","showStatusActivities":false,"showErrors":false,"unitid":"3","dataType":"Coil","adr":"1","quantity":"1","server":"bd31c2d6.405ec","x":900,"y":260,"wires":[[],["6efb46a8.ced388"]]},{"id":"a2689cb4.baa78","type":"inject","z":"81ec175d.437d98","name":"True","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":730,"y":240,"wires":[["988d202b.0d8e6"]]},{"id":"6efb46a8.ced388","type":"debug","z":"81ec175d.437d98","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":1110,"y":240,"wires":[]},{"id":"ffd6e866.8ad168","type":"inject","z":"81ec175d.437d98","name":"False","topic":"","payload":"false","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":730,"y":300,"wires":[["988d202b.0d8e6"]]},{"id":"70a311c3.ad4c3","type":"modbus-write","z":"81ec175d.437d98","name":"UNIT ID 4 Fault","showStatusActivities":false,"showErrors":false,"unitid":"4","dataType":"Coil","adr":"0","quantity":"1","server":"bd31c2d6.405ec","x":900,"y":380,"wires":[[],["9adc79a5.7609d8"]]},{"id":"38f73f6f.cc0b9","type":"inject","z":"81ec175d.437d98","name":"True","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":730,"y":360,"wires":[["70a311c3.ad4c3"]]},{"id":"9adc79a5.7609d8","type":"debug","z":"81ec175d.437d98","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":1110,"y":360,"wires":[]},{"id":"c164d4c6.de2198","type":"inject","z":"81ec175d.437d98","name":"False","topic":"","payload":"false","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":730,"y":420,"wires":[["70a311c3.ad4c3"]]},{"id":"ab9b09e0.9b77c8","type":"modbus-write","z":"81ec175d.437d98","name":"UNIT ID 4 CMD","showStatusActivities":false,"showErrors":false,"unitid":"4","dataType":"Coil","adr":"1","quantity":"1","server":"bd31c2d6.405ec","x":900,"y":500,"wires":[[],["ae8f1533.3aa2a8"]]},{"id":"9c609fed.a28e9","type":"inject","z":"81ec175d.437d98","name":"True","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":730,"y":480,"wires":[["ab9b09e0.9b77c8"]]},{"id":"ae8f1533.3aa2a8","type":"debug","z":"81ec175d.437d98","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":1110,"y":460,"wires":[]},{"id":"6a79feb1.c2606","type":"inject","z":"81ec175d.437d98","name":"False","topic":"","payload":"false","payloadType":"bool","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":730,"y":540,"wires":[["ab9b09e0.9b77c8"]]},{"id":"7961661d.09e688","type":"modbus-flex-getter","z":"81ec175d.437d98","name":"Read_MB","showStatusActivities":false,"showErrors":false,"logIOActivities":false,"server":"716ecd11.6a2384","useIOFile":false,"ioFile":"","useIOForPayload":false,"x":1020,"y":700,"wires":[[],["7ad52f9a.ff38f"]]},{"id":"329ec0ae.73d1b","type":"inject","z":"81ec175d.437d98","name":"Read","topic":"","payload":"true","payloadType":"bool","repeat":"","crontab":"","once":false,"x":110,"y":700,"wires [... cut ...] Timeout":"1000","reconnectTimeout":"2000"}]
  
  * Device :
     Linux: v
         - Name: IOT 2040 running Linux Yocto
         - Version:2.1
   
   * Description of the connecting Modbus system:
     - Name   : JACE 8000
     - Version: Niagara 4: 4.1 
     - Vendor : Tridum
     - Link   : https:// https://www.tridium.com/~/media/tridium/enterprisesecurity/documents/jace%208000%20ds%20new%20tridium.ashx?la=es-mx
     
    
* **node-red-contrib-modbus version:** ``
   `node --version` =v0.20.5

opt-natter, 2019-05-15:
Maybe use a Flex-Server?
If you could provide your flow properly formated (code tags would be nice). I could have a look.

ABatGitHub, 2019-05-17:
Thanks for the reply.
I tried the Flex-Server but was not able to resolve the issue.
I had success testing the node-red flow using a third party Modbus slave simulator.

closed 2019-05-17 by ABatGitHub