BiancoRoyal/node-red-contrib-modbus #143: Node-RED crashes when 2 Flex-Server nodes are created with the same port but different unit IDs

thatcadguy, 2019-06-17:
## Problem
 - (X) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - (X) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
Creating two TCP flex server nodes with the same IP/port but different unit IDs makes Node-RED crash.
 
 **Expected behavior**
Ideally the unit ID should be respected according to the standard and making TCP requests to the same port but different unit IDs should be allowed to respond differently with different registers. At a minimum, the error should be caught so that Node-RED doesn't crash.
 
 **Minimal reproduction of the problem with instructions**
 
   1. Add two Modbus Flex Server nodes. Level everything as default, except change the unit ID on one of them.
   2. Deploy.
   3. Observe crash information from commandline:

17 Jun 15:03:27 - [info] Starting flows
17 Jun 15:03:27 - [info] Started flows
17 Jun 15:03:27 - [info] Server now running at http://127.0.0.1:1880/
17 Jun 15:03:43 - [info] Stopping flows
17 Jun 15:03:43 - [info] Stopped flows
17 Jun 15:03:43 - [info] Starting flows
17 Jun 15:03:43 - [info] Started flows
17 Jun 15:03:43 - [red] Uncaught Exception:
17 Jun 15:03:43 - Error: listen EADDRINUSE: address already in use 0.0.0.0:11502
    at Server.setupListenHandle [as _listen2] (net.js:1290:14)
    at listenInCluster (net.js:1338:12)
    at doListen (net.js:1471:7)
    at process._tickCallback (internal/process/next_tick.js:63:19)

C:\Users\thatc>
   
### Your Environment
**Please tell us about your environment:**
All the latest on Windows 10 x64

  * ( ) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - (x) Windows: v 10 x64
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
none
     
    
* **node-red-contrib-modbus version:** ``

* **Node:** (min. 6.x)
   `node --version` = latest v10 LTS

biancode, 2019-06-18:
Hi, two flex servers need two ports. That is TCP. If you need multiple UnitID's, then you need to implement this inside the flex server to handle them. It is not available out of the box.

see: Error: listen EADDRINUSE: address already in use 0.0.0.0:11502 - Address and port in use

dceejay, 2019-06-18:
Why are you closing this so quickly ? Surely it would be easy enough to detect the port is already in use and either fail gracefully - or warn the user - rather than breaking completely ? When you say it should be handled inside the flex server - is this not your code ? Would it not be better to enhance the one node rather than force a fork ?

biancode, 2019-06-18:
@dceejay I mean therefore node-red got a new start flag to handle such config problems and the node-red team decided (we discussed this in a slack conversation) node-red has to crash on such problems, shouldn't it anymore?

dceejay, 2019-06-18:
Yes - It should crash if it gets an async error it can't handle - BUT if a node can detect an error (like there being two instances of itself on same port) then it should do so. The overall node instance could have a global array of "portsInUse" and add to it each time a specific node is added/configured... and if there is a duplicate then raise an error before trying to connect twice. 

EG see core udp node - around https://github.com/node-red/node-red/blob/ccc3809daa3b03e7171cb6f4ccf4d6e0f1188bef/packages/node_modules/%40node-red/nodes/core/io/32-udp.js#L83

Ekristoffe, 2019-11-15:
@dceejay you said: 
> Expected behavior
> Ideally the unit ID should be respected according to the standard and making TCP requests to the same port but different unit IDs should be allowed to respond differently with different registers. At a minimum, the error should be caught so that Node-RED doesn't crash.
The answer is yes and no. in pure Modbus TCP/IP the Unit ID should be ignored. The Unit ID should only be used by a Modbus RTU <-> Modbus TCP/IP converter (or gateway).
In @biancode case, the Flex-Server node is only made for pure Modbus TCP/IP. I understand it totally ignore the Unit ID.
I understand the rage about Node-RED crashing and also I think it is not clean. 
Yes it should be better to return an error in the Node-RED debug windows since this is a configuration error and it can be easily fixed by disabling the flow / server instead of pushing the kill switch.

biancode, 2019-11-17:
With the new `node-red --safe` parameter of Node-RED, it is also possible to start, correct it, and deploy with a working state of the configuration.

biancode, 2019-11-17:
There is a try/catch as in MQTT, but in Node-RED it do not catch it inside the node if it is an exception like a network error.

biancode, 2019-11-17:
in link with #125

joearkay, 2019-12-04:
FYI - I too have come across this problem whilst deploying two Modbus RTU slave devices. I was hesitant to open an issue until some further feedback and research, but here's mine detailed: https://discourse.nodered.org/t/multiple-modbus-rtu-slaves-causes-error/18783

biancode, 2019-12-08:
A port can only be used once - the unitid is to handle inside the server as it is also in other modbus server implementations, please.

closed 2019-12-08 by biancode