BiancoRoyal/node-red-contrib-modbus #144: Node-RED v1.x and NodeJS 12

biancode, 2019-06-20:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer), please!

## Feature
 - ( x) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

* **node-red version:** ``
v1.x

* **Node:** (min. 6.x)
   `node --version` = 12

racer77rus, 2019-10-02:
Looks like modbus node doesn't work with node-red 1.0.0. It shows "Active" state but without any output from the node. (node-red 1.0.0, nodeJS v10)

vampywiz17, 2019-10-03:
Same with me. I try with Node-red 1.0.0 and Node JS v10

I get error if i start node-RED with console (i not remember the error message, i switch back to a previous version)

Somebody able to paste it?

biancode, 2019-10-03:
@racer77rus and @vampywiz17 correct. We stopped contributing to Node-RED for now. I have to write my master thesis first.

biancode, 2019-10-05:
Sponsors welcome https://contact.bianco-royal.com/

trenty27, 2019-10-09:
> Same with me. I try with Node-red 1.0.0 and Node JS v10
> 
> I get error if i start node-RED with console (i not remember the error message, i switch back to a previous version)
> 
> Somebody able to paste it?
``
(node:27964) UnhandledPromiseRejectionWarning: TypeError: a is not a function
at activateSending (/home/pi/.node-red/node_modules/node-red-contrib-modbus/modbus/maps/modbus-client.js:587:20)
(node:27964) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 719)
``

kpm-at-hfi, 2019-10-29:
I'm wondering if this issue title is slightly off; I think that `node-red-contrib-modbus` 4.1.3 has this `UnhandledPromiseRejectionWarning` when running in **Node-RED >=1** on **Node.js >=10** (not just 12 as the current title suggests).

Also, is it true that `node-red-contrib-modbus` 4.1.3 works fine in **_Node-RED <1_** on **Node.js >=10**?

fabiomoli, 2019-10-30:
@kpm-at-hfi yes, `node-red-contrib-modbus` works fine in Node-RED 0.20.8

I think the problem is related to the fackt that Node-RED 1.0.x is fully asynchronous [https://nodered.org/blog/2019/09/30/version-1-0-released#asynchronous-message-passing](https://nodered.org/blog/2019/09/30/version-1-0-released#asynchronous-message-passing) and is now necessary to andle promises [https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Promise](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Promise)

kpm-at-hfi, 2019-10-30:
Thanks @fabiomoli ...  yeah, I see now that Node-RED 0.20.8 + Node.js 10 + this module are fine. I've been kind of fixated on wanting to be on Node-RED 1.0, so wasn't thinking outside of that box.

FWIW I was able to get my company to sponsor a little bit relating to this issue... if others might be able to do that, there's a Sponsor link up top of this page. That might help. :)

racer77rus, 2019-10-30:
It seems that the module works fine with nodejs 10 and node-red 1.0.2.

kpm-at-hfi, 2019-10-30:
@racer77rus interedsting... what do you mean working? Are you able to communicate with modbus devices? Ethernet or serial?

FWIW, I'm definitely seeing the `UnhandledPromiseRejectionWarning: TypeError: a is not a function at activateSending` error reported above when I try to work with modbus serial, the cause likely being what @fabiomoli noted.

kpm-at-hfi, 2019-10-30:
In the blog post that @fabiomoli noted above, there's an **immediately depreciated** `runtimeSyncDelivery: true` setting noted that does allow everything to work, meaning that it seems I can communicate over modbus serial using this module when running on Node-RED 1.0.2 using Node.js 10.

So that could be a short term fix for those who might need something now.

kpm-at-hfi, 2019-10-30:
Upon reading @trenty27's comment, I redeployed the same code removing the `runtimeSyncDelivery: true` setting and _it is working_. I can't explain it.

* Node-RED version: v1.0.2
* Node.js version: v10.17.0
* NPM 6.11.3
* Linux 4.19.66-v7+ arm LE

This is a custom Docker build created using https://github.com/node-red/node-red-docker/tree/master/docker-custom

I'm flummoxed.

biancode, 2019-11-17:
All early release testers could help to test here. You could test with install cmd like: 

into Node-RED
`npm i https://github.com/BiancoRoyal/node-red-contrib-modbus/archive/5.0.1-alpha.0.tar.gz`

or into Global NPM
`npm i -g https://github.com/BiancoRoyal/node-red-contrib-modbus/archive/5.0.1-alpha.0.tar.gz `

biancode, 2019-12-03:
Node-RED v1.x since v5.1.0

biancode, 2019-12-08:
next step to test more with v5.2.0

kpm-at-hfi, 2020-01-07:
Just a quick note that this has been working for me in initial testing. I'll be seeing this module in action more over the next two weeks and will let you know how it goes.

Thanks for working this, @biancode!

RMutharaju, 2020-02-12:
Just a question, will there any further release for Mindconnect (v5.3.0?) to support Node.Js 12 and Node-RED 1.0.3. If yes, by when it is planned to be released?

currently, I see that Mindconnect v5.2.0 is not building with the above combination 
refer:  https://github.com/BiancoRoyal/node-red-contrib-modbus/issues/176

biancode, 2020-03-28:
Please, test v5.4.0 and open a new issue information if the problem with Nodejs v12 or v13 is not fixed.

closed 2020-03-28 by biancode