BiancoRoyal/node-red-contrib-modbus #145: Multiple Modbus RTU Slaves using the same port pon Raspberry Pi

joedtara, 2019-07-03:
Hi, I'm working on a IIOT demostration module and using a Raspberry Pi 3+ for running Node-Red, I already communicated with a CFW500 Drive using a USB-RS485 converter and the modbus nodes (everything using Modbus RTU BUFFERED). But now I'm trying to communicate with a M2M Analyzer (with modbus) using the same USB port (and the same converter) connecting the 2 cables from M2M in parallel with the cables from the CFW500 and it's not working. 
I already configured the 2 clients (CFW500 and M2M) with their respective Modbus ID (CFW500: 3, M2M: 2).
I have proved the connection separately with both devices and I'm sure the wiring is correct because I can read the registers but when I use the modbus nodes with both Clients configured
in the same port (/dev/ttyUSB0) one of them just doesn't connect.

I think it's because the nodes can't access to the same usb port so I need some guidelines to make it work. Thanks

## Support/Advice
Support for connecting multiple slaves with RTU.

## Problem
Can't work with multiple modbus slaves connected to the same modbus rtu network.
   
### Your Environment
Running on Raspberry Pi3+
Node.js version 10.16.0
Node Red version 0.20.5

biancode, 2019-07-06:
maybe an issue #125 duplicate

biancode, 2019-11-17:
The port should only be used once (1 client connector) because it is serial and then it should work over the UnitID via Modbus if they have different (i read 2, 3 here)

closed 2019-11-17 by biancode