BiancoRoyal/node-red-contrib-modbus #146: Error connectin modbus wifi thermostat MC6

tty74it, 2019-07-06:
I'm trying to connect to a china MC6 thermostat wifi modbus
With modbustcp node work

The error is:

"Error: Data length error, expected 7 got 8"

Thank's
Tiziano
[MC6-MODBUS.pdf](https://github.com/biancode/node-red-contrib-modbus/files/3364703/MC6-MODBUS.pdf)

biancode, 2019-07-06:
@yaacov https://github.com/yaacov/node-modbus-serial/issues?utf8=%E2%9C%93&q=wifi

biancode, 2019-07-06:
@tty74it could you test with a simple node-modbus-serial script on wifi, please? Do you get the same result there?

yaacov, 2019-07-06:
> @tty74it could you test with a simple node-modbus-serial script on wifi, please? Do you get the same result there?

Example of node-modbus-serial  scripte for debugging:
https://github.com/yaacov/node-modbus-serial/blob/master/examples
https://github.com/yaacov/node-modbus-serial/blob/master/examples/logger.js

tty74it, 2019-07-06:
Similar problem using logger.js

pi@hassbian:~/.node-red/node_modules/modbus-serial/examples$ node logger-test.js 
Connected
Data length error, expected 11 got 8

`/* eslint-disable no-console, spaced-comment */

// create an empty modbus client
//var ModbusRTU = require("modbus-serial");
var ModbusRTU = require("../index");
var client = new ModbusRTU();
var timeoutRunRef = null;
var timeoutConnectRef = null;

var networkErrors = ["ESOCKETTIMEDOUT", "ETIMEDOUT", "ECONNRESET", "ECONNREFUSED", "EHOSTUNREACH"];

// check error, and reconnect if needed
function checkError(e) {
    if(e.errno && networkErrors.includes(e.errno)) {
        console.log("we have to reconnect");

        // close port
        client.close();

        // re open client
        client = new ModbusRTU();
        timeoutConnectRef = setTimeout(connect, 1000);
    }
}

// open connection to a serial port
function connect() {
    // clear pending timeouts
    clearTimeout(timeoutConnectRef);

    // if client already open, just run
    if (client.isOpen) {
        run();
    }

    // if client closed, open a new connection
    client.connectTCP("192.168.88.24", { port: 502 })
        .then(setClient)
        .then(function() {
            console.log("Connected"); })
        .catch(function(e) {
            checkError(e);
            console.log(e.message); });
}

function setClient() {
    // set the client's unit id
    // set a timout for requests default is null (no timeout)
    client.setID(1);
    client.setTimeout(1000);

    // run program
    run();
}

function run() {
    // clear pending timeouts
    clearTimeout(timeoutRunRef);

    // read the 3 registers starting at address 0
    client.readHoldingRegisters(0, 3)
        .then(function(d) {
            console.log("Receive:", d.data); })
        .then(function() {
            timeoutRunRef = setTimeout(run, 1000); })
        .catch(function(e) {
            checkError(e);
            console.log(e.message); });
}

// connect and start logging
connect();`

yaacov, 2019-07-07:
Thanks :+1:

a - can you check with bufferedtcp:
https://github.com/yaacov/node-modbus-serial/blob/master/examples/buffertcp.js

b - can you enable debug output for both tests:
```
 export DEBUG=modbus-serial
```
https://github.com/yaacov/node-modbus-serial/blob/master/examples/debug.js

tty74it, 2019-07-07:
pi@hassbian:~/.node-red/node_modules/modbus-serial/examples$ node buffertcp-test.js 
Connected
Timed out

pi@hassbian:~/.node-red/node_modules/modbus-serial/examples$ node debug-test.js 
  modbus-serial TCP port: signal connect +0ms
  modbus-serial { action: 'send tcp port',
  modbus-serial   data: <Buffer 01 03 00 00 00 03 05 cb>,
  modbus-serial   buffer: <Buffer 00 01 00 00 00 06 01 03 00 00 00 03>,
  modbus-serial   unitid: 1,
  modbus-serial   functionCode: 3,
  modbus-serial   transactionsId: 1 } +12ms
Connected
  modbus-serial { action: 'receive tcp port strings',
  modbus-serial   data: <Buffer 00 01 00 00 00 06 01 03 06 01 00 00 00 00 34> } +71ms
  modbus-serial { action: 'parsed tcp port',
  modbus-serial   buffer: <Buffer 01 03 06 01 00 00 14 82>,
  modbus-serial   transactionId: 1 } +4ms
Data length error, expected 11 got 8
  modbus-serial TCP port: signal close: false +10ms

modbus-serial TCP port: signal close: false +10ms
pi@hassbian:~/.node-red/node_modules/modbus-serial/examples$ node buffertcp-test.js 
  modbus-serial { action: 'send tcp rtu buffered port',
  modbus-serial   data: <Buffer 01 02 00 00 00 03 38 0b>,
  modbus-serial   buffer: <Buffer 00 01 00 00 00 06 01 02 00 00 00 03>,
  modbus-serial   transactionsId: 1 } +0ms
Connected
  modbus-serial { action: 'receive tcp rtu buffered port',
  modbus-serial   data: <Buffer 00 01 00 00 00 06 01 82 01>,
  modbus-serial   buffer: <Buffer 00 01 00 00 00 06 01 82 01> } +71ms
  modbus-serial { protocolID: 0, msgLength: 6, bufferLength: 9, cmd: 130 } +3ms
  modbus-serial { protocolID: 0, msgLength: 1537, bufferLength: 9, cmd: 1 } +1ms
  modbus-serial { protocolID: 6, msgLength: 386, bufferLength: 9, cmd: undefined } +0ms
  modbus-serial { protocolID: 1537,
  modbus-serial   msgLength: 33281,
  modbus-serial   bufferLength: 9,
  modbus-serial   cmd: undefined } +1ms
Timed out

Thank's
Tiziano

yaacov, 2019-07-07:
Well, the controller is answering with a none correct packets :-(

**request**:
00 01 00 00 00 06 01 03 00 00 00 03

Part of Data Package | Description | Value
-- | -- | --
00 01 | Transaction identifier | 0x0001 (1)
00 00 | Protocol identifier | 0 = MODBUS protocol
00 06 | Length | 0x0006 (6)
01 | Unit identifier | 0x01 (1)
03 | Function code | 0x03 (3) - Read Holding Registers
00 00 | Starting address | 0x0001 (1)
00 03 | Quantity | 0x0003 (3)

**response** from controller is:
00 01 00 00 00 **06** 01 03 06 01 00 00 00 00 34  <- Error
00 01 00 00 00 **09** 01 03 06 01 00 00 00 00 34  <- Should have been
 

Part of Data Package | Description | Value
-- | -- | --
00 01 | Transaction identifier | 0x0001 (1)
00 00 | Protocol identifier | 0 = MODBUS protocol
00 09 | Length | 0x0009 (9)
01 | Unit identifier | 0x01 (1)
03 | Function code | 0x03 (3) - Read Holding Registers
06 | Byte count | 0x06 (6)
01 00 00 00 00 34 | Register value | 0x0100 (256), 0x0000 (0), 0x0034 (52)

tty74it, 2019-07-07:
What can I do to make it work? With modbustcp node it works and also with modbus pool (windows) it works with all registers

Thank's
Tiziano

yaacov, 2019-07-07:
> What can I do to make it work?

You will have to add a new port ( like the tcp buffered one ) that ignore the length argument, and accept the length of the packet instead, see:
https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L58
here we use the parameter sent from the controller, if we want the packet to pass check, we need to ignore this value and use `data.length` from:
https://github.com/yaacov/node-modbus-serial/blob/master/ports/tcpport.js#L56
to calculate the actual length value, like I did when I fixed the packet from `length = 06` to `length = 09` in the comment above.
It looks like this controller always send `length=6` without calculating real length :-(

yaacov, 2019-07-07:
to check this go to the code in line 58, and hardcode length to be `9`, things should start work for you query of 3 registers starting from adress 0

yaacov, 2019-07-07:
I think the calculated length value should be `data.length - 6` you can try that too

yaacov, 2019-07-07:
you can also try asking for just one register, instead of 3, that may also work without changing the code.

tty74it, 2019-07-07:
pi@hassbian:~/.node-red/node_modules/modbus-serial/examples$ node debug-test.js 
Connected
Receive: [ 256, 0, 52 ]

I've hardcoded in tcpport.js
length = 9;

256 is room temp
0  is floor temp
52 is humidity

With other value of length, is not working

Thank's
Tiziano

yaacov, 2019-07-07:
did you try `data.length - 6` ?

tty74it, 2019-07-07:
Yes
pi@hassbian:~/.node-red/node_modules/modbus-serial/examples$ node debug-test.js 
Connected
Data length error, expected 11 got 8

tty74it, 2019-07-07:
with debug

pi@hassbian:~/.node-red/node_modules/modbus-serial/examples$ node debug-test.js 
  modbus-serial TCP port: signal connect +0ms
  modbus-serial { action: 'send tcp port',
  modbus-serial   data: <Buffer 01 03 00 00 00 03 05 cb>,
  modbus-serial   buffer: <Buffer 00 01 00 00 00 06 01 03 00 00 00 03>,
  modbus-serial   unitid: 1,
  modbus-serial   functionCode: 3,
  modbus-serial   transactionsId: 1 } +12ms
Connected
  modbus-serial { action: 'receive tcp port strings',
  modbus-serial   data: <Buffer 00 01 00 00 00 06 01 03 06 00 ff 00 00 00 34> } +70ms
  modbus-serial { action: 'parsed tcp port',
  modbus-serial   buffer: <Buffer 01 03 06 00 ff 00 04 b2>,
  modbus-serial   transactionId: 1 } +4ms
Data length error, expected 11 got 8
  modbus-serial TCP port: signal close: false +9ms

yaacov, 2019-07-07:
strange because data.length should be `15` and `15 - 6` is `9`, so it's the same as hardcode `9` ... 
if 9 works ... just use it

yaacov, 2019-07-07:
can you print out what data.length is in this point ( maybe also what data.length - 6 is also ... )

tty74it, 2019-07-07:
I confirm  is 6
i've printed lengthdebug:  ( lengthdebug = data.readUInt16BE(4); )

pi@hassbian:~/.node-red/node_modules/modbus-serial/examples$ node debug-test.js 
  modbus-serial TCP port: signal connect +0ms
  modbus-serial { action: 'send tcp port',
  modbus-serial   data: <Buffer 01 03 00 00 00 03 05 cb>,
  modbus-serial   buffer: <Buffer 00 01 00 00 00 06 01 03 00 00 00 03>,
  modbus-serial   unitid: 1,
  modbus-serial   functionCode: 3,
  modbus-serial   transactionsId: 1 } +13ms
Connected
  modbus-serial { action: 'receive tcp port strings',
  modbus-serial   data: <Buffer 00 01 00 00 00 06 01 03 06 00 fe 00 00 00 34> } +64ms
  modbus-serial { action: 'parsed tcp port',
  modbus-serial   buffer: <Buffer 01 03 06 00 fe 00 00 00 34 09 76>,
  modbus-serial   transactionId: 1,
  modbus-serial   lengthdebug: 6 } +5ms
Receive: [ 254, 0, 52 ]

pi@hassbian:~/.node-red/node_modules/modbus-serial/examples$ cat debug-test.js 
/* eslint-disable no-console, spaced-comment */

/* DEBUG environment variable is used to enable debug logging

  Set the DEBUG variable to be "modbus-serial" before running this example:
  export DEBUG=modbus-serial

  To stop debugging, unset the DEBUG variable:
  unset DEBUG
 */

// create an empty modbus client
//var ModbusRTU = require("modbus-serial");
var ModbusRTU = require("../index");
var client = new ModbusRTU();

// open connection to a serial port
//client.connectRTU("/dev/ttyUSB0", {baudRate: 9600})
client.connectTCP("192.168.88.24", { port: 502 })
    .then(setClient)
    .then(function() {
        console.log("Connected"); })
    .catch(function(e) {
        console.log(e.message); });

function setClient() {
    // set the client's unit id
    // set a timout for requests default is null (no timeout)
    client.setID(1);
    client.setTimeout(1000);

    // run program
    run();
}

function run() {
    // read the 4 registers starting at address 5
    client.readHoldingRegisters(0, 3)
        .then(function(d) {
            console.log("Receive:", d.data); })
        .catch(function(e) {
            console.log(e.message); })
        .then(close);
}

function close() {
    client.close();
}

yaacov, 2019-07-07:
great so you have a solution, by hardocding 9 ( i do not know why data.length - 6 does not work, but for you needs, always getting 3 registers, 9 should work fine)

tty74it, 2019-07-08:
Ok with hardcoded 9, but when i use node-red-contrib-modbus i've receive the same error
"Error: Data length error, expected 7 got 8"

Thank's
Tiziano

yaacov, 2019-07-08:
@biancode ^^

p.s.
If I will change something upstream it will be to `data.length - 6` and not hardcode anything :-)
@tty74bit are you sure `data.length - 6` does not work ?

tty74it, 2019-07-08:
I've misunderstood  data.lenth - 6  changing the code.

I confirm it work:

 length = data.length - 6;           -> 9  (and it work as same hardcoded 9)
 lengthdebug = data.length;      -> 15

pi@hassbian:~/.node-red/node_modules/modbus-serial/examples$ node debug-test.js
  modbus-serial TCP port: signal connect +0ms
  modbus-serial { action: 'send tcp port',
  modbus-serial   data: <Buffer 01 03 00 00 00 03 05 cb>,
  modbus-serial   buffer: <Buffer 00 01 00 00 00 06 01 03 00 00 00 03>,
  modbus-serial   unitid: 1,
  modbus-serial   functionCode: 3,
  modbus-serial   transactionsId: 1 } +11ms
Connected
  modbus-serial { action: 'receive tcp port strings',
  modbus-serial   data: <Buffer 00 01 00 00 00 06 01 03 06 00 fd 00 00 00 32> } +70ms
  modbus-serial { action: 'parsed tcp port',
  modbus-serial   buffer: <Buffer 01 03 06 00 fd 00 00 00 32 cd 74>,
  modbus-serial   transactionId: 1,
  modbus-serial   lengthdebug: 15 } +5ms
Receive: [ 253, 0, 50 ]
  modbus-serial TCP port: signal close: false +11ms

tty74it, 2019-07-08:
I've changed  the  /node-red-contrib-modbus/node_modules/modbus-serial/ports/
 with  data.lenth - 6

And all work in node-red

Thank's
Tiziano

yaacov, 2019-07-08:
> And all work in node-red

@tty74it great :+1: 

@biancode do you think we should make default tcp port ignore the length parameter ?

biancode, 2019-07-30:
@yaacov - good question spec says:

"When  MODBUS  is  carried  over  TCP,  additional  length  information  is  carried  in  the  MBAP  header  to  allow  the  recipient  to  recognize  message  boundaries  even  if  the  message  has  been  split  into  multiple  packets  for  transmission. The existence of explicit and implicit length rules, and use of a CRC-32 error check code (on Ethernet) results in an infinitesimal chance of undetected corruption to a request or response message."

biancode, 2019-07-30:
[Modbus Specs v3](http://modbus.org/docs/Modbus_Messaging_Implementation_Guide_V1_0b.pdf)

yaacov, 2019-07-30:
@biancode  Thanks ! will see if we can use this ...

biancode, 2019-11-17:
I close it here - I mean it will be updated with the next version of modbus-serial and here v5.0.0+

closed 2019-11-17 by biancode

rigibe, 2019-12-04:
> I'm trying to connect to a china MC6 thermostat wifi modbus
> With modbustcp node work
> 
> The error is:
> 
> "Error: Data length error, expected 7 got 8"
> 
> Thank's
> Tiziano
> [MC6-MODBUS.pdf](https://github.com/biancode/node-red-contrib-modbus/files/3364703/MC6-MODBUS.pdf)

I would like to buy the same device to do some testes. Is it a specific modbus version or the standard wifi version?

Thanks
Riccardo

rigibe, 2019-12-04:
> I'm trying to connect to a china MC6 thermostat wifi modbus
> With modbustcp node work
> 
> The error is:
> 
> "Error: Data length error, expected 7 got 8"
> 
> Thank's
> Tiziano
> [MC6-MODBUS.pdf](https://github.com/biancode/node-red-contrib-modbus/files/3364703/MC6-MODBUS.pdf)

I would like to buy this device to do some testing. Is the WiFi version good or is there a specific Modbus version?

tty74it, 2019-12-04:
No the wifi version is good
There is a menu procedure to convert standard config managed in cloud app
in to local modbus config.

In this config the cloud app stop working

Bye
Tiziano

tty74it, 2019-12-04:
> I would like to buy this device to do some testing. Is the WiFi version good or is there a specific Modbus version?

No the wifi version is good
There is a menu procedure to convert standard config managed in cloud app
in to local modbus config.

In this config the cloud app stop working

Bye
Tiziano

mentr, 2020-05-29:
What is the menu procedure to activate modbus tcp on the MC6?

tty74it, 2020-05-29:
[image]