BiancoRoyal/node-red-contrib-modbus #147: Unable to address outside of 65536

stevejfrank, 2019-07-19:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - (X) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
I work with PLCs that have data types with Modbus addresses outside of the range allowed.
 
 **Expected behavior**
I should be able to read and write outside of 65536
 
 **Minimal reproduction of the problem with instructions**
 
   1. FC2 Read address 400000
   2. No request processed and no response
   3. FC2 Read address 0
   4. Request processed

See this manufacturer's default Modbus architecture

Automation Direct
Productivity Suite
https://www.automationdirect.com/microsites/productivity/software/help/Content/198.htm
   
### Your Environment
**Please tell us about your environment:**

  * (X) I am using just the Node-RED package and got [modbus-serial](https://github.com/yaacov/node-modbus-serial) package installed with it
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) package global from source (using git clone)
  * ( ) I have installed [modbus-serial](https://github.com/yaacov/node-modbus-serial) as a global package (using npm i -g)
  
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - ( ) Linux: v 
    - ( ) Windows: v 
    - (X) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : Productivity 1000
     - Version: P1-540
     - Vendor : Automation Direct
     - Link   : https://www.automationdirect.com/microsites/productivity/software/help/Content/198.htm
     
    
* **node-red-contrib-modbus version:** ``

* **Node:** (min. 6.x)
   `node --version` =

thatcadguy, 2019-07-29:
The address field on these nodes is really offset; so address (offset) 0 really means 400001; 65535 means 465536.

stevejfrank, 2019-07-29:
Thanks. Pretty sure I tried that, but will give it another look.

Cheers.

Steve

UdoUSA, 2019-09-18:
The offset field in a Modbus message is only 16 bit and can only take on a value between 0 and 65535. The way register addressing is displayed is not standardized and consistent. Some manufactures use the offset value other the old fashion register numbers (starting with1, 10001, 30001 & 40001). Then the are the ones which put a 0, 1, 3 or 4 in front of the offset to show the register type. In my 25+ years of working with Modbus I have seen it all and sometimes it comes down to trial and error to figure out what addressing scheme is used. Also be careful when reading Devices which feature 32 or 64 bit numbers. Some will send you an error message or ignore you if you do not read the full 32 or 64 bit registers, or if you try to read the second register without the first.

Looking at the link you provided it appears that Automation Direct is adding the register type as an prefix before the offset in the first address column but also shows the traditional register numbering plus the prefix. No idea why the show it both ways, but stick with the first column and strip off the prefix. This will give you the offset.

biancode, 2019-11-17:
reopen if you need @UdoUSA and @thatcadguy special thanks for the good response here

closed 2019-11-17 by biancode