BiancoRoyal/node-red-contrib-modbus #148: [Question] Reading from Master

Sineos, 2019-07-26:
My setup:
- A PLC acting as Master
- A `Modbus Server` Node 
- A `Modbus Read` Node

My understanding:
- The Master (PLC) connects to the the Slave (Server Node)
- The Master pushes its registers to the Slave
- The Read Node reads the relevant registers from the Server

This works and I can get the data for further processing in the flow.

I would think that the Slave gets the data basically in real time but for processing of the data in the flow, I need to regularly pull the server.

Is there a way to have the `Modbus Server` Node actively pushing new data down the flow or is my understanding of the concept false?

biancode, 2019-07-30:
Hi @Sineos, Modbus is polling take a look here [Modbus.org](http://modbus.org/tech.php)

biancode, 2019-07-30:
You could build some with Node-RED and the context inside Node-RED to watch values and to send events on value changes. Does it clearify your question?

Sineos, 2019-07-30:
@biancode many thanks for coming back on this. 

Do I understand this correctly that the Slave/Server is polling the Master/Client?

jason-turner2, 2019-10-28:
Hi @Sineos . In Modbus, the Server cannot initiate the exchange, so it does not poll. But you can put a Modbus client on the same Node Red flow as the server, and have the new client do the extra polling in order to push the processing if you need to. I hope I've answered your question. If not, please feel free to ask.

biancode, 2019-11-17:
reopen if you need more

closed 2019-11-17 by biancode