BiancoRoyal/node-red-contrib-modbus #149: Writing to Flex Server Buffers with Message

jackmarshall9357, 2019-07-31:
**I'm submitting a ...**  (check one with "x")

## Support/Advice
 - support/advice request => **Please do not submit support requests here**, instead use [the link to our support page](https://bianco-royal.cloud/supporter/#/content/backer), please!

## Feature
 - ( ) feature request
 
 **What is the motivation / use case for changing the behavior?**
 

## Simple question to get some hint
 - ( ) simple question => please visit the Wiki, the YouTube tutorials and the Info tab by clicking on the node inside Node-RED before
 - ( ) help wanted => you read all information and watched relevant YouTube tutorials, but you need more help

## Problem
 - ( ) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - (x) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 
 When trying to write values (with a message) to a Flex Server node I get no response or change in the servers buffers. I also get no error.
 
 **Expected behavior**
 
 i would Expect Either an Error if the payload is in the wrong format, or if correct the valuess of th buffers to change.
 **Minimal reproduction of the problem with instructions**
 
   1. Flow with problem
   `[
    {
        "id": "7c901ce1.97f324",
        "type": "tab",
        "label": "Flow 3",
        "disabled": false,
        "info": ""
    },
    {
        "id": "9f1c31db.f86f2",
        "type": "inject",
        "z": "7c901ce1.97f324",
        "name": "",
        "topic": "",
        "payload": "[0]",
        "payloadType": "bin",
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "x": 170,
        "y": 160,
        "wires": [
            [
                "fcd0afb0.7a1cf"
            ]
        ]
    },
    {
        "id": "fa376833.7c3268",
        "type": "inject",
        "z": "7c901ce1.97f324",
        "name": "",
        "topic": "",
        "payload": "[1]",
        "payloadType": "bin",
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "x": 170,
[... 343 lines cut ...]
  Yes, modbus-serial is the name but it does all, TCP and Serial.
  The node-modbus package is just for the Modbus Server node.  
  
  * Device :
    
    - ( ) macOS: v 
    - (x) Linux: v 
    - ( ) Windows: v 
    - ( ) Raspbian: v 
    - ( ) Other:
         - Name:
         - Version:
   
   * Description of the connecting Modbus system:
   
     - Name   : node-red-contrib-modbus
     - Version: 4.1.3
     - Vendor :
     - Link   : https://
     
    
* **node-red-contrib-modbus version:** `4.1.3`

* **Node:** (min. 6.x)
   `node --version` = `10.15.3`

biancode, 2019-08-30:
Hi, there is no way to send data directly to the server input. You have to use the read/write operations via modbus here. You could work with the Node-RED context, but this is not implemented for now and would be a feature request.
Happy wiring!

jackmarshall9357, 2019-08-30:
Hi thanks for the Reply, I get using the Node-Red Context but to me at least it makes more sense to do this with a msg directly into the server?  

But without either feature it is impossible to write into the 2 Buffers (Discrete Inputs, and Holding Registers) that you can't use Modbus to write into, which makes having them present on the server rather redundant? it also makes it impossible to use the server in the scenario of providing Read Only Values.

but PR #140 should make writing in via msg work? I was hoping that there was a test implementation I could see, but as you can see from the flow I submitted got nothing in the way of feedback or errors. Would you be happy with a new PR to write in via msg to the flex-server input?

biancode, 2019-08-30:
Oh, I forgot this change, sorry ... then we should reopen this one.

biancode, 2019-12-03:
PR #140 released with 5.1.0

biancode, 2019-12-03:
[image]

closed 2019-12-03 by biancode