BiancoRoyal/node-red-contrib-modbus #151: Modbus Read IO File Question

Psens666, 2019-08-26:
Hello,

in the modbus read node under optionals is a point use io file.
Is there a way to import a file with my registers i want to read?

In witch format this file shoud be?

Dominic

biancode, 2019-08-30:
Hello, IO is to use Codesys CSV exports to name register variables - see the example inside Node-RED Import named "CODESYS-CSV-To-IO". With that you are able to create the JSON IO file for the modbus nodes. You have to play with to see how it works for now. Maybe I find some time later to record a video.

closed 2019-08-30 by biancode

Kortenbach, 2022-06-18:
Hi, biancode,
I have looked at the CODESYS-CSV-To-IO example. It has some errors in the "filter named register" function (I changed the example documentation here to include a few changes to the code) I'm able now to generate the sdm630.json file but in another folder than in the example (I'm running a docker image). This is where I get stuck. How does Modbus read refer to this JSON file. There is no documentation and the example code doesn't work. Can you give me a few hints on how to proceed?

Kortenbach, 2022-06-18:
@[Psens666](https://github.com/Psens666) 
DId you manage to get this working?