BiancoRoyal/node-red-contrib-modbus #153: Unhandled Promise Rejection error in Node-RED 1.0.0

jhottell, 2019-10-03:
#120  Problem Bug Report
 node-red-contrib-modbus does not work properly on Node-RED version 1.0.0
Tested on three different machines, some clean node-red installs one a version upgrade. 

 **Current behavior**
node-red-contrib-modbus does not read or write data in Node-RED version 1.0.0. The modbus server does seem to work.

 **Minimal reproduction of the problem with instructions**
To recreate the issue, simply drop a Modbus Server and a Modbus Read on a screen, configure the Modbus read to read a value from the server. This works fine on Node-RED 19.x. On Node-RED 1.0.0 and it starts barfing these errors:

```
(node:4352) UnhandledPromiseRejectionWarning: TypeError: a is not a function
    at cberr (C:\Users\FILEA\.node-red\node_modules\node-red-contrib-modbus\modbus\maps\modbus-client.js:565:19)
    at <anonymous>
    at process._tickCallback (internal/process/next_tick.js:189:7)
(node:4352) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 26)
```
### Your Environment
**Please tell us about your environment:**
Node-Red 1.0.0

  * Device :
     - (x) Linux: Ubuntu 18.4 
     - (x) Windows: 10 Home 10.0.18362
   
   * Description of the connecting Modbus system:
None, connecting to its own server.  
   
* **node-red-contrib-modbus version:** ``
node-red-contrib-modbus 4.1.1
also tried 
node-red-contrib-modbus 4.1.3
Installed from the Node-Red  palette manager.v

* **Node:** (min. 6.x)
   `node --version` = 8.12.0 and also tried on node 10.16.3

Thanks!

jhottell, 2019-10-03:
Indeed rolling Node-RED back to 0.20.8 it works again. After reading some on the release of v1.0 I guess there are quite a few breaking changes, so this may not be completely unexpected.

biancode, 2019-10-04:
#144

closed 2019-10-04 by biancode