BiancoRoyal/node-red-contrib-modbus #155: Modbus Flex Getter - Type Error: a is not a function

ralieghnet, 2019-10-08:
#120  Problem
 - ( x) bug report => please search github for a similar issue or PR and test Modbus problems with the [modbus-serial](https://github.com/yaacov/node-modbus-serial) package before submitting
 - ( x) unexpected behavior => please visit the Wiki before
 
 **Current behavior**
 Attempting to use Modbus flex Getter
 
 **Expected behavior**
 Expecting to get result from a single holding register FC3
 
 **Minimal reproduction of the problem with instructions**
 Basic single line setup. 
Inject -> Function -> modbus glex getter -> debug
pulling single message from unitid 245

function:
  fc: 3,
  unitid: 245,
  address: 0,
  quantity: 1

console debug message:

2019-10-08T23:28:53.478Z modbus-serial TCP port: signal connect
2019-10-08T23:28:57.161Z modbus-serial { action: 'send tcp port',
  data: <Buffer f5 04 00 00 00 01 24 be>,
  buffer: <Buffer 00 01 00 00 00 06 f5 04 00 00 00 01>,
  unitid: 245,
  functionCode: 3,
  transactionsId: 1 }
2019-10-08T23:28:57.167Z modbus-serial { action: 'receive tcp port strings',
  data: <Buffer 00 01 00 00 00 05 f5 04 02 08 22> }
2019-10-08T23:28:57.169Z modbus-serial { action: 'parsed tcp port',
  buffer: <Buffer f5 04 02 08 22 8f 3c>,
  transactionId: 1 }
(node:17) UnhandledPromiseRejectionWarning: TypeError: a is not a function
    at activateSending (/data/node_modules/node-red-contrib-modbus/modbus/maps/modbus-client.js:571:20)
    at process._tickCallback (internal/process/next_tick.js:68:7)
(node:17) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 4)

I parsed the received message and it is correct.

biancode, 2019-10-09:
Node-RED v1.x?

ralieghnet, 2019-10-09:
I'm currently running 1.0.1

biancode, 2019-10-09:
It is not v1.x compatible see #144

closed 2019-10-09 by biancode